jgyates/genmon #450: Upgraded from V1.14.33 to v1.15.02 - Lost Web Interface

USAFPride, 2020-10-11:
Just upgraded via web interface (and then manually) to v15.02 and I've lost the web interface.

`./startgenmon.sh start` (or restart) does not bring it back
`/var/log/genloader.log` shows the last error to be:
`2020-09-19 17:36:51,417 : Warning: required library pyotp not installed. Attempting to install....`
but that is properly installed (plus that error is from prior date).
```sudo pip install pyopt
Requirement already satisfied: pyopt in /usr/local/lib/python2.7/dist-packages (0.84)
```
htop shows genmon.py is running (along with genserv)

What is interesting is the permissions on the *conf* files in the /etc folder
`
- -rw-r--r--  1 root root  155 Nov 30  2019 genalexa.conf
- -rwxr-xr-x  1 root root  309 Jun 19  2019 genemail2sms.conf
- -rwxr-xr-x  1 root root 1.7K Jun 19  2019 genexercise.conf
- -rwxr-xr-x  1 root root  205 Jun 19  2019 gengpioin.conf
- -rwxr-xr-x  1 root root 2.7K Oct 10 22:00 genloader.conf
- -rwxr-xr-x  1 root root  12K Oct 10 21:25 genmon.conf
- -rwxr-xr-x  1 root root 1.9K Jun 19  2019 genmqtt.conf
- -rwxr-xr-x  1 root root  172 Jun 19  2019 genpushover.conf
- -rwxr-xr-x  1 root root  556 Jun 19  2019 genslack.conf
- -rwxr-xr-x  1 root root  441 Jun 19  2019 gensms.conf
- -rw-r--r--  1 root root  361 Sep  5 09:56 gensnmp.conf
- -rw-r--r--  1 root root  550 Sep  5 09:56 gentankdiy.conf
- -rw-r--r--  1 root root  530 Jun 19  2019 gentankutil.conf
- -rw-r--r--  1 root root  651 Sep  5 09:56 gentemp.conf
`
Either way, I'm a bit stumped on how to get it working again.

jgyates, 2020-10-11:
To get a data point, log in to your pi via ssh and type this:

     pip freeze | grep pyotp

you should get this:

     pyotp==2.3.0

If you had another version of pyotp installed this could be the problem. If you have 2.3.0 then there is some other issue.

USAFPride, 2020-10-11:
`:/var/log $ pip freeze | grep pyotp
pyotp==2.3.0
`

jgyates, 2020-10-11:
Next would be to determine if the problem is that genmon.py is not loading or genserver.py is not loading.

From the genmon directory type this:

      python ClientInterface.py 

If genmon is not loaded it will timeout in about 10 seconds and display "Genmon is not loaded". If genmon.py is loaded it will display a prompt. At the prompt type:

      generator: sendlogfiles 
     
Then type:

      exit

to quit the program. This will send me your log files. I would be interested in looking at /var/log/genserv.py

If you got "Genmon is not loaded" then look at /var/log/gennon.log to check for errors and past them to this thread.

USAFPride, 2020-10-11:
Logs sent.

Below is /var/log/genserv.py (you can see where I was trying to force it to start before I decided to raise the white flag). No "Genmon is not loaded" errors.

> 2020-09-19 17:36:17,846 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
> 2020-09-19 17:36:24,422 : Root
> 2020-09-19 17:36:28,515 : Root
> 2020-09-19 17:36:39,147 : Script: /bin/bash /home/pi/genmon/genmonmaint.sh -u -n
> 2020-09-19 17:36:46,945 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart -c /etc/genmon/
> 2020-09-19 17:37:07,049 : Root
> 2020-09-19 17:37:08,621 : Root
> 2020-09-19 17:37:22,860 : Root
> 2020-09-19 17:37:32,588 : Root
> 2020-09-19 17:37:46,590 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
> 2020-10-10 21:25:15,334 : Script: /bin/bash /home/pi/genmon/genmonmaint.sh -u -n
> 2020-10-10 21:25:24,783 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart -c /etc/genmon/
> 2020-10-10 21:25:40,768 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
> 2020-10-10 21:29:29,442 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
> 2020-10-10 21:33:47,838 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
> 2020-10-10 21:39:25,306 : Starting genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
> 2020-10-10 21:41:06,961 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
> 2020-10-10 21:43:41,788 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
> 2020-10-10 21:49:56,198 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
> 2020-10-10 21:59:01,370 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
> 2020-10-10 22:08:21,450 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True

jgyates, 2020-10-11:
Do you have outgoing email enabled? I have not received your log files.

If you were able to use ClientInterface.py to connect to genon then that means you genmon.py is loaded so the problem could be with genserv.py. What happens if you type this from the genmon directory:

       sudo python genserv.py &

This will load genserv.py manually, bypassing genloader.py. To kill it manually you can type:

        sudo pkill -u root -f genserv.py

Post any error messages to this thread. If it loads then see if you can access the web interface before killing it.

USAFPride, 2020-10-11:
Yes, outgoing mail is enabled. Where is the log file saved off to?  I can always manually email it.  That being said, 
`sudo python genserv.py &`

This does allow me to load the webpage successfully.

jgyates, 2020-10-11:
then type:

     sudo pkill -u root -f genserv.py

To kill the copy of genserv.py you loaded. Then type:

     ./startgenmon.sh restart

This should reload genmon. If it works after this, the I am not sure what occurred. It could just be a delay for some reason in the reload.

The log files are in /var/log/. The log files for genmon start with gen* and my*

Still have not received any log files.

jgyates, 2020-10-11:
you may just need to restart your pi. The Flask library may need a refresh, but that is just a guess.

USAFPride, 2020-10-11:
No joy on the 
`./startgenmon.sh restart`

Nor the reboot. Logs would be coming from a NVR gmail address.  If you still don't have them, shoot me a message of where to upload them manually.

jgyates, 2020-10-11:
Was there any output on the ./startgenmon.sh restart command?

You can email them to generatormonitor.software@gmail.com

jgyates, 2020-10-11:
You mentioned this message:

      Requirement already satisfied: pyopt in /usr/local/lib/python2.7/dist-packages (0.84)

was this in genloader.log?

You also said that htop shows genmon.py and genserv.py loaded. Is this correct?

jgyates, 2020-10-11:
Try typing this from the genmon directory:

      sudo python genloader.py -r -c /etc/genmon/

jgyates, 2020-10-11:
if the above command fails then I would try to replace your /etc/genmon/genloader.conf file. From the genmon directory type:

     sudo cp ./conf/genloader.conf /etc/genmon

or just remove the file and genloader.py will recopy it:

      sudo rm /etc/genmon/genloader.conf

If you do this you will need to re-enable any add on programs. Your add on settings will still be preserved.

USAFPride, 2020-10-11:
Was there any output on the ./startgenmon.sh restart command? - Just the normal Stopping and starting
was this in genloader.log? - Yes
htop shows genmon.py and genserv.py loaded. Is this correct? - Yes
sudo python genloader.py -r -c /etc/genmon/ - No change
sudo rm /etc/genmon/genloader.conf - **_Success_** (after ./startgenmon.sh restart )
Logs also emailed

jgyates, 2020-10-11:
looks like your genloader.conf got corrupted. I am glad it is working for you now.

closed 2020-10-11 by jgyates