jgyates/genmon #451: Remote access with https enabled

DJT1423, 2020-10-11:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

When I enable security in the web interface, I am prompted for my username and password as expected when connecting to the interface internally via http://xxx.xxx.xxx.xxx:8000. I had assumed I would see the same thing when connecting from outside my home network, however so far I am missing something in the configuration,  or misunderstood the purpose of securing access.

### Actual Behavior

To access the web interface from outside my network, I have created an account with No-Ip, and I have opened port 8000 on my router and forwarded it to port 8000 on my pi (tried a few other combinations as well). However so far I have been unable to remotely access the web interface. I have tried:
http://mynoipdns.net:8000
http://mynoipdns.net
https://mynoipdns.net:8000
https://mynoipdns.net

Whatever combination I try simply fails to load or times out. 

I am sure I have missed something simple, but can't  seem to get it to work. I want to be able to access the interface in the event I get an alarm or error while away from home.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Evolution 2 
- Generator Registers: Submit via the About page
- Genmon Version: the latest

skipfire, 2020-10-11:
I had this setup briefly with mine to install the certificate so I know it can work.

1. If you only have https enabled, then port 8000 isn't the one you need to forward, you would need 443.
2. If you check the IP of your noipdns entry with nslookup or ping or something does it return the right value?
3. Have you tried from outside your home network?  I had a couple of routers in the past that did not like coming back in from the outside.
4. Is the port possibly blocked by your ISP?  If you have telnet installed, you can try 'telnet mynoipdns.net 8000' and if it comes up with an all blank screen it connected successfully and it should not be a port issue.
5.

DJT1423, 2020-10-11:
1. If you only have https enabled, then port 8000 isn't the one you need to forward, you would need 443.

Thank you, that did it. I have port forwarded 443 external to 443 internal on the pi. I also learned I need to use https:// on my ddns address, however even then Google tries to block it as unsafe.

2. If you check the IP of your noipdns entry with nslookup or ping or something does it return the right value?

Yes, it worked fine.

3. Have you tried from outside your home network? I had a couple of routers in the past that did not like coming back in from the outside.

Yes, I use my phone with wifi turned off so I am coming in from outside on the cellular network.

4. Is the port possibly blocked by your ISP? If you have telnet installed, you can try 'telnet mynoipdns.net 8000' and if it comes up with an all blank screen it connected successfully and it should not be a port issue.

Port 443 is not blocked and works fine, once I use the correct port :-)

Thanks again. External access is a significant benefit, with the added security of password protection.

David

closed 2020-10-11 by DJT1423

skipfire, 2020-10-11:
Get a valid cert installed and that chrome warning will go away.

https://github.com/jgyates/genmon/wiki/Appendix-K-Adding-LetsEncrypt-certificates-to-Genmon