jgyates/genmon #452: sendmail broken?

bbrown-sc, 2020-10-11:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Didn't receive my gen exercise email this morning. Test Email Settings from the settings page doesn't work. 

I haven't changed any configurations. I have only have done software updates.

### Actual Behavior

Get an error dialog "An error occured: Error sending test email : Connection unexpectedly closed Please try again." BTW, occurred is spelled wrong : )

From mymail.log

_2020-10-11 15:30:39,483 : Error SMTP sendmail: (535, '5.7.0 (#AUTH005) Too many bad auth attempts.') : mymail.py:435
2020-10-11 15:30:39,485 : Error in SendMailThread, retrying (2): Connection unexpectedly closed : mymail.py:463_

### Steps to Reproduce (including precondition)

Test Email Settings from the settings page

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: QT02524, H-100, HTS xfer sw
- Generator Registers: Submit via the About page - can't, email doesn't work : )
- Genmon Version: V1.15.02

jgyates, 2020-10-11:
Can you log into your email provider with the same credentials? Nothing relating to email has changed in a while. The error "Too many bad auth attempts" implies that the email pw or username has changed or is incorrect. Also, the file /var/log/mymail.log should indicate when you started seeing authentication errors, or at least in the ballpark since genmon does not send emails unless a state of the generator changes. 

I corrected the typo. Thanks

jgyates, 2020-10-11:
Also, depending on your email provider, you may have to enable use less secure apps in the email settings of gmail, yahoo, etc. This thread has some info on this topic: https://github.com/jgyates/genmon/issues/351

bbrown-sc, 2020-10-12:
The error occurs every 2 minutes so all my logs (0-5) were filled with that. Looking into enable use less secure apps. Trying to figure out how to do that with ATT/Yahoo mail.

jgyates, 2020-10-12:
yahoo does tend to hide this setting from what I have seen. I use gmail. I generally setup an account that I only use for sending from apps like genmon. Look in Account Info -> Account Security - > Generate App Password, then select Mac Mail (this is IMAP mail client so it should work). 

This link should have server address and port:
https://support.sequentum.com/hc/en-us/articles/360040569954-Error-Connecting-to-Yahoo-Mail-SMTP-Server#:~:text=Server%20Address%3A%20smtp.mail.,Number%3A%20465%20(With%20SSL)

bbrown-sc, 2020-10-12:
Looks like ATT mail made changes to their authentication requirements. Now using OAuth or secure mail keys. I switched to a different outgoing mail server and it's working now. 

I do still get this in the mail logs.

_2020-10-11 18:12:58,324 : Error in MyConfig:ReadValue: extend_wait: invalid literal for int() with base 10: '' : myconfig.py:90_

closed 2020-10-12 by bbrown-sc

jgyates, 2020-10-12:
That message is OK. It just means you don't have a value for extended_wait so it defaults to zero. this parameter is on the advanced settings page (double click the gear icon in the top right on the web interface)