jgyates/genmon #454: Question and possible enhancement for MQTT Tank Level

GlennGoddard, 2020-10-13:
More of a feature request then a bug.
Ever since I pulled my Tank Utility into Home Assistant, I disabled external tank data in GenMon.  This was not really a problem since I port all of GenMon via MQTT to Home Assistant.  I just turned it off in GenMon (months ago).  I just tried to turn it back on and no matter what I do I can not display External Tank Level data.  I had also set to "Natural Gas" just so I did not get any gauge at all.

I read the recent change that computed time left on the tank.  So I thought I would look at it.
So, could we have a External Tank Level read via MQTT to import into GenMon?
Or (so I don't have to riffle through you code) let me know how you did the calculation so I can just make that calculation in HA...or if put a fuel usage out to MQTT I could easily use that (ie 1 Gal/min usage or what ever unit per time).  After I read the not about time left on tank, I thought that was a good idea.  I guess I could calculate (guess) fuel load off of generator load but it would be much easier (for me) to just expand off (i.e. plagiarize) your work.

I see (in MQTT) Fuel consumption in last 30 days, but that is it.  It might be that the value is there but only published when generator is in use.  Ok, I see it in Outage Log but that would be a pain to pull out of that line.

Also, I have seen several people put Genmon MQTT as sensors into Home Assistant, as have I; I am wondering if you use Home Assistant or if you would want a YAML template of what I use for use by others.  I could compile it and maybe you put it up in Wiki.

jgyates, 2020-10-13:
I would need to see your logs to determine what the issues was with your external tank. You can send them on the about page if your outbound email is setup and working.

You asked for this:
"could we have a External Tank Level read via MQTT to import into GenMon?"

You want to import a tank level via MQTT? or do you want to read the tank level via MQTT? I am not clear on what you want.

the calculations are based the fuel consumption specs for each model. I take the fuel used at half load and full load and create a point slope equation (percent load is one axis, fuel used is the other). then i take the current power as a percentage of total rated kw and plug it into the equation to solve for the fuel used at any given time.

https://github.com/jgyates/genmon/blob/master/genmonlib/controller.py#L1317

Fuel consumption numbers are stored in /genmon/data/  in the files EvoAC_Fuel.txt and EvoLC_Fuel.txt. For industrial generators you input the fuel usage at 50% and 100% manually on the advanced page.

I don't use HA, but if you want to post your template to this thread I am sure that others will benefit.

jgyates, 2020-10-13:
I recieved your log files.

You two have a repeating entries in your genmon.log file:

      Invalid input for UnitsOut: 21.70 cubic feet
      Invalid input for UnitsOut: 43.05 cubic feet

UnitsOut is a function that will strip away the units and just return the numeric portion. I would look for these two numbers in /etc/genmon/genmon.conf and remove them.

Also, this error is in gentankutil.log:

     Error in MyConfig:ReadValue: poll_frequency: could not convert string to float

It looks like you have non numeric value for the parameter poll_freq. I would remove this value from /etc/genmon/gentankutil.conf

I would guess that these values are cause from either editing the conf files manually or modifying the source code. I would not recommend editing the conf files directly as there are a lot of settings and the type checking is done at runtime but the web interface provides some visual type checking when entering the data. Let me know if that helps or if you have any other questions.

GlennGoddard, 2020-10-13:
Import tank level into GenMon via MQTT.
I can publish % and gallons (with calculations...easy)
I have Tank Utility being read in Home Assistant and from there I can publish to MQTT.
I just don't know if it is worth your effort for this edge case to read from MQTT.

Logs submitted.

I also have the tank level stored in InfluxDB and then read into Grafana for graphing.
[image]
[image]

I will pull out my YAML for the genmon later this week.

GlennGoddard, 2020-10-13:
I don't think I have ever manually edited the config files; the only source code I change previously was the section to change the version so I could verify the update available in MQTT, but that was overwritten by newer versions since then.
I hope the SD card isn't going out again; I had to start from scratch (not hard mind you) a few months ago because my SD card decided it wasn't going to play any more.  That SD card lasted for a few years thou.

jgyates, 2020-10-13:
I may have found an issue with the code. The UnitsOut function is complaining because the units is two words while it was coded to expect only one. I am not sure where this is coming from in the code. Your fuel type is natural gas in your log data. Is that the case? You graph above shows propane. Check your fuel type on the setting page.

GlennGoddard, 2020-10-13:
I changed it to Natural Gas to get rid of the gauge off the status page since I could not get Tank Utility Level working.

GlennGoddard, 2020-10-13:
Configurations for Home Assistant.

```
##sensor.yaml
#Tank Utility (Propane Tank Level)
- platform: tank_utility
  email: !secret TankUtilityUser
  password: !secret TankUtilityPassword
  devices:
    - !secret TankUtilityDevice

##sensor.yaml
#Generator Monitor
- platform: mqtt
  name: "Utility Voltage"
  state_topic: 'generator/Outage/Utility Voltage/value'
  unit_of_measurement: 'V'
  icon: mdi:transmission-tower
- platform: mqtt
  name: "Generator Voltage"
  state_topic: 'generator/Status/Engine/Output Voltage/value'
  unit_of_measurement: 'V'
  icon: mdi:flash
- platform: mqtt
  name: "Generator Frequency"
  state_topic: 'generator/Status/Engine/Frequency/value'
  unit_of_measurement: 'Hz'
  icon: mdi:sine-wave
- platform: mqtt
  name: "Generator Battery"
  state_topic: 'generator/Status/Engine/Battery Voltage/value'
  unit_of_measurement: 'V'
  icon: mdi:car-battery
- platform: mqtt
  name: "Generator Current"
  state_topic: 'generator/Status/Engine/Output Current/value'
  unit_of_measurement: 'A'
  icon: mdi:current-ac
- platform: mqtt
  name: "Generator Power"
  state_topic: 'generator/Status/Engine/Output Power (Single Phase)/value'
  unit_of_measurement: 'kW'
  icon: mdi:flash-circle
- platform: mqtt
  name: "Generator Speed"
  state_topic: 'generator/Status/Engine/RPM/value'
  unit_of_measurement: 'RPM'
  icon: mdi:fan
- platform: mqtt
  name: "Generator Pi Time"
  state_topic: 'generator/Status/Time/Monitor Time'
  unit_of_measurement: ''
  icon: mdi:clock
- platform: mqtt
  name: "Generator Time"
  state_topic: 'generator/Status/Time/Generator Time'
  unit_of_measurement: ''
  icon: mdi:clock

## binary_sensor.yaml
#Generator Monitor
- platform: mqtt
  name: "Genmon Power Outage"
  state_topic: 'generator/Outage/System In Outage'
  device_class: problem
  payload_on: 'Yes'
  payload_off: 'No'
- platform: mqtt
  name: "Genmon Update Available"
  state_topic: 'generator/Monitor/Generator Monitor Stats/Update Available'
  device_class: moving
  payload_on: 'Yes'
  payload_off: 'No'

```

jgyates, 2020-10-13:
I just checked in an update that will likely get rid of the "Invalid input for UnitsOut" error. This would occur if the fuel type is Natural Gas and MQTT or SNMP was enabled.

I will think on the setting the tank data via MQTT. If I can do this with a simple python library then I will consider it. If it requires installing mosquitto or some other message broker then I will opt not to do this as this would be a bit beyond the scope of the project.

GlennGoddard, 2020-10-13:
Well that is an interesting set of conditions to cause an issue.   I had two of the three.
I, just like you have scripts that publish to MQTT.  I will have to look to see if I have any scripts that read MQTT.

jgyates, 2020-10-13:
I looked into the setting of the tank data. I could just subscribe to the topic of you tank data and your broker/server would notify genmon (that ends up being the easy part). The part that complicates things is that genmon must know about the intention to use external tank data when it starts. This requires a conf file setting in genmon.conf. Setting this conf file entry in genmqtt.py would make a gauge show up, but if one is using genmqtt and not using this feature then an additional unused gauge shows up. If I use a separate add on for this it would require a lot more code so I am opting not to do this as the maintenance required outweighs the benefit.

closed 2020-10-13 by jgyates