jgyates/genmon #455: GenMon HAT?

skipfire, 2020-10-17:
Has there been any discussion of making a HAT specifically for this?  What I'm thinking is:
- 8-pin coming in
- 12v-5v buck converter and boost converter for the older ones with the flaky 5v
- MAX3232 for the TTL

This would replace 
- ttl converter
- DB9 adapter
- motorcycle USB power adapter
- cable making
- all the potential assembly problems

jgyates, 2020-10-17:
@lmamakos had mentioned this in the past, but I don't know if everyone has the same idea regarding features. I do like the idea of making things easier to install / setup. He had mentioned an isolated RS232 interface which sounded like a good idea. That combined with what you are suggesting may be the right combination. 

FYI: @lmamakos has created this HAT in the past:

https://github.com/lmamakos/RPi-pHat-Thermocouple

skipfire, 2020-10-17:
I've created about a dozen HATs, including a thermostat HAT which has the 12v-5v circuitry.  I haven't pulled in the 5v booster or validated the Tx/Rx connections yet.  Might also make a 3D-printable case that has spots for neodymium magnets for mounting as the 8-pin in 2x4 form would not fit in the HighPi or any other case.

[image]

skipfire, 2020-10-17:
If anyone wants to collaborate I'm up for it.

lmamakos, 2020-10-18:
I built a board that has 3 thermocouple interfaces on it, some break-out for the I2C signals and one-wire bus on the Raspberry Pi, and a galvanicly isoloated RS-232/TTL converter.   

The idea is that the one side of that isolated interface circuit is powered by the +5V and (isolated) ground coming from the Generac controller, and there's a LDO regulator to drop down the input voltage to run that half of the circuit (only 20-50mA depending on how bright the LED is).
[image]

As it turns out, the thermocouple interfaces are not as useful as I hoped, and probably not really necessary. 

I'm thinking of respinning the board to drop all the thermocouple stuff - that's built around a part that has a buried pad on the bottom of the package and is impossible to hand solder.  The rest, while using all SMD parts, can be hand soldered.  I spent a little bit of time and I think that I can adapt the board to use parts that are available on the JLCPCB assembly service (except for the through-hole connectors).

I had not considered doing a 12V power supply to run the Raspberry Pi, but that LM2596 part looks like it might be a good match, and is also available in an SMD version that JLCPCB has available in their inventory.    I was a little unsure of doing a power supply to run the Pi, given the rather nasty environment this thing runs in, but if your power supply seems to work OK, that's a good start!

My goal was to be able to have this design mostly fabricated by some PCB assembly house on demand.  I'm not sure I want to be in the business of selling/supporting Raspberry Pi hats, so this is an important consideration for me.  I was hoping to spend some time on this over the winter during some vacation..  In the mean time, you can take a look at the thing that I did, posted on github.  I used KiCAD for the PCB design.

skipfire, 2020-10-18:
I've used the 12v power design in a few things both inside and outside, but you bring up a good point with using it in the generator where temps get even higher when it runs.  I just rechecked and the weakest component are the caps which go -40 to 85 matching the MAX3232 and that still exceeds the Pi's CPU temp allowance.  The advantage to using the 12v is the reports of low voltage errors when running from 5v and that would probably be worse with a Pi 4.  That said, the 12 isn't available on the older controllers, so to get rid of the low-voltage when using the ~5v pin a booster circuit to feed the 12v buck side, though I haven't found specs on max current output from the RS232 5v on the controller.  Then a few options could be done from jumpers to control what power paths are done, a couple diodes and a transistor to kill the 5v side if the 12v is powered, to having the 5v components being left off by default to reduce cost.  So far the parts at single quantities are $15 and the prototype run of 10 boards is $190, so cost at least seems reasonable.  I've been using Pad2Pad for design and pcb manufacturing, but I've always done assembly myself.

I'll take a look at KiCAD.

lmamakos, 2020-10-18:
My understanding is that the power available from the controller is somewhat limited, so you end up powering the Raspberry Pi from power derived from the generator's starter battery.   

It was this fact that directly led me to the isolated RS-232 interface in the design - because of the potential ground-loop situation.  In particular, the ground reference for the RS-232 signals from the controller.  If you only connect RX and TX signals, and rely on the power supply connection via the battery for the reference.. I think you'd see terrible noise on the signals when, e.g., the starter on the generator kicks in and drops a large inductive load.  And then you have the possibility of a running way excessive current through the ground connection to the Raspberry Pi and controller if the main ground connection to the generator and starter is compromised, somehow.   The isolation component from Analog Devices was pretty inexpensive, around $2 or so and a good solution.

I was looking at the data sheets for those switching buck converter parts, and I'm wondering if there's enough space on the board to layout the circuit using their recommended board layout designs.  There's some high-current and steep edges in the switching components that's going to required some large, low-inductance traces on the board.  I suppose I'll have to try to lay it out and see how it fits.

I have to say, the switching regulator gives me pause; I do this fiddling around as a hobby but I try to know what I don't know.  I think there's some amount of voodoo in getting these circuits running reliably and ideally without having to do too many board spins to get it right.  There's also a question of how much pre-filtering is required for the input power.  My earlier comment was more directed towards the ugly electrical environment when the starter for the generator engages and stops -- that's got to be thrilling!   And the large input capacitor probably needs an inductor/choke to both filter and limit the in-rush current when it's connected to power.  And reading the datasheet, you have to be real careful about having the load disconnect or short and maybe having a diode back to the input to dump the energy.   I'll have to read that data sheet very carefully..

Anyway, as @jgyates had mentioned, there's a github repo at https://github.com/lmamakos/RPi-pHat-Thermocouple which has all the files, including a PDF of the existing schematic that should illustrate the isolated RS-232 interface.  I'd re-use that pretty much as-is, except that I'd choose a different reverse-polarity diode part in a more common package that would be cheaper and easier to solder.  I stupidly just searched for an 0603 diode and found one... however, there are zillions of parts in a more commonly available package.  Also change out the 3.3v LDO regulator there for one in a SOT89 package that JLCPCB stocks that's more common as well.

skipfire, 2020-10-18:
I have a test power board of this and I have semi-annual service on Tuesday.  I'll throw it in with some monitoring and see what happens with all the starts and stops.  I will hook up a Pi for load and get it doing some processing so it is producing a higher than normal load.  However, given that the 12v on the board is used to power the WiFi add-on, I am pretty confident that it is a sufficiently clean and stable power source to use, especially with a device that only needs 5v behind a buck circuit.

USAFPride, 2020-10-18:
For those of us with older controllers, it would be nice to have the ability to add [SCT-013-000](https://www.amazon.com/SCT-013-000-Non-invasive-Current-Sensor-Transformer/dp/B07FZZZ62L) current transformers and associated electronic bits to read current draw when the generator is running.  eg: http://lechacalshop.com/gb/internetofthing/63-rpizct3v1.html

skipfire, 2020-10-18:
Seems like a reasonable idea, though it would of course require some software as well.  There should be enough room for an MCP3008, the resistors, and the jacks.  I haven't used those CTs but a quick pass looks like it they shouldn't be any problem.

jgyates, 2020-10-18:
I can incorporate the software. One thing to consider if you are going to add the ability to add CTs is that you make the burden/terminator resistor relatively easy to change. This will give more flexibility for people with larger rated lines if they have to upgrade the CT to a larger one. The openenergy monitor project has a lot of discussion and information on this topic. I use this project with a dedicated pi to monitor my utility lines continuously.

https://learn.openenergymonitor.org/electricity-monitoring/ct-sensors/interface-with-arduino?redirected=true

skipfire, 2020-10-19:
Any chance that could be as simple as a pair of resistors in series with a bypass jumper on the second?

jgyates, 2020-10-19:
You could probably get most cases for consumer rated generators with that configuration, but there are many variants of CTs. This link has some details on calculating the value and typical burden resistors for a few different types of CTs from a few manufacturers.

https://learn.openenergymonitor.org/electricity-monitoring/ac-power-theory/use-in-north-america#connecting-the-current-transformers

I had to change the resistors on my openenerg monitor board. They used thru hole resistors for that value so it was very easy. I am not sure of all the trade offs of mixing thru hole and SMT so I will leave that up to you.

skipfire, 2020-10-19:
Mixing them isn't a problem, but I was planning PTH anyway.  Tiny manufacturing problems like a bad solder joint are more likely to happen and be an issue with SMT than PTH, especially in vibration heavy environments.  At least that was the case back when I was in Electrical Engineering in college back in a different millennium.  Of course we didn't have access to switching regulators back then.

skipfire, 2020-10-19:
@jgyates I've got what I think is a good solution.  A pair of empty holes so any resistor can be added without having to try to remove a jumper, then a 3-pin bypass for jumper where either 1-2 (standard) or 2-3 (custom) is bypassed.

jgyates, 2020-10-19:
that sounds perfect

skipfire, 2020-11-03:
So testing came back very mixed.  Multiple Pis were tried and they all stayed powered and I even had the official touchscreen being powered by the circuit, however they all reported low voltage the whole time even with an scope showing constantly 4.91-4.97v.  I also saw nothing more than small magnetic induced ripples at startup or when running indicating it is a well regulated 12v output (was 12.08-12.12v).  Unless I figure out why the low voltage is being reported I probably won't go forward with this.  I tried several circuit variations, but never fixed it.  With a standalone 12vdc power supply it works exactly as expected with no low-voltage errors.

jgyates, 2020-11-03:
You may have this info already, but this thread has what appears to be some info related to creating a power supply from a 5V source.

https://www.raspberrypi.org/forums/viewtopic.php?t=102015

jgyates, 2020-11-13:
I am going to close this issue for now but feel free to continue the discussion if anyone is interested in keeping this alive. I can always re-open if there is more interest.

closed 2020-11-13 by jgyates

lwbeam, 2020-12-17:
Just in case you decide to have another run at this, may I suggest the following. Consider adding a small, surface-mounted, normally-open pushbutton to the circuit board, that connects GPIO pin 3 to ground. This and the addition of the following firmware overlay to the _/boot/config.txt_ file, would allow the RPi to be safely shutdown/restarted in the field, without having to have access to a phone, tablet, or computer.

`dtoverlay=gpio-shutdown,gpio_pin=3,active_low=1,gpio_pull=up,debounce=100`

For more information on this, see the [RPi firmware documentation](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L913).

jgyates, 2020-12-18:
that is a good idea

lmamakos, 2020-12-18:
On my board, I've already brought GPIO2 and GPIO3 out to a 4-pin header (J9
near the top left), along with Ground and +3.3V.  The intention was to use
this for external I2C peripherals if desired.  GPIO3 is pin 4 on that
header, intended for the I2C SCL signal.  It should be easy to wire-up a
button between the ground signal on pin 1 and SCL/GPIO3 on pin 4 if
desired.   If you won't want the pull-ups for some reason, you can cut a
couple of traces on the back of the board to disconnect each pull-up
resistor independently.

There's gerber files for it in the repo
https://github.com/lmamakos/RPi-pHat-Thermocouple if you want to spin one.
You could choose to just not populate any of the thermocouple stuff which
turns out to be more difficult to use in practice since the thermcouples
need to be electrically isolated and most of the inexpensive ones available
are not.

[image: Front]
<https://github.com/lmamakos/RPi-pHat-Thermocouple/blob/genmon/meta/Rev3.1%20Front.jpg>

smeisner, 2021-01-03:
@Louis, have you ever priced up having JLCPCB create these boards with the
SMT pieces attached? I am in the middle of building a new Genmon RPi and
have been thinking more about your HAT idea. I cannot do surface mount
soldering, but I do know JLC has very good low volume prices.

If not, would you mind if I researched this and maybe coordinated a group
buy? I know their minimum for bare PCBs is 5.

Steve

lmamakos, 2021-01-03:
This is something I investigated, but they didn't carry all the parts (even as "extended") parts.  Plus I know that I made a stupid component choice for the reverse polarity protection diode on the generator side of the isolated RS-232 interface for no good reason because I was looking for an 0603 package when diodes are typically in a different package type..  They also didn't have the MCP1754ST 3.3v regulators in stock, but an more common part in a different footprint is available.  

However, they do have the MAX3221PWR RS-232/TTL chip as well as the ADUM1201ARZ-RLZ digital isolation device as available parts which are probably the most difficult to solder.  And of course, they could assemble the capacitor passives and LEDs as well.

They don't carry the thermocouple interfaces in stock, but they are difficult to use, requiring K-type thermocouples where the junctions are electrically isolated.  Probably more trouble than is worth, but you could leave all those parts off if you just wanted the isolated RS-232 interface.

It's been some months since I've looks, so the component situation might have changed if they've added other parts.   Of course, if you look at my BOM, the specific vendors for the capacitors, resistors and LEDs are, of course, pretty much inconsequential..

Also note that JLCPCB _only_ does SMD assembly, so you'll still be stuck with soldering on the connectors for the Raspberry Pi GPIO header and the power and RS-232 connections.  

I have no problems at all if you want to get boards fabricated or if you want to modify the design.  It's open-source for a reason!

Also, SMD assembly isn't as hard as you think!  If you have a soldering iron with a small enough tip, very small gauge solder and perhaps a flux pen, some good tweezers that you can use to pick up the components, and whatever you need (if anything) to boost your eyesight so you can see the parts.. it's not too bad.  The most problematic component on the board I did is those thermocouple interfaces because they have a "buried" pad on the bottom of the package where you can't reach it.  That pretty much requires using an oven or hot-air rework tool with solder paste to deal with.  But all the other parts have exposed leads (small as they might be) and are suitable for hand soldering.  I now prefer doing SMD assembly; you get to skip the part of trimming off the component leads and the assembly can actually go faster.

alabamatoy, 2021-01-03:
I know nothing about such hardware design and construction, but I certainly understand the software and RPi side, and would really like to have such a board to use with my genset.  I would especially like to see lmamakos board design revised to circumvent the isolated thermocouple issue.  A few tcouple connections for simple cheap type tcouples would be wonderful.  I will support a group buy etc, but can't offer much in the way of assistance except as cheerleader/support and perhaps testing by a limited-knowledge user.

lmamakos, 2021-01-03:
@alabamatoy there is no simple fix to address the thermocouple isolation issue.  It's just a property of the thermocouple interface device that's being used.  You can find electrically isolated thermocouples, but they are stupid expensive as compared to the cheap type-K thermocouples you can buy on Amazon or eBay.

I think I'd investigate using higher temperature thermistors, like used in grill thermometers or in you electric stove.  You could use a cheap multichannel A/D converter and 12 bits of resolution would probably be plenty.

I think the most value comes from the isolated RS-232 interface; you could take my board, as-is and just leave off all the thermocouple related components and populate the rest.  If I started over, I'd make some different component choices that JLCPCB has in their parts catalog for doing assembly.  Some of those parts, while functionally equivalent, may be available only in a different package, which would require a different PCB footprint and re-spinning the board.  It took me 2 tries before I got it right, with a layout problem on my first version.

griffths, 2021-11-14:
@skipfire ,

This is unorthodox but until you release your updated HAT I thought I would give this a shot but am having no luck.

I have enabled the "External Temperature Sensor" app from the add-on page and named my only sensor "Battery" but I don't have the file structure referenced in the wiki (/sys/bus/w1/devices...) so I am guessing that the sensor isn't being detected but not sure if it is a power or data issue.

Can you comment (constructively please :-) ) on my modification and whether the +5V header is an appropriate place to power the sensor from.

I seem to recall you inquiring in a recent thread about which GPIO pins to use but couldn't find it again.

Based on a search on the web GPIO 4 is what other have used for these sensors but do I need to enable it or modify a .conf file to tell the Pi where to look for the sensors?

Thanks 

Jay
[image]

griffths, 2021-11-14:
[image]

skipfire, 2021-11-14:
@griffths pin 4 is the 1-wire pin which is where many temperature sensors do expect to be connected.  As for the 5v, that will depend on the sensor.  Some sensors are 5v, others 3.3v, and some are tolerant for a range.  The add-on I don't know about yet.  I hope to receive my prototypes for the accessory board this week.

griffths, 2021-11-14:
@skipfire 
So the fan header is live by default on your HAT and is capable of supplying sufficient power?

I am currently powering the Pi/HAT combination with the USB power port although I use the battery terminals with your custom cable when in the generator if that makes any difference 

The data sheet for the sensors I purchased indicate 3.3-5.25V range so I am good there.

Would you please take a look at my work and confirm that I have connected everything correctly.

[image]

Thanks

skipfire, 2021-11-15:
Sensors tend to be extremely low draw so I would not expect any problems, and the fan pins are connected to 5v and Gnd pins on the Pi so as long as you are not getting undervoltage warnings from Raspbian I would say you are fine.

As for the wiring, it looks right to me, and the resistor and capacitor there look like what is used by the DS18B20, which matches what the accessory HAT will use.

Be careful with the wires running around the 3-pin header to be sure you don't pull it over and short anything.  You can clip those 3 pins, use the 5v and Gnd on it, etc.

griffths, 2021-11-15:
Thanks for checking my work.

No under voltage alarm.

Time to swap parts and see if I am a victim of a faulty part somewhere.

skipfire, 2021-11-15:
@griffths have you tried accessing it without GenMon to see if the Pi is recognizing it?  Perhaps GenMon doesn't read the thermometer if there is no data from the generator?

griffths, 2021-11-15:
Not sure how I would do that.

I am not very experienced with Linux / Pi stuff so am learning/teaching myself as I go.

I did try the following command I found from something related to the HAT project by @lmamakos but it didn't work.  I suspect the command _w1thermsensor_ refers to a script that isn't on my Pi but that is only speculation since I am not real familiar with Linux / Pi stuff

w1thermsensor all --type DS18S20

skipfire, 2021-11-15:
Here's an article for it: https://bigl.es/ds18b20-temperature-sensor-with-python-raspberry-pi/.

griffths, 2021-11-15:
Thanks!

Will read up and see how far I get

griffths, 2021-11-15:
@skipfire 

Success!

Thank you for the link.

Followed the instructions to install and enable the 1wire stuff via the GUI

and now I have external temp added to my Genmon screen.

Satisfying to have gotten it working but it would have taken me forever, if ever, to figure this out without your assistance.

[image]

Jay