jgyates/genmon #456: Upgrade from 1.14.27 to 1.15.05 web server doesn't auto start

danorum, 2020-10-19:
I just upgraded from 1.14.27 to 1.15.05 and I genserv no longer starts automatically.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

I would expect the web server to start

### Actual Behavior

genmon starts automatically, but the web server does not

### Steps to Reproduce (including precondition)

I am able to start it manually using ./stargenmon.sh start, but after a reboot it doesn't start automatically.
I checked the crontab and even changed the sleep from 30 to 45 in case it needed more time.

I see it running in the process list, but connection is refused:
root       695 34.8  4.5 124272 20184 ?        Sl   11:57   0:38 /usr/bin/python /home/pi/genmon/genmon.py -c /etc/genmon/
root       721  2.9  3.5  28728 15580 ?        Sl   11:57   0:03 /usr/bin/python /home/pi/genmon/genserv.py -c /etc/genmon/
root       722  1.3  2.3  32724 10480 ?        Sl   11:57   0:01 /usr/bin/python /home/pi/genmon/genemail2sms.py -c /etc/genmon/

when I look at the listening ports, 8000 isn't listed:
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN      382/sshd
tcp        0      0 0.0.0.0:9082            0.0.0.0:*               LISTEN      695/python
tcp6       0      0 [::]:ssh                [::]:*                  LISTEN      382/sshd
udp        0      0 0.0.0.0:mdns            0.0.0.0:*                           209/avahi-daemon: r
udp        0      0 0.0.0.0:39460           0.0.0.0:*                           209/avahi-daemon: r
udp        0      0 0.0.0.0:bootpc          0.0.0.0:*                           345/dhcpcd
udp6       0      0 [::]:mdns               [::]:*                              209/avahi-daemon: r
udp6       0      0 [::]:dhcpv6-client      [::]:*                              345/dhcpcd
udp6       0      0 [::]:47782              [::]:*                              209/avahi-daemon: r

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: 1.15.05

danorum, 2020-10-19:
Also, once I start it manually, it looks like 2 copies of genserv are running -

root       695 31.7  4.5 148848 20276 ?        Sl   11:57   1:42 /usr/bin/python /home/pi/genmon/genmon.py -c /etc/genmon/
root       721  1.3  3.5  28728 15580 ?        Sl   11:57   0:04 /usr/bin/python /home/pi/genmon/genserv.py -c /etc/genmon/
root       722  0.7  2.3  32724 10480 ?        Sl   11:57   0:02 /usr/bin/python /home/pi/genmon/genemail2sms.py -c /etc/genmon/
root      1925  8.5  3.6  29248 16080 pts/0    Sl   12:02   0:03 /usr/bin/python /home/pi/genmon/genserv.py -c /etc/genmon/
root      1938  3.6  2.5  33164 11312 pts/0    Sl   12:02   0:01 /usr/bin/python /home/pi/genmon/genemail2sms.py -c /etc/genmon/

And then I can see port 8000 listening:
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN      382/sshd
tcp        0      0 0.0.0.0:9082            0.0.0.0:*               LISTEN      695/python
tcp        0      0 0.0.0.0:8000            0.0.0.0:*               LISTEN      1925/python
tcp6       0      0 [::]:ssh                [::]:*                  LISTEN      382/sshd
udp        0      0 0.0.0.0:mdns            0.0.0.0:*                           209/avahi-daemon: r
udp        0      0 0.0.0.0:39460           0.0.0.0:*                           209/avahi-daemon: r
udp        0      0 0.0.0.0:bootpc          0.0.0.0:*                           345/dhcpcd
udp6       0      0 [::]:mdns               [::]:*                              209/avahi-daemon: r
udp6       0      0 [::]:dhcpv6-client      [::]:*                              345/dhcpcd
udp6       0      0 [::]:47782              [::]:*                              209/avahi-daemon: r

jgyates, 2020-10-19:
instead of starting manually like this:

     ./stargenmon.sh start

Do this:

     ./stargenmon.sh restart

The first commands just blindly loads without first unlading any loaded version. Explicitly loading two copies will result in two copies being loaded. genmon.py will check that it is not loaded twice, but genserv.py and add on programs will load a second instance if you do not unload it first.

Your logs look like they are showing that genserv (the web server) is loading OK.  Your logs also indicate that you are not using secure HTTP so the URL should be 

     http://youripaddress:8000

Is this the url you are using?

jgyates, 2020-10-19:
One thing you could try is this:

from the genmon directory type:

        ./startgenmon.sh stop
        ./genmonmaint.sh -C
        ./genmonmaint.sh -u -n 
        ./startgenmon.sh restart 

Upper case "C" option will clean the *.pyc files in the genmonlib directory

The -u -n option will download the latest again. If that timed out or was interrupted it could cause an issue.

danorum, 2020-10-19:
All of those commands seemed to execute successfully, but after reboot, my genserv doesn't appear to be listening on 8000. Even after doing a ./startgenmon.sh restart doesn't trigger listening on 8000 even though genserv is running.

If I do a ./startgenmon.sh start, then I get listening on 8000

This is very odd and not an issue I have ever had with genmon.

jgyates, 2020-10-19:
There are few ways we can go to debug from here:

1) Validate this is not specific to one computer , anti-malware software,  browser or a plugin on a broswers keepting the connection from occurring

2) If you have telnet installed on a system on your network you can type:
      
      telnet ipaddress 8000

This will attempt to connect to ipaddress on port 8000. If the pi is listening it should connect. This will validate that your network is not blocking things. I have seen that once on another system. i.e. overly aggressive router attempting to block any ports that are non standard.

3) You can use your browser's developer console (on chrome it is settings -> More Tools > Developer Tools. From here you can view network traffic your browser sends out
4) Load things manually (without the script or genloader) and see if there is any additional info displayed by the Flask libraries to the console. To do this:

       a) unload genmon, from the genmon directory type:    ./startgenmon.sh stop
       b)  sudo python genmon.py &
       c)  sudo python genserv.py &

If you open a browser you should see browser requests on the console like this:

      OK - Init complete.
      * Serving Flask app "genserv" (lazy loading)
      * Environment: production
        WARNING: Do not use the development server in a production environment.
        Use a production WSGI server instead.
      * Debug mode: off
      * Running on http://0.0.0.0:8000/ (Press CTRL+C to quit)
     192.168.11.179 - - [19/Oct/2020 16:15:51] "GET / HTTP/1.1" 200 -
     192.168.11.179 - - [19/Oct/2020 16:15:51] "GET /libraries.min.css HTTP/1.1" 200 -
     192.168.11.179 - - [19/Oct/2020 16:15:51] "GET /libraries.min.js HTTP/1.1" 200 -
     192.168.11.179 - - [19/Oct/2020 16:15:51] "GET /cmd/start_info_json HTTP/1.1" 200 -
     192.168.11.179 - - [19/Oct/2020 16:15:51] "GET /cmd/getfavicon HTTP/1.1" 200 -

To kill the processes we started above, the startgenmon.sh script will not work. Instead use this:

     sudo pkill -9 -u root -f genserv.py
     sudo pkill -9 -u root -f genmon.py

danorum, 2020-10-19:
I can tell for sure it isn't listening : 
```
pi@genmon:~/genmon $ sudo netstat -ltup
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN      408/sshd
tcp        0      0 0.0.0.0:9082            0.0.0.0:*               LISTEN      1464/python
tcp6       0      0 [::]:ssh                [::]:*                  LISTEN      408/sshd
udp        0      0 0.0.0.0:mdns            0.0.0.0:*                           233/avahi-daemon: r
udp        0      0 0.0.0.0:bootpc          0.0.0.0:*                           348/dhcpcd
udp        0      0 0.0.0.0:45652           0.0.0.0:*                           233/avahi-daemon: r
udp6       0      0 [::]:mdns               [::]:*                              233/avahi-daemon: r
udp6       0      0 [::]:dhcpv6-client      [::]:*                              348/dhcpcd
udp6       0      0 [::]:42380              [::]:*                              233/avahi-daemon: r
```

If I then run ./startgenmon.sh start, after a little bit, it is then listening. 
```
pi@genmon:~/genmon $ sudo netstat -ltup
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN      408/sshd
tcp        0      0 0.0.0.0:9082            0.0.0.0:*               LISTEN      27153/python
tcp        0      0 0.0.0.0:8000            0.0.0.0:*               LISTEN      27967/python
tcp6       0      0 [::]:ssh                [::]:*                  LISTEN      408/sshd
udp        0      0 0.0.0.0:mdns            0.0.0.0:*                           233/avahi-daemon: r
udp        0      0 0.0.0.0:bootpc          0.0.0.0:*                           348/dhcpcd
udp        0      0 0.0.0.0:45652           0.0.0.0:*                           233/avahi-daemon: r
udp6       0      0 [::]:mdns               [::]:*                              233/avahi-daemon: r
udp6       0      0 [::]:dhcpv6-client      [::]:*                              348/dhcpcd
udp6       0      0 [::]:42380              [::]:*                              233/avahi-daemon: r
```

After a reboot, it is then not listening again, and the ./startgenmon.sh restart doesn't fix it, but a ./startgenmon.sh start does. I also just tried a stop followed by start and that also doesn't seem to get the server started. 

It is odd that I have 2 instances of genserv running in order to get that port 8000 open to listen.
```
pi@genmon:~/genmon $ ps -aux | grep gen
avahi      233  0.2  0.7   6520  3228 ?        Ss   13:37   0:13 avahi-daemon: running [genmon.local]
root     27153 32.0  4.5 148852 20400 pts/0    Sl   15:25   1:29 /usr/bin/python /home/pi/genmon/genmon.py -c /etc/genmon/
root     27166  1.3  3.5  28732 15608 pts/0    Sl   15:25   0:03 /usr/bin/python /home/pi/genmon/genserv.py -c /etc/genmon/
root     27167  0.7  2.3  32728 10396 pts/0    Sl   15:25   0:02 /usr/bin/python /home/pi/genmon/genemail2sms.py -c /etc/genmon/
root     27967  4.4  3.7  63296 16592 pts/0    Sl   15:28   0:06 /usr/bin/python /home/pi/genmon/genserv.py -c /etc/genmon/
root     27968  1.3  2.5  33164 11296 pts/0    Sl   15:28   0:01 /usr/bin/python /home/pi/genmon/genemail2sms.py -c /etc/genmon/
pi       29302  0.0  0.4   4364  2112 pts/0    S+   15:30   0:00 grep --color=auto gen
```

jgyates, 2020-10-19:
Then do number 4 above. The actual listening is done by the Flask libraries so number 4 will give us some more insight

danorum, 2020-10-19:
stopping genmon and running manually works.

```
pi@genmon:~/genmon $ ./startgenmon.sh stop
Stopping genmon python scripts
Stopping....
Stopping genemail2sms.py
Stopping genserv.py
Stopping genmon.py
pi@genmon:~/genmon $ ps -aux | grep gen
avahi      208  0.4  0.7   6516  3144 ?        Ss   15:37   0:00 avahi-daemon: running [genmon.local]
pi         921  0.0  0.4   4364  2108 pts/0    S+   15:39   0:00 grep --color=auto gen
pi@genmon:~/genmon $ sudo python genmon.py &
[1] 1042
pi@genmon:~/genmon $ ps -aux | grep gen
avahi      208  0.2  0.7   6516  3144 ?        Ss   15:37   0:00 avahi-daemon: running [genmon.local]
root      1042  0.5  0.7   7212  3352 pts/0    S    15:39   0:00 sudo python genmon.py
root      1046 81.3  4.4 107636 19688 pts/0    Sl   15:39   0:08 python genmon.py
pi        1105  0.0  0.4   4364  2112 pts/0    S+   15:39   0:00 grep --color=auto gen
pi@genmon:~/genmon $ sudo python genserv.py &
[2] 1203
pi@genmon:~/genmon $ OK : Auto, Off - Ready
 OK - Init complete.
 * Serving Flask app "genserv" (lazy loading)
 * Environment: production
   WARNING: Do not use the development server in a production environment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://0.0.0.0:8000/ (Press CTRL+C to quit)
192.168.0.201 - - [19/Oct/2020 15:40:26] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:27] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:27] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:29] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:30] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:30] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:31] "GET / HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:31] "GET /libraries.min.js HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:31] "GET /libraries.min.css HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:31] "GET /cmd/getfavicon HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:31] "GET /cmd/start_info_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:31] "GET /cmd/getreglabels HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:31] "GET /favicon.ico HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:31] "GET /images/transparent.png HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:31] "GET /images/sprites.png HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:32] "GET /images/ui-icons_222222_256x240.png HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:32] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:32] "GET /cmd/power_log_json?power_log_json:%2010000 HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:32] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:32] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:35] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:36] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:36] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:37] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:41] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:41] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:41] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:44] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:47] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:47] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:47] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:50] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:51] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:51] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.201 - - [19/Oct/2020 15:40:53] "GET /cmd/gui_status_json HTTP/1.1" 200 -
sudo pkill -9 -u root -f genserv.py
[2]+  Killed                  sudo python genserv.py
Killed
pi@genmon:~/genmon $  sudo pkill -9 -u root -f genmon.py
[1]+  Killed                  sudo python genmon.py
Killed
pi@genmon:~/genmon $
```

After this, if I do ./startgenmon.sh start, I do not have genserv listening on 8000. I just have genmon listening on 9082.

jgyates, 2020-10-19:
That is quite strange. The only difference between loading manually like you just did and loading with the startgenmon.sh script is that the script calls genloader.py which loads the modules based on /etc/genmon/genloader.conf. If a module is enabled it just loads "sudo python genmon.py" etc in the background. The only real difference is the way the standard out, and standard error streams are handled.

Are you running raspbian, if so what version. If not what OS and version are you running? It is almost as if something is not right with stream redirection / handling and flask is choking because it can not output the messages it normally does. genloader redirects the output to null.

try this:

      sudo python genmon.py &
      sudo python genmon.py > /dev/null 2>&1 &

That should be very close to what genloader is doing to genserv.py.

danorum, 2020-10-19:
Yes Raspbian. Things have been humming a long for a long time, and this is the first update to do this. Before I got your last reply, I tried running

`sudo python genloader.py -s &`

I had to run it twice before genserv was listening on 8000. I see it running in the process list, but not listening.

Should that second line be genserv.py?

```
  sudo python genmon.py &
  sudo python genserv.py > /dev/null 2>&1 &
```

danorum, 2020-10-19:
I had to run 

`sudo python genserv.py > /dev/null 2>&1 &`

twice in order for it to be listening on 8000. I can see it running in the process list, but can't connect to it (and appears to not be listening).

danorum, 2020-10-19:
I wonder if that is it.. If there is no delay between running genmon and genserv, something in genserv is failing.

jgyates, 2020-10-19:
/etc/genmon/genloader.conf has a setting named "postloaddelay". this is the number of seconds to delay before loading another module. You could try to make this value 10 or 20 as a test and then load with the startgenmon.sh script.

in the [genmon] section of /etc/genmon/genloader.conf make postloaddelay 10 or 20. The default is 4

danorum, 2020-10-19:
Setting it to 10 fixed it for me. Sorry about all the back and forth troubleshooting. Luckily, I tried pasting in multiple commands or I may have never found this. 

This is a raspberry pi zero that I have been running for a long time, so the only thing I can think of is something got "bigger" and genmon was just slower to start now than it was with the last successful version.

I did just reboot a second time and the web came back up after the delay.

Thanks again for the help!

jgyates, 2020-10-19:
Great! I am glad it is working now.

closed 2020-10-19 by jgyates