jgyates/genmon #457: When i reboot my wifi, i always lose the genmon and have to reboot it.

grsthegreat, 2020-10-20:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}
This is just a simple inquiry. I added a ups supply to prevent wifi restart upon power failure, but every so often i have to reboot the wifi system to get rid of accumulated glitches that occures over time. These glitches just seem to add up, and sooner or later a wifi reboot is used to clear everything up. Every other device in the house handles reboot just fine, but the genmon unit always gets lost and i have to reboot the genmon unit itself to reconnect.

Is this just standard behavior with the raspberry pi units in general?

swvalenti194, 2020-10-20:
https://gist.github.com/mharizanov/5325450

jgyates, 2020-10-20:
The above link is used to setup a cron job that will run every 5 min and check if wifi is connected. This is needed because linux does not continuously check if wifi is connected and reconnect as needed, but the provided script does do this.

You would use the script in the file and setup a crontab entry to make is run every x number of minutes.

jgyates, 2020-10-20:
I added this one to the tips and trick wiki

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon

grsthegreat, 2020-10-20:
Hmm, im not really sure where to insert this

grsthegreat, 2020-10-20:
Would i just copy and paste the entire thing in putty?

jgyates, 2020-10-20:
I have attached a modified version this that may be a bit better for a pi.

[WiFi_Check.txt](https://github.com/jgyates/genmon/files/5412173/WiFi_Check.txt)

From there you need to transfer the file to your pi and put it in /home/pi. I am not sure how to do this with putty. I use the command line program called scp. From the directory where you saved it type:

      scp ./WiFi_Check.sh pi@ipaddressOfYourPi:./

You will prompted for the password of the pi user. Once you enter it the file will be copied to the /home/pi directory.

Then change the permissions on the file so it is executable. Log into your pi and type:

      chmod 0755 /home/pi/WiFi_Check.sh

From here you can test the file to make sure it works on your system by typing:

       ./WiFi_Check.sh

Then we need to add this to the crontab:

      sudo crontab -e

If prompted to choose and editor choose nano. Then add this line at the bottom of the file:

     */5 * * * * /usr/local/bin/WiFi_Check

Save the file by typing Ctrl+X, then Y then Enter.

grsthegreat, 2020-10-21:
Thank you, ill try that

grsthegreat, 2020-10-21:
well, i think im to stupid to mess with this. i tried fiddeling with putty, and im afraid ill crash the entire thing. i think ill leave good enough alone.

closed 2020-10-21 by grsthegreat

Ariveraq, 2024-03-20:
Sorry to re-open this.  The file you provided has a .txt extension but the file in this command (scp ./WiFi_Check.sh pi@ipaddressOfYourPi:./) has a .sh extension.  Stupid question, How do you convert one to the other?

jgyates, 2024-03-20:
That depends on where you saved the file and what operating system you are using. You typically just rename the file by right clicking on it and changing the name, but you may have to make a settings change on your operating system to show the extensions. Alternatively you could transfer the file to the pi:

     scp ./WiFi_Check.txt unsername_on_pi@ipaddressOfYourPi:./

Then on the pi type

    mv ./WiFi_Check.txt ./WiFi_Check.sh

Ariveraq, 2024-03-23:
Thanks again, scrip successfully installed!!
[image]