jgyates/genmon #459: Gauges on status page

RoyRomeo, 2020-10-27:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

6 gauges on status page

### Actual Behavior

Only 4 gauges showing.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

[image]

{Please write here}

### Your Environment

- Generator Model: G0062440
- Generator Registers: Submit via the About page
- Genmon Version: V1.15.07

jgyates, 2020-10-27:
solely based on the register data you submitted the contents of your registers look normal. I would need to see your log data to know for sure, but your nominal RPM value does not look right. I am not sure why it is. You can save your settings on the about page and then restore your /etc/genmon/genmon.conf. From the genmon directory:

      sudo cp ./conf/genmon.conf /etc/genmon/genmon.conf
      ./startgenmon.sh restart

jgyates, 2020-10-27:
FYI, the nominal RPM is a setting that is a numeric value that is read from the registers and saved to the conf file. For some reason, you value is text  (your name) instead of a number.

jgyates, 2020-10-27:
I got your logs,yes, that is the problem. You can try the above or you can try to set Nominal RPM on the settings page to 3600 and save your settings.

RoyRomeo, 2020-10-27:
The Nominal RPM is already set to 3600.

jgyates, 2020-10-27:
Change it to another value and then back to 3600. You should see a green check mark if you change the value. If you see a green check mark and click "save" then it will write the config file. The web interface is  populating the default, but you have a non numeric value for this setting. Your log is riddled with errors related to this. If this does not work you should replace the file with the default as described above.

RoyRomeo, 2020-10-27:
That worked!  Thanks very much for your help.

jgyates, 2020-10-27:
No problem

closed 2020-10-27 by jgyates