jgyates/genmon #460: JSON Switch State, Engine State, System in Alarm Codes

snakeeyes5, 2020-10-28:
Would you consider adding codes to the status_num_json output?

`        {
          "SwitchCode": 2,
          "Switch State": "System in Alarm"
        },`

This helps with some parsing. Tried to search for it but was unable to find exact possible values for Switch State, Engine State, System in Alarm, could you point me to those?

jgyates, 2020-10-28:
I am not clear on what you are asking. Are you wanting the register values that correspond to the text output? You can get that from registers_json

I am assuming you are asking about Evolution Controllers, this is different for industrial. The lower half of register 0001 contains the value for the switch state and alarm.

GetSwitchState:
https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L2840

GetAlarmState:
https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L2594

If you are using clientInterface to send commands you can also use the command "generator: readregvalue=0001". This will not work on the web interface. This returns a string, not JSON.

snakeeyes5, 2020-10-28:
Thanks! I think I got what I need.

closed 2020-10-28 by snakeeyes5