jgyates/genmon #462: Text messages are no longer being received

mxplea10, 2020-10-30:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Text messages are no longer being sent for outage, error, warning, and info.  I noticed this last week after an update and genmon restarted.  Had an outage last night and did not receive a notification.

### Steps to Reproduce (including precondition)

Verified that the text message address is correct by sending a message from my email address.  I added my email address to the Notification Recipients' list and received an email when genmon was restarted.  I've checked the settings.  What can I do to fix this issue?

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0070432
- Genmon Logs: Submit via the About page
- Genmon Version: V1.15.08

jgyates, 2020-10-30:
I got you logs. 

Sorry, I am am at a disadvantage due to an home internet outage, power outage and my laptop is getting the keyboard replaced. I doubt I will be able to reproduce this until my laptop is back from repair. Nothing related to email has changed recenetly in the genmon code base so I am guessing that either this is some setting that has changed or google (assuming you are using gmail) or some other provider has changed the way they are enforcing some rules.

I do see an error in the log. One question, do you just have one recipient or are there multiple? If so how many. This may be relevant to testing. Emails are working fine for me. One thing to try is to see if you get the same results with one recipient. Also, in a separate test, you can try to make the "Sender Name" field blank.

I am curious if this still occurs if you try both of the above tests.

One thing that *might* correct the problem is to try to update the system software of your pi. This will potentially update python and also update the smtplib python module. This is the python library that is giving the error. This is a long shot but worth a try. To do this:

     1) log into the pi via ssh 
     2) type:
               sudo apt-get update
               sudo apt-get dist-upgrade

You will be prompted to hit 'Y' to update the software if updates are found. One the updates are finished then restart your pi.

          sudo reboot

mxplea10, 2020-10-30:
With all of that going on, I appreciate you even getting back with me.  I added my gmail as test and removed it.  I did several test before making the "Sender Name" BLANK but didn't receive a text.  After making the "Sender Name" BLANK, text messages are being received again.

I updated the Pi this morning.  Thanks for help.

jgyates, 2020-11-03:
I just checked in an update the should fix your issue. To test this:

1) go to the about page and click the button to update the software (the version number has not changed, I will wait until you validate this fixes the issue to change the version number).
2) Once the software has updated, then put the "Sender Name" field back to how you originally had it and test

Let me know the results.
Thanks
Jason

mxplea10, 2020-11-04:
It's working with the "Sender Name" field again.  To test, I restarted the Raspberry Pi.  Is there a better way of testing this without restarting the Raspberry Pi or going to the generator?

jgyates, 2020-11-04:
when you update the software or save your settings the software restarts so you don't have to restart the pi every time.

Thanks, I will close this issue. A new release will go out soon with this change included.

closed 2020-11-04 by jgyates

mxplea10, 2020-11-08:
@jgyates, the text notifications have stopped again.  I removed the "Send Name" but it still doesn't work.  I noticed the problem yesterday.  Can you check this again?

jgyates, 2020-11-08:
I would need to see you log files to look into this as I am not seen any errors on my end or reported by others.

mxplea10, 2020-11-08:
Ok, logs sent.

jgyates, 2020-11-08:
I received your logs. I do not see any errors reporting in the log. Can you manually send from your gmail account to the addresses in your notification? Also can you send a test email to the recipients on the settings page. You can send a test email to multiple recipients by entering them, separated by commas.

mxplea10, 2020-11-08:
Yes, I can manually send to the recipient address from my gmail account.  But, I do not receive a message when sending to the recipient address on the settings page.  However, when a message is sent to my gmail account, the message is received.

jgyates, 2020-11-09:
What are the ending domain names of the email addresses you are using to send to and from (i.e. @gmail.com, @yahoo.com)

since I can not duplicate this, the difference must be in something besides gmail as I can send and receive with gmail OK.

mxplea10, 2020-11-09:
Settings > Optional - Outbound Email Settings > Email Account (@gmail.com)

Notifications > Notification Recepients > Email Address (@mms.att.net)

swvalenti194, 2020-11-09:
isnt it phone number @mms.att.net

mxplea10, 2020-11-09:
Yes, the phone number is before the domain name.

zekyl314, 2020-11-09:
I know it doesn't help, but I gave up on AT&T and the email to MMS gateway.  I always had hit or miss luck with it.  I even opened a case with AT&T support for my work account (Unrelated to GenMon personal software usage) as we had delivery issues with the AT&T mms gateway there too.  

I ended up moving to an MMS service from Twilio.  It 100% always works for me since I switched to that service.

jgyates, 2020-11-09:
@mxplea10 

I tested this with a sender gmail and a reciever gmail and @mms.att.net and it worked for me, although it was slow and often messages were received out of order. I did check in one minor tweak to the mail support, but my testing showed it worked both before and after the fix for me. You can checkout the latest and see if that fixes it, but my guess is that this is just some strange problem with ATT email to txt service.

mxplea10, 2020-11-10:
Notifications are now being received after the update.  I've gotten 5 notifications after the update for some reason.  @zekyl314, thanks for the recommendation.  I'll try Twilio if I keep having trouble with AT&T.  Thank both you for your help.