jgyates/genmon #463: TX Series ATS and H-100 Controller

unsword01, 2020-11-03:
At one facility we have an HTS Series ATS and Genset with H-100 Controller, all nicely monitored using Genmon thanks to great work from @jgyates to implement the HTS support.

At another facility we are planning to install a new Genset with H-100 controller but advised by the dealer that the HTS series ATS is no longer being sold and that the replacement is the TX Series. But digging into the documentation for the TX series it states:

*Note: This control is not currently communication compatible with Generac Gensets*

I'm struggling to understand on what planet that makes any sense ! - The ATS to Genset start is back to traditional 2-wire method.

The TX ATS does have Modbus capability and thus could/should be a slave (since it supports Modbus write) and has a default Address of 240. However looking at the Modbus maps for the HTS and the TX they use very different register numbers, so I'm suspecting the H-100 can't poll the TX (unless maybe the H-100 gets new firmware to understand the TX series registers).

So I have  two questions:

1. Anybody got experience of the TX series (specifically a TX 611) with an H-100 controller ?
2. If they truly don't talk nicely could Genmon poll both devices separately and consolidate the data, instead of reading the ATS data via the H-100 ?

jgyates, 2020-11-03:
I don't have any experience with the TX switch. In theory, yes they could both be polled, but the software does not currently support multiple slave devices on a single genmon instance. Before I made any changes we would want to validate the TX series can communicate with us over modbus by using the modbusdump,py program in this projects source repository. modbusdump can probe serial or serial over TCP devices for modbus values.

The other issue is that I don't have any register definitions for the TX switch. If you have any you can send them to me at generatormonitor.software@gmail.com

unsword01, 2020-11-03:
Thx for the reply - I figured it would be a fairly significant architectural change. Making some more inquiries on the Generac side right now.

unsword01, 2020-11-03:
The 'Official Word':

The H-100 controller & TX Controller are independent devices of which both would have to be integrated into a building management system independently over the RS 485 communication port

bbrown-sc, 2020-11-05:
@unsword01 Sorry to hijack your thread here, but as a HTS & H-100 owner I was wondering if you could answer a question for me. I have an industrial genset (QT025A) with a H-100 control panel and a single-phase 400A HTS transfer switch. I have the HTS "Instrument Package" that includes current transformers (WICC Ltd. A-600-01-L36) on each leg of the split-phase that's suppose to measure the current from the utility. Since the initial installation in 2008 GenLink-DCP and now Genmon both read erroneous values for utility current & power. The user manual says to orient one CT in one direction and the other CT in the opposite direction. This seems wrong. Wouldn't the CTs just cancel each other out? If you have a single (split-phase) HTS can you verify the correct orientation for the CTs? Thanks.

[HTS- instrument package.pdf](https://github.com/jgyates/genmon/files/5491353/HTS-.instrument.package.pdf)

jgyates, 2020-11-13:
@unsword01 

I have added this to the feature list:

https://github.com/jgyates/genmon/blob/master/feature_requests.md

I am going to close this issue for now. When you are ready to test you can repost to this thread.

closed 2020-11-13 by jgyates