jgyates/genmon #464: SNMP values sennding data that cannot be prosesed by NMS (librenms)

typecookie, 2020-11-05:
I am attempting to integrate genmon with Librnms. Librenms graphs values and so expects numbers. ie  snmp .1.3.6.1.4.1.9999.0.1.0 reports 243 v and i need an output of 243.  I have been looking a the code but am at best a scriptmonky. 
Is there a way to strip out the following letter in the snmp output or to not include this in the web output? I am more then happy to spent the time but wold appreciate a pointer in the right direction (the flow from the hex from the generator through the web out put to snmp output).  
### Expected Behavior

 snmpwalk -v2c -c WyoCore 10.15.255.27 .1.3.6.1.4.1.9999.0.0.0.4
iso.3.6.1.4.1.9999.0.0.0.4 = STRING: "12.60"

### Actual Behavior

 snmpwalk -v2c -c WyoCore 10.15.255.27 .1.3.6.1.4.1.9999.0.0.0.4
iso.3.6.1.4.1.9999.0.0.0.4 = STRING: "12.60 V"

- Genmon Version: V1.15.08

jgyates, 2020-11-05:
I did a similar thing for the MQTT add on. I should be able to add this soon. I will let you know once it is complete.

typecookie, 2020-11-05:
thank you for such a quick response

as I am sure you know this example, the voltage, will of corse apply to any data sent involving a number and hopefully anything that could be represented in binary. ie on/off states or ok/error states

jgyates, 2020-11-05:
The only thing that my implementation would do is remove the units of numeric values. It would not convert on/off states to binary. That would require a more significant change. If that is not sufficient let me know.

typecookie, 2020-11-06:
Sensor states in Librenms as well as several other NMS platforms use nagios methods.
https://docs.librenms.org/Developing/Sensor-State-Support/
This I understand is a large change but would make this service more viable in larger networks.

```
states:
                        - { value: -1, generic: 3, graph: 0, descr: 'null' }
                        - { value:  0, generic: 0, graph: 0, descr: open }
                        - { value:  1, generic: 2, graph: 0, descr: closed }
```
and 
```
 states:
                        - { value: 0, generic: 0, graph: 0, descr: normal }
                        - { value: 1, generic: 1, graph: 0, descr: info }
                        - { value: 2, generic: 1, graph: 0, descr: warning }
                        - { value: 3, generic: 2, graph: 0, descr: error }
                        - { value: 4, generic: 2, graph: 0, descr: critical }
                        - { value: 5, generic: 2, graph: 0, descr: failure }
```

jgyates, 2020-11-06:
Nagios has a robust method for parsing content. I use Nagios on my home network and it currently alerts me if the SNMP fuel sensor on my diesel generator is low.

This page has info on using the nagios plugins with librenms:

https://raymii.org/s/tutorials/Service_checks_in_LibreNMS_nagios_plugins.html

The main plugin you would use would be check_snmp. 

https://www.monitoring-plugins.org/doc/man/check_snmp.html

In my nagios setup I use this command:

define command{
 command_name check_gen_fuel
 command_line $USER1$/check_snmp -H $HOSTADDRESS$ -P 1 -C $USER3$ -o .1.3.6.1.4.1.9999.0.1.0.7 -r '([2][5-9]|[3-9][0-9]) %'
}

This calls the function:

      /usr/lib/nagios/plugins/check_snmp -H 192.168.11.14 -P1 -C public -o .1.3.6.1.4.1.9999.0.1.0.7 -r '([2][5-9]|[3-9][0-9]) %'

And it returns this value:

      SNMP OK - "93 %" | 

This translates to a SNMP request to OID 1.3.6.1.4.1.9999.0.1.0.7 to IP address 192.168.11.14  using SNMP V1 and community string of public. The result should be formatted (using regular expressions) with a percent sign and a range of 25-99.

The numeric return value follows the standard nagios format:

0 - OK
1 - Warning 
2 - Critical
3 - Unknown

The use of regular expressions allows you to parse the output in many ways.

For another example this would check that the switch state is in auto:

     /usr/lib/nagios/plugins/check_snmp -H 192.168.11.14 -P1 -C public -o .1.3.6.1.4.1.9999.0.0.0.0 -R 'Auto'

you can also set warning and critical levels on the command line when dealing with numeric values line voltages.

The following will return OK if the battery voltage falls between 12.00 and 14.90

     /usr/lib/nagios/plugins/check_snmp -H 192.168.11.14 -P1 -C public -o .1.3.6.1.4.1.9999.0.0.0.4 '1[2-4].[0-9][0-0] V'

typecookie, 2020-11-06:
Using nagios plugins is an option with librenms although I have had mixed luck using the alerts built in to Librenms pulling from the plugin.  I can pull all the data and create alerts with only the letters removed from the snmp reply.  the binary was only a thought.

Librenms is a bit picky without building a full yaml file and a mib imported. I have found the easy way of adding custom devises is with the add custom oid option.
[image]

This helps eliminate the need of creating a yaml file and mib file. It seems to be the most accessible method for adding custom devises although it is picky about only getting numbers.

Just my thoughts if I am more then willing to help in any way I can including beer money and testing.
Thank you for your time.

jgyates, 2020-11-07:
I have added the ability to remove the units from the numeric values as an option in the gensnmp add on program. I will have to think on adding the additional items you are asking for. Presently I don't see the benefit will out weight the maintenance cost since this would cover an extremely large subset of data. This project currently supports 6 different residential generators and two different industrial generators. Defining unique numbers for all state and error conditions is quite a lot of data. Using regular expressions within librenms would likely allow you to accomplish your task. having said that I will think on this for a few days and let you know. I may try to load up librenms and see if I can do some testing.

typecookie, 2020-11-09:
I should have posted this on the 6th but,l the changes you made have allowed me to implement the efficiently in our network and an now push it to 4 other sites and possibly more soon. Thank you very much.

jgyates, 2020-11-13:
@typecookie I am glad this is working for you. I am going to close this issue now.

closed 2020-11-13 by jgyates