jgyates/genmon #465: Please allow geolocation access for this to work.

RoyRomeo, 2020-11-06:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Display current weather (at bottom of setup page)

### Actual Behavior

When I click on Look Up I get the message Please allow geolocation access for this to work.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: G0062440
- Genmon Logs: Submit via the About page
- Genmon Version: V1.15.09

jgyates, 2020-11-06:
Your browser is denying the web app to use your location to automatically set the location for the weather api.

In chrome use this URL to change this setting:

     chrome://settings/content/location?search=location

[image]

Your could have this setting on a site specific setting. You can change that here but you need to search for the site address (192.168.1.170):

      chrome://settings/content/all?search=location

RoyRomeo, 2020-11-06:
When I go to "chrome://settings/content/all?search=location", I do not see "192.168.1.170" anywhere in the list.

jgyates, 2020-11-06:
Your browser can prohibit geolocation across all sites *or* on specific sites. The first link is for all sites, the second one is for specific sites. What is the setting on the first link?

RoyRomeo, 2020-11-06:
Ask before accessing.

jgyates, 2020-11-06:
On this page:

    chrome://settings/content/all?search=location

in the search box at the top right type:

        192

This will show all sites that have 192 in the address. You should see a site listed. Click the three dots on the right of the entry then click "Reset Permissions". Then refresh your browser. 

If you pi IP address has ever changed then it may be a different IP address grouped into one entry. What has happened is that you were asked by your browser and then "deny" was clicked and your browser saved the settings.

RoyRomeo, 2020-11-06:
When I type in "192", I do not see "192,168.1.170".  I don't think my IP address ever changed.

jgyates, 2020-11-06:
I looked into this further, google will not allow geolocation to work unless you are using https with a valid certificate. What are your settings for secure web access on the genmon settings page.

RoyRomeo, 2020-11-06:
[image]

jgyates, 2020-11-06:
There are two problems. 

1) Your connection is not secure so the browsers built in geolocation function is failing for security reasons 
2) The genmon web app is allowing the Lookup button to occur when you don't have a secure connection.

I checked in an update that should correct #2. You can update on the about page, I did not change the version number from yesterday's update. 

You will have to enter your city manually. You can use this link:

     https://openweathermap.org/find?q=

In most cases you can enter your city in this format:

       City, State, Country

For example:

       Kalamazoo, MI, USA

I am going to close this issue but feel free to post here if you have any questions.

closed 2020-11-06 by jgyates

RoyRomeo, 2020-11-06:
Okay, thanks very much for your help.

RoyRomeo, 2020-11-06:
So in the "location to report weather" do I put "39.8948,-80.1617"?  And where does the weather information display?

jgyates, 2020-11-06:
That number is the longitude and latitude of the lookup. That would be correct for Morrisville PA. Or you could just put 

       Morrisville, PA, USA

in the box or your could put the number 4480285 as this is the City ID of Morrisville PA.

RoyRomeo, 2020-11-06:
Where should the weather information display?

jgyates, 2020-11-06:
on the monitor page of the web app