jgyates/genmon #466: Disconnected from Server" after Addon

curtis1757, 2020-11-06:
This is a minor issue, but something that may have an easy fix.

When "Optional - Webserver Security Settings" is ON and then you either turn ON or OFF an Add-On (I tested with 'External Temperature Sensors' and some others) then Save the new setting, then I get the 'Disconnected from Server' message and it will not reconnect.  However,  with "Optional - Webserver Security Settings" is OFF, it will also disconnect, but after about 10 seconds will reconnect and all is fine.  With "Optional - Webserver Security Settings" is ON and I do this, I can just refresh the webpage and log in again and all is good.  So I'd guess that since there maybe some internal webpage refresh being done and it does not know the user name and password so it can not reconnect and the 'Disconnected from Server' message goes on forever.

Like I said very minor issue and easy to work around, just more of an FYI.

Curtis

jgyates, 2020-11-06:
Thanks for pointing this out. This is related to some overlapping logic in the web app. If you refresh your browser you should be prompted to login again. Once settings are saved the web app restarts genmon and genserv (the web server) so the session info expires (your login info goes away) and you are forced to login again. I will put this on the list to clean this up. Thanks

curtis1757, 2020-11-06:
Yup, that is what I've beendoing and all good....  Thanks

jgyates, 2020-11-06:
I checked in an addendum to V1.15.09 that should fix this issue. Thanks for letting me know and let me know if you have any issues.

closed 2020-11-06 by jgyates