jgyates/genmon #467: Power output not accurate

jjmacneil, 2020-11-07:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

The power output status should display actual output

### Actual Behavior

The power output status was showing 0.11kW and typically the average has been between 6kW - 7kW during power outages.

### Steps to Reproduce (including precondition)

We experienced a power outage recently and when viewing the power output status I noticed the odd value 0.11kW

### Screenshot or Pictures relating to the problem (if possible)

See attached
[image]

### Your Environment

- Generator Model: Generac 22kW
- Genmon Logs: Submit via the About page
- Genmon Version: 1.15.08

jgyates, 2020-11-07:
this is known issue with 22kw systems. See number 5 on this page:

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

Also see:

https://github.com/jgyates/genmon/issues/409

You can add your own Current Transformer to the 22kW model. Some folks have done that in this thread:

https://github.com/jgyates/genmon/issues/403

I am closing this issue as it has been covered but feel free to post to this thread if you have questions.

closed 2020-11-07 by jgyates

jjmacneil, 2020-11-07:
Thank you Jason. I'm a bit confused as the status was accurate previously in prior versions so what changed to alter the output or is now causing the inaccurate output number?

jgyates, 2020-11-07:
Previously the software read one set of registers (register 003a) for what was assumed to be the current transformer (CT) output. This was later changed to read registers 05f4 and 05f5 and has been proven to be the accurate CT reading. I am assuming the 03a value is a throttle value of some type as it does appear to be related. If you want to revert to using register 003a as the source of the current output your can go to the advanced page in the web interface by clicking the icon in the upper right twice, then enable the "Use Legacy Power Calculation" option. This will revert to the old method. The conversion between the throttle value and the power was performed empirically so it works because these testing make it work out to be close enough. If you want it to be more accurate you can add two CTs and connect them to your controller, then the new method will work. Let me know if you have any other questions.

jjmacneil, 2020-11-07:
Thanks Jason. I enabled the legacy option for now which will hopefully provide a more accurate reading until I have a chance to install the CT’s. Is there a diagram showing where to clamp the CT’s in the generator?

jgyates, 2020-11-07:
This thread has some diagrams and discussion on adding CTs

jgyates, 2020-11-07:
https://github.com/jgyates/genmon/issues/403