jgyates/genmon #469: Status Page and some data not loading in Chrome

rastam4n, 2020-11-10:
### Expected Behavior

Genmon is expected to work as it did before update

### Actual Behavior

After I updated to the latest version the status page no longer loads properly in chrome. I also have a system warning stating that the system is initializing and not receiving data. 
I can however load the page no problem using IOS 

### Steps to Reproduce (including precondition)

I just opened the shortcut in chrome like I always have

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: G0070310
- Genmon Logs: Submit via the About page
- Genmon Version: V1.15.09

jgyates, 2020-11-10:
There is not a lot of info to go on in your description. I have not received any logs on this issue. Is your outgoing mail working? You can only submit logs if your outgoing email is functional. 

I can not duplicate this problem. I would suggest disabling any browser plugins that may be preventing chrome from loading the page properly. If you want to provide more details here are the things that would be helpful:

1) Logs. 
2) Pictures of the monitor page in the web interface
3) what operating system are you using. You can run chrome on various windows version, Mac and Linux
4) What browser extensions are you using? you can see this in chrome from this url: chrome://extensions/

jgyates, 2020-11-13:
I am closing this issue due to inactivity. Feel free to post to this thread if you wish to pick this up again.

closed 2020-11-13 by jgyates

rastam4n, 2020-11-24:
Hello sorry for the delay. I continue to have issues, it seems as though
the genmon is not receiving data from the Generator. I have submitted a log
via the log page.
Here is a picture of the web interface status page
It does not matter if I access it via ios, or windows or linux all of them
have no data, although IOS does load the gauges regardless.I am using
chrome on all devices
[image: image.png]
extensions
[image: image.png]
Thank you for taking a look at this. It was working fine for years until I
pushed that upgrade button.... WHY did i do that!
Kind regards
Peter Rastonis

jgyates, 2020-11-24:
Your images do not show up if you respond to the emails. you must attach them in the web interface at github.com.

Also, I have not received any logs today. Is your outbound email working? Sending logs requires the outbound email to be functional.

rastam4n, 2020-11-24:
[image]
[image]
[image]

rastam4n, 2020-11-24:
I have repaired my email issue and resent the logs, Thank you

rastam4n, 2020-11-24:
So I have disabled Malwarebytes, that seems to have fixed the gauges not appearing but the RPI is still not communicating with the Generator.

jgyates, 2020-11-24:
I got your logs. Looks like it was quite a while since you have upgraded so I am guessing that part of the issues is that there were several libraries that needed to be installed. The genloader.py program will attempt to install the missing libraries, but since it has been a while since you updated it may take several minutes. To make sure that this is not an issue let's go through the install procedure again.

Log into your pi via ssh then type:

     cd genmon
    ./startgenmon.sh stop
    ./genmonmaint.sh -u

Type Y to confirm updating to the latest software. Then type:

      ./genmonmaint.sh -i

This will resinstall the libraires needed for genmon. You should be prompted to install the config files but you can say No. Also you can indicate No to the other questions (install at boot, etc) as you already have that done.

Once this is done you can update your libraries to the latest:

       ./genmonmaint.sh -r

Then restart genmon:

       ./startgenmon.sh restart

Then open the web interface (or refresh your browser) and see if you are communicating. If you are not then the next step is to validate your cable and Rs-232 converter. To do that follow the steps on this page:

     https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

Let me know how it goes or if you have any questions.

rastam4n, 2020-11-24:
I am getting a lot of errors during  the update, and a lot of instability in the terminal window, long waits, lag etc. 
I did manage to run the python serialtest.py and it failed so it seems I will need to see what happened to my cable. Maybe someone damaged it while servicing the generator etc. Will update. Thank you

rastam4n, 2020-11-26:
So the cable was fine, but the act of unplugging and plugging it in seems to have solved the problem. I had obviously cycled the power many times before so it couldn't be a "reboot needed"... maybe just a weak connection? 
That was a lot of effort for a simple solution.

jgyates, 2020-11-26:
I am glad it is working now. Thanks for letting me know.