jgyates/genmon #470: GENMON SYSTEM WARNING - Log file is reporting errors

stronzo31, 2020-11-13:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

At the bottom of the web page, the last line was light gray with no errors displayed,

### Actual Behavior

At the bottom of the  web gui, there is a red bubble with an exclamation mark. While hovering over the bubble, the message "GENMON SYSTEM WARNING Log file is reporting errors" appears. 

### Steps to Reproduce (including precondition)

Rebooted the pi server, stopped and started Genmon, problem persists. Started happening after an outage today. Did not happen during exercise tests or when remote starting/stopping prior to today. I have not performed any remote starting or exercise tests since this condition started.

### Screenshot or Pictures relating to the problem (if possible)

I checked the genmon.log file and this only started happening after an outage, but continues. The log entry displays the following:

2020-11-12 20:47:17,352 : Error in  GetAveragePower: time data '\x00\x00\x00\x00
\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
\x00\x00\x00\x0011/12/20 10:14:28' does not match format '%x %X' : controller.py
:1051
2020-11-12 20:47:41,644 : Error in  GetAveragePower: time data '\x00\x00\x00\x00
\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
\x00\x00\x00\x0011/12/20 10:14:28' does not match format '%x %X' : controller.py
:1051

Log entries appear about every 20 seconds or thereabouts.

### Your Environment

- Generator Model: G0070351, Generac Air Cooled, 16KW
- Generator Controller: Evolution 2 
- Genmon Logs: Submit via the About page
- Genmon Version: V1.15.10

[image]

[image]

Thank you,
John

jgyates, 2020-11-13:
John,

Looking at your logs and registers it looks like there are two issues:

1) You generator has an alarm log entry that the software has not encountered. If you can look at the exact text in the last entry in your controllers alarm log and post that entry here I can add it to the software.
2)  Your power log appears to be corrupt. The easiest thing to do is to reset the power log on the Maintenance page. This will make the fuel readings off unless you either wait and do it when your tank is filled or your can use the Subtract Fuel parameter on the advanced page (double click the gear icon in the top right for the advanced page) to estimate the amount of fuel to subtract from a full tank to get you to the actual tank level. If you use the Subtract Fuel parameter be sure to zero out this the next time your fill your tank and reset the power log. Alternatively you can edit your log file (/etc/genmon/kwlog.txt) and try to delete any entries that are invalid. This approach may not work, depending on how bad the file is corrupt.

I am not sure what your log is corrupt, it could be a SD card starting to fail, or it could be a power issue. I have added some additional checking that may make the software more resilient if this does occur again.

Let me know what you find on item one above and also let me know if item 2 works for you.

Thanks
Jason

stronzo31, 2020-11-13:
Hi Jason,

Thank you for your quick response! Much appreciated!

Regarding #1, I accidentally entered "Yes" on my generator to setup the WiFi module when I had the connector unplugged because I was using Genmon. So that error is due to the absence of the WiFi module.

[image]

Regarding issue #2, I took your advice and edited the /etc/genmon/kwlog.txt file and found one bad entry (attached). I removed the bad characters in front of that entry and everything is all good.

[image]

Please let me know if you need anything else.

Thank you again and great job!
John

jgyates, 2020-11-13:
Thanks John,

I updated the software with the new error code. I did not increase the version number so just go to the About page and click update and when the software restarts you should see the correct log entry on the log page. Let me know if you have any other issues.

closed 2020-11-13 by jgyates

stronzo31, 2020-11-13:
Thank you Jason. Updated the software and the log entry is correct now. All issues are resolved! Many thanks.
John