jgyates/genmon #471: Unable to establish connection between raspberry pi zero w and Evolution 2.0 FW 1.11

bgalderisi, 2020-11-13:
### Expected Behavior

Unable to establish connection between raspberry pi zero w and Evolution 2.0 FW 1.11

### Actual Behavior

genmon.log
GenMon Loaded for site: Porter using python 2.7
WARNING: Initilization not complete after 30 seconds, possible communication failure. Check cabling.

### Steps to Reproduce (including precondition)

New installation.  Genmon retrieved via git and installed via genmonmain.sh -i.  Raspberry pi zero w with 3.3v female ttl to 232 serial converter purchased from amazon.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}
[image]

### Your Environment

lrwxrwxrwx 1 root root 7 Nov 12 22:23 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Nov 12 22:23 /dev/serial1 -> ttyS0
Raspberry pi zero w with 3.3v female ttl to 232 serial converter purchased from amazon.  Male to female adapter was not used.  Rather, the wires from the molex were soldered to the 232 serial adapter.

- Generator Model: 16KW
- Genmon Logs: Submit via the About page
- Genmon Version: 1.15.10

logs sent from house1733*@*si.com email address.

jgyates, 2020-11-13:
Have you tried the steps listed on this page? 

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If so what are the results? A quick thing to try is to swap your TX and RX and try again. This is probably the most common problem. If that does not work then you need to go thru the steps in the link above to validate what is an is not working.

jgyates, 2020-11-17:
I am closing this issue due to inactivity. If you want to pick this up you can respond to this thread.

closed 2020-11-17 by jgyates

bgalderisi, 2020-11-20:
Sorry for the delay.  Below is what I have found.

pi@pi-generac:~/genmon/OtherApps $ sudo python serialconfig.py -c

Checking : BT service disabled: OK
Checking : Serial console command line removed: OK
Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console service disabled: OK

Serial port settings are OK. A reboot is needed if changes were made.

pi@pi-generac:~/genmon/OtherApps $ python ./serialtest.py 

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

After loopback and config test, the below is found in the genmon.log.

2020-11-20 00:16:36,786 : GenMon Loaded for site: Porter using python 2.7
2020-11-20 00:17:09,116 : WARNING: Initilization not complete after 30 seconds, possible communication failure. Check cabling.

I then shut it down.  
2020-11-20 00:21:54,337 : Generator Monitor Shutdown

I then verified that the pins from the molex were correctly routed to the right inputs on the serial male adapter, but switched them just to check.  

2020-11-20 00:24:41,274 : GenMon Loaded for site: Porter using python 2.7
2020-11-20 00:25:13,570 : WARNING: Initilization not complete after 30 seconds, possible communication failure. Check cabling.

No change, so I shut it down and reverted it.
2020-11-20 00:26:51,285 : Generator Monitor Shutdown

There are no recent messages in myserial.log.

I tailed the mymodbus.log and found the following messages during startup.
[tail_modbus.log](https://github.com/jgyates/genmon/files/5570527/tail_modbus.log)

jgyates, 2020-11-20:
I would be useful to see the rest of your logs. The fact that the encryption is failing is problematic:

Error in MyCrypto:Encrypt: initializer for ctype 'unsigned char *' must be a cdata pointer, not bytearray : mycrypto.py:59

Based on the data I have from this one log, this error is causing packets not to be sent. If your outgoing email is working your can send your logs to me on the About page.

bgalderisi, 2020-11-20:
Here is the picture of my serial connections between the molex and the ttl to rs232.
[image]

Here is a picture of the connections between the pi and the ttl to rs232.  This picture shows the markings on the ttl to rs232 better than it shows the pins on the pi itself (below).
[image]

The ttl to rs232 is powered from 3.3v.
[image]

bgalderisi, 2020-11-20:
Logs should be sent just now from house####*@*si.com email address.

bgalderisi, 2020-11-20:
Actually I see in the genmon.log that when I click to send logs I get the below error:

`2020-11-20 01:04:53,868 : Unable to convert type <type 'NoneType'> in GetDispatchItem
2020-11-20 01:04:53,877 : Item: None`

jgyates, 2020-11-20:
The error:

     2020-11-20 01:04:53,868 : Unable to convert type <type 'NoneType'> in GetDispatchItem 2020-11-20 01:04:53,877 : Item: None

is normal if you have not received any valid packets yet, but it should not keep your from emailing your logs. Have your email settings changed since the 13th? I looked back and I received logs from you on the 13th, but not today, not sure what it was working before but not now. Have you re-installed since the 13th?

Are you seeing Modbus Exceptions on the Monitor page of the web interface Monitor->Communication Stats->Modbus Exceptions. I am guessing you are seeing this number increase ever few seconds? Is this correct? If this is the case then serial communications is working and what is not working is the python encryption libraries.

At this point we should do some validation on your install. Log into the pi and from the genmon directory type:

        ./startgenmon.sh stop
        ./genmonmaint.sh -u

Then yet 'Y' when prompted. This will update to the latest software. Then type:

       ./genmonmaint.sh -r

This command will update the python libraries that where installed during the initial setup. As this run, look for any errors that may occur. Then type:

        ./startgenmon.sh restart

Then load the web interface and send a screen capture of the first 2 sections.

jgyates, 2020-11-20:
If the above does not work then try this:

      genmonmaint.sh -i

This will reinstall the libraries. There are a couple of steps here that if skipped would cause the crypto stuff to fail. If you type Y, when prompted to write the config files it will overwrite any config files you have saved.

bgalderisi, 2020-11-20:
I probably did an upgrade since the 13th, but I am able to receive test emails.

Are you seeing Modbus Exceptions on the Monitor page of the web interface Monitor->Communication Stats->Modbus Exceptions.
**Yes**

    Communication Stats :
        Packet Count : M: 2850, S: 2849
        CRC Errors : 0
        CRC Percent Errors : 0.00%
        Timeout Errors : 0
        Timeout Percent Errors : 0.00%
        Modbus Exceptions : 2849

1. (re)updated to 1.15.11 (previously installed)
2. updating libraries (pyserial updated from 3.2.1 to 3.4, and pyopenssl updated seemed relevant since this seems to be a communications/(de/en)cryption issue issue).  A bunch of other things were updated and already up to date as well.  

Side question: does updating with -u (or through web ui) not refresh libraries as well?

Updating (-r) is taking a very long time.  My internet connection is string with low latency and high bandwidth.  I will provide an update on what happens after the restart.

When you say "steps here that if skipped would cause..." is that something I could have possibly skipped?  This was a fresh setup/install of 1.11 just a couple of weeks ago and I installed it with "genmonmain.sh -i".

bgalderisi, 2020-11-20:
Still updating...

Not an error, but a notice I spotted during the update.

Successfully installed cffi-1.14.3 cryptography-3.2.1 enum34-1.1.10 ipaddress-1.0.23 pycparser-2.20 pyopenssl-19.1.0 six-1.15.0
/usr/local/lib/python2.7/dist-packages/OpenSSL/crypto.py:12: CryptographyDeprecationWarning: **Python 2** is no longer supported by the Python core team. **Support for it is now deprecated in cryptography, and will be removed in a future release**.
  from cryptography import x509

Not sure if it is pertinent.

jgyates, 2020-11-20:
The libraries are not updated unless you run the genmonmaint.sh -r command. This prevents the libraries from being updated unless something is not working. The update from the web interface or with the -u command line option only updates the genmon software from github.

"steps here that if skipped would cause..." this means that if your internet connection takes to long the operation may timeout or some other error retrieving the info on the internet.

"My internet connection is string with low latency and high bandwidth." Is this a typo? I am not sure what you intended to say here?

bgalderisi, 2020-11-20:
string=strong.  

I was intending to say that it doesn't make a whole lot of sense why it is still updating after close to 30 minutes.

That being said, after restarting the software (./startgenmon.sh restart) it seems to be communicating for the first time.  There are a few errors in the modbus log:

2020-11-20 02:43:30,271 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2020-11-20 02:43:30,276 : Error Receiving slave packet for register 01f4
2020-11-20 02:43:30,311 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2020-11-20 02:43:30,313 : Error Receiving slave packet for register 0000
2020-11-20 02:43:30,347 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2020-11-20 02:43:30,349 : Error Receiving slave packet for register 0064

[image]

jgyates, 2020-11-20:
ok. This looks normal. The software uses the modbus exceptions to detect if an Evolution 2.0 firmware with encryption is present. After three exceptions it switches over to encrypted mode.

My guess is that this line may have timed out in the script:

     sudo apt-get -yqq install build-essential libssl-dev libffi-dev python-dev

this builds the encryption libraries.

Anyway, it looks like it is working now. Let me know if you have any other issues.

jgyates, 2020-11-20:
is your email working now as well? Fixing the encryption issue may have helped email also.

bgalderisi, 2020-11-20:
Seems like it is. I see it in my sent items.

jgyates, 2020-11-20:
I received your logs. Looks like you are all set. Let me know if you have any other issues.

bgalderisi, 2020-11-20:
Thanks.  And thanks for doing what you do.