jgyates/genmon #472: External Tank Fulel Monitor shows Zero after Genmon update to v1.15.11

jwtaylor310, 2020-11-16:
I updated Genmon from v1.15.10 to v1.15.11 this morning.  I am using the External Tank Fuel Monitor add-on to read my fuel level with TankUtility.  After the upgrade I received a 'low fuel' email and the status page displays a level of 0%.  When I log on to the TankUtility web page it shows a normal value (currently 76%).  I have re-entered my TankUtility password in the add-on without success.  Can you help?  Thanks!

jgyates, 2020-11-16:
Have you refreshed yoru browsers? If that does not correct the issue, I would need to see your logs.

jwtaylor310, 2020-11-16:
I have refreshed my browser with no change.  Since the system sent me an email warning that the "estimated fuel in the tank is at or below 10%" I don't think it is a browser issue.  I also power cycled the Pi with no improvement.   I have submitted a copy of the logs.  Please let me know if you need anything else.

jgyates, 2020-11-16:
Your logs look good with one exception. 

     Error : The program gentankutil is already loaded

One of the things that was added in V1.15.11 is that each add on is now checked to make sure it is not loaded twice. Could you be loading this program (gentankutil.py) explicitly? If it was power cycled then that would rule out a zombie copy hanging out in memory. Are you doing loading the application the standard way (via the startgenmon.sh script as a crontab job) or are you doing anything unique in your setup?

One thing you can try is to disable (comment out) the cron job that starts genmon and then reboot. Then check the processes (with the top command) to see if gentankutil.py  is loaded. This will let us know if it is loaded outside of the normal load process. Let me know if you can do that your if you need instructions.

jwtaylor310, 2020-11-16:
I am starting startgenmon.sh as a cron job.  Cron also has a job that runs a daily program at 00:30 that I wrote named "TankUtility.py".  That program connects to the Tank Utility web site and verifies that it can log on to my account, checks the date of the last reading and sends an email if the last reading is over 24 hours old.  It does not invoke gentankutil.py and should not have been active when I ran the Genmon update or when I did the cold reboot after that.  However, I commented out that job, rebooted (a warm restart) and Genmon then started reporting the tank level correctly.  I re-enabled the cron TankUtility job, rebooted (another warm restart) and Genmon continues to read the tank level correctly.  That doesn't make a lot of sense to me.  However, everything appears to be working now so I can't complain!  I will check to see if my TankUtility program runs correctly tomorrow morning and will post the results here.  In the event of any future problems, I have the option of moving my TankUtility.py program to another Pi.  Thanks for your help!

jgyates, 2020-11-16:
It is this code that is having an issue:

https://github.com/jgyates/genmon/blob/master/genmonlib/mysupport.py#L414

I will do some testing and try to make this a bit more bullet proof. In the meantime if this is any issue you can go to the advanced page and enable "Allow Multiple Genmon Instances". This will disable this check. Now that I know the name of the script I should be able to duplicate and fix this. Thanks. I will let you know if I check in a fix.

jgyates, 2020-11-16:
I checked in an update that should fix this. Now the python script name must be an exact match. Let me know if this works for you.

jwtaylor310, 2020-11-17:
My "TankUtility.py" script ran at 00:30 as usual and this morning Genmon is continuing to correctly report the external tank level.  A few minutes ago I ran "Upgrade to latest version" from the "About" screen to get the fix and everything continues to work normally.  Thanks for the great support!  You can close this issue.

closed 2020-11-17 by jgyates