jgyates/genmon #473: Incorrect Frequency 120Hz Displayed

brett-kimball, 2020-11-18:
This issue seems similar to one reported by @moab11 at the end of issue https://github.com/jgyates/genmon/issues/68#issuecomment-621499366_

There was no further follow up on that report.

When running, genmon is displaying a frequency of 120Hz instead of 60Hz.  This is also reflected in the running email message.

Model : QT08054JNAX
Generator Serial Number : 3003683772
Controller Detected : Nexus, Liquid Cooled
Nominal RPM : 3600
Rated kW : 80 kW
Nominal Frequency : 60 Hz
Fuel Type : Natural Gas
Engine Displacement : 5.4 L

[image]
[image]
[image]

brett-kimball, 2020-11-18:
The Nexus controller appears to be properly identified.  The genmon.conf file was created after the communications to the controller were fully tested and operational.

jgyates, 2020-11-18:
Sending your logs would help. I don't have a lot of info to go on with what you have posted so far. You can send them on the about page.

liltux, 2020-11-18:
@jgyates this frequency value is read from a register, could there be more conversion on this value on the QT 3600 RPM units?  Maybe this register is a raw value and not a processed value.  Both units in question at QT(nexus controlled) and 3600 RPM rather than 1800 (which would be double).  I am curious if there are other users out there with Nexus controlled units running at above 1800 RPM and if their frequency values are accurate or similarly off?

liltux, 2020-11-18:
I was reading more of the original post, and now I am wondering if you are dealing with a firmware difference.  The Nexus controlled generators are not field updatable for firmware.  The new controllers, signified with a letter identifier, would have updated firmware.  Wonder what the Firmware version was from the original post compared to this post?

jgyates, 2020-11-18:
the way the software works now is this:

Evolution Liquid cooled: modbus reg 0008 / 10  (i.e. register is freq with a tens decimal place)
Evolution and Nexus Air Cooled: modbus reg 0008 (i.e. register is frequency with no decimal places)
Nexus Liquid cooled : modbus reg 0008 * 2

It is possible that there are firmware differences between the Nexus Air cooled units, but so far I have not gotten a set of registers that shows this (while the system is running) so I can add any code that I can test.

brett-kimball, 2020-11-18:
I have sent the logs.  Let me know if you need additional information or register info when the unit is running.

jgyates, 2020-11-18:
I got your logs. Thanks. If you can send the logs again when the generator is running that would be helpful. If the running logs show register 0008 is the correct frequency (without multiplying by 2) I will add an option to make calculation different for Nexus liquid cooled units.

brett-kimball, 2020-11-18:
Looks like the frequency is shown accurately: "003c".

jgyates, 2020-11-18:
I checked in a fix for this. There is now an option on the advanced page (double click the gear icon in the top right for the Advanced page) named "Use Nexus Legacy Frequency". Set this value to False and it should work for you. I made the default value True as to not disrupt other users who have this installed and it is working for them.

I also added this to the known issues list:

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

Thanks for the feedback. Let me know if you have any issues. I am closing this thread but feel free to post here if you have any issues or questions.

closed 2020-11-18 by jgyates

brett-kimball, 2020-11-18:
Thank you!  It seems to be working fine.