jgyates/genmon #474: Opto-Isolator solved my communication problem

jwtaylor310, 2020-11-23:
The Raspberry Pi for my Genmon installation is located in my garage and is connected to the generator with an RS232 cable about 75 feet long.  I have a Uxcell a14041500ux12 RS232-to-TTL converter installed at the Pi.  This has worked great with virtually no CRC or timeout errors.  However, I found that  anytime I needed to power down the Pi it caused the generator's controller to disable its RS232 port.  To get communications working again, I had to power up the Pi and then restart the generator's controller.

I solved the problem by adding an opto-isolator at the Pi end of the the RS232 line.  It isolates the transmit, receive and ground pins on the generator side from those on the Pi's side, using led's and photo-electric cells to transmit the data between the two sides.  The opto-isolator that I installed has a female DB9 connector on one end and a male DB9 on the other, so installation was a simple matter of inserting the opto-isolater into the RS232 line at the Uxcell level converter.  The opto-isolator is powered from the data lines, so it needs only a 3-wire RS232 connection (transmit data, receive data and ground) and it does not require an external power supply.

If you have a similar installation or are experiencing other communications issues, this may work for you.  It should be helpful for problems associated with different ground potentials between Pi and the generator.

The specific device that I installed is a Dtech DT9011 (https://www.dtechelectronics.com/dtech-dt-9011-passive-rs232-photoelectric-isolation-protector_p90.html).  I purchased mine from Amazon for $17.98: https://www.amazon.com/gp/product/B01EACRB0K/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Other brands/models may work similarly, but be aware that some "port powered" opto-isolators take their power from RS232 lines other than the Transmit/Receive wires and will require additional wiring or an external power supply in a typical Genmon installation.

jgyates, 2020-11-23:
That is interesting, thanks for sharing this info. This is probably some weird anomaly of the controllers RS232 interface (and possibly the Pi's RS232 converters) coupled with the cable length.

Out of curiosity, how is your RS-232 cable constructed? is it CAT-5 or CAT-6?

This site has some interesting data on cable length, not sure this is relevant but it could be..:

https://www.lammertbies.nl/comm/info/rs-232-specs#phys

"Cable length is one of the most discussed items in RS232 world. The standard has a clear answer, the maximum cable length is 50 feet, or the cable length equal to a capacitance of 2500 pF. The latter rule is often forgotten. This means that using a cable with low capacitance allows you to span longer distances without going beyond the limitations of the standard. If for example UTP CAT-5 cable is used with a typical capacitance of 17 pF/ft, the maximum allowed cable length is 147 feet."

jgyates, 2020-11-23:
I added a link to this thread in the tips and tricks wiki entry:

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon

jwtaylor310, 2020-11-23:
I am using a cable that is marketed for LED lights:  https://www.amazon.com/gp/product/B01MQPN2L2/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1
I chose it instead of Cat5/6 cable because it is outdoor rated and sunlight resistant (I live in Florida) and it is stranded which makes it better able to handle vibration from the generator.  It has a 14 gauge ground wire plus two 18 gauge wires which I am using for the transmit and receive data lines.  The manufacturer doesn't provide capacitance per foot information since that isn't pertinent to its intended purpose (led lighting).   However, I doubt that cable length/capacitance is an issue here since excess capacitance would typically result in a high error rate.  In my case, prior to installation of the isolator Genmon typically reported zero communication errors even after weeks of continuous operation.  The only problem occurred when the Pi was shut down.  Once that happened, Genmon would show 100% timeout errors until the generator's controller was restarted.

I expect that the "port lockup" issue that I experienced was the result of the Uxcell RS23-to-TTL putting an out of bounds load on it's Receive Data line (the Transmit Data line from the generator's perspective) when the Uxcell lost power.  That could cause the generator's Transmit Data line driver to latch up (or down) requiring a power cycle to reset.  Another possibility is that when the Pi was powered down it did something to the ground reference that it shares with the generator that the generator didn't like.  When the opto-isolator is in the circuit the generator's Transmit Data line always sees the same load (the led emitter in the opto-isolator) and the Pi's ground is no longer shared with the generator.  

With the isolator in the circuit Genmon reports zero communication errors over prolonged periods of time and I can power down the Pi and/or unplug its power supply and when it is brought back up Genmon immediately "sees" the generator controller and begins normal operation.

The isolator may provide additional value by protecting the generator's controller from possible damage if the Pi is hit by a power surge.  For $18 installing an isolator is probably good insurance even if "port lockup" isn't a problem.  (This would only apply to installations like mine where the Pi is at a distance from the generator.  Genmon users who mount their Pi in the generator enclosure with power from the generator battery shouldn't have a problem with this.)

closed 2020-11-27 by jgyates

kc-nj, 2021-01-03:
Anyone happen to know if a loopback test should pass with the DT-9011 in place?

I'm setting up a new system. With only the RS232 to TTL adapter & RS232 connector in place (loopback on the connector side) the loopback test passes. With the DT-9011 in place, it fails. Based on my (albeit limited) understanding, I would expect it to still pass even though it is an optical connection, not a physical one. 

I'm looking to use the DT-9011 as a preventative measure, since I'm going to have the Pi inside of the house, so there is a longer run between the Pi & the genset.

jgyates, 2021-01-03:
The website says it is half duplex (from the site, did not find a manual) so that would imply that it can transmit or receive, but not at the same time. This should be OK for general use for gnemon, but a loopback will not work in half duplex mode.

jwtaylor310, 2021-01-03:
I would expect the DT-9011 to fail a hardware loopback test.

 

The DT-9011 derives its power from the data lines.  By design, it isolates each side of the connection from the other side.  If you are trying to do a loop-back by connecting the transmit pin to the receive pin on one side of the DT-9011, that side won’t have any power and data won’t flow to/from it.

 

That doesn’t mean the DT-9011 isn’t functional.  If you connect it to a ‘live’ port like the generator it should then see power on that side and start passing data.  If you have a notebook or PC with a serial port (or another Pi) you can connect it in the place of the generator and use terminal emulator software on both sides to test the connection.  However, that is probably not worth the trouble since it is likely that the DT-9011 will work just fine once it has ‘live’ connections on both sides.  

 

My recommendation is to initially make the connection from the Pi to the generator without the DT-9011 in the circuit, get Genmon running and then install the DT-9011.

 

My installation is similar to yours, with the Pi in my garage and a 75’ cable running to the generator outside.  The DT-9011 is installed at the Pi end between the data line and the RS22 to TTL adapter.  It has worked flawlessly and has eliminated my previous problem with the generator communication  going off-line anytime the data line was disrupted.

 

Good luck!

kc-nj, 2021-01-03:
Thanks for the reply. I was typing up an update when your reply came through. 

I did essentially what you stated. I connected to the genset without the isolator in place and had communication. I then installed the isolator and had no communication. Removed it and communication was restored. 

It's questionable if I really need it, my run is only about 30 feet. I am using CL3R rated wire however, which my research shows has a higher than ideal capacitance rating.

jwtaylor310, 2021-01-03:
Not sure which web site you are viewing.  The manufacturer’s site (https://www.dtechelectronics.com/dtech-dt-9011-passive-rs232-photoelectric-isolation-protector_p90.html) refers to both “full” and “half” full duplex.  I think they mean it runs full duplex in a point-to-point configuration (like a Pi talking to a generator) but half duplex when running in a multipoint configuration with multiple devices sharing the communications line.  In any event, it is working fine on my installation of Genmon.  Nothing to configure or fiddle with – I just unplugged the serial cable at the Pi end and inserted the DT-9011 in the line.

 

 

--John

kc-nj, 2021-01-03:
I was looking at that mfg site. If you're asking about the website I was using regarding the capacitance of the wire, I was looking at the site Jason posted above, and Googled CL3R specs to find the capacitance. 

Maybe the DT-9011 needs power on one side and I have it backwards for what it wants? My RS232 to TTL adapter is Male, so the Female end of the isolator is on the Pi side.

jgyates, 2021-01-03:
Yeah, it says full duplex then right below it mentions two modes, both of which are half duplex so it is hard to say which one is the typo. If you are not getting comms with both the loopback and with genmon then I would check swap your tx and rx and see if that fixes the issue. Some devices are labeled what they are and some are labeled what to connect them to (DTE vs DCE). Are there LEDs on the oscillator to indicate traffic?

jgyates, 2021-01-03:
I would try swapping the power first

kc-nj, 2021-01-03:
No LEDs on any of the components involved. When you say swap power, you mean get gender changers or different serial adapters so the isolator is going the other direction?

With the isolator removed I get communication with the generator.

kc-nj, 2021-01-03:
The directions that came with the isolator mention having the Female side to the PC, so it seems to be going the correct way. 

The RS232 to TTL adapter instructions called for 3.3V DC from the Pi. I tried the 5V line when I was running the loopback tests and it still failed. I did not try the 5V line when I had it connected to the generator.

jgyates, 2021-01-03:
yes. That is just a guess, but if one side is not producing the voltage levels needed reversing the isolator is probably not going to fix it but it worth trying. If this is powered from the RS-232 signals instead of a dedicated supply then it would probably (again a guess) have two independent circuits , each supplying power for isolation. RS-232 line levels are between +-5 and +-15 volts. The isolator may require something like 7 as a minimum and  one side is only outputting 5 or 6 volts.

kc-nj, 2021-01-03:
The mfg says it is powered from the TXD, RTS or DTR lines. They don't say what voltage is needed or if it is needed on both ends though. 
Strange that it was plug and play for jw. 
At this point, I will run it as is and hope that with my shorter run, I will not have the same problems jw had which led him to install an isolator.

jwtaylor310, 2021-01-03:
While it is possible that the problem is the voltage level, I think it is more likely that you have the send and receive data lines reversed.  My RS232 to TTL adapter has a female DB9 connector and the male connector of the isolator goes to it.  That is consistent with the convention of using a female connector on computers and male connector on external devices, which is the reverse of what you have described for your system.

 

If I am correct, you will need to swap pins 2 and 3 on both sides of the connection.  If your RS232 to TTL adapter is connected to the GPIO pins on the Pi, you can just reverse the jumpers at the GPIO header  to take care of the connection on the Pi.  You will then need to make the same swap on the connector attached to your data line leading to the generator.  With the pins reversed at both places, the circuit should still work when connected without the isolator, but it should also work with the isolator in the circuit.  

 

If available, you can try a pair of “gender changer” adapters which would allow you to reverse the direction of the isolator.  If you try this, you will need to use gender changers that have “straight-through” connections (i.e, pin 2 goes to pin 2 and pin 3 goes to pin 3 on both sides of the adapter).  Some devices that appear to be gender changers are actually “null modems” that reverse the send/receive data pins which in this case would put you back where you started.

jwtaylor310, 2021-01-03:
Also ... if the problem is reversed send/receive lines, it is likely that if you fix that the system will still fail a loopback test since there will be no power on one side of the isolator.  However, it should work with Genmon when the generator is attached since both sides of the isolator will then see power.

It is "plug and play" for my installation because I have a female connector on the Pi end.  With that configuration the send and receive lines are connected to the correct pins on the isolator.  With a male connector on the Pi end, the Pi is sending data out on the same line that the isolator is trying to use to send data to the Pi, and the Pi is listening on the same line that the isolator is  using to listen for data coming from the Pi.  That's a classic RS232 configuration problem.

kc-nj, 2021-01-03:
I was wondering why you were thinking I needed to reverse the tx/rx lines when I had communication with the isolator removed, but I understand your thinking now. 

I have done a few complete shutdowns and removal of the power supply to the Pi and so far it has reconnected to the generator each time. Hopefully that means with my shorter run and maybe different spec'd cable, I will not have the problems you had. 

For now I'm going to let it run as is and see what happens. 

Thank you both!