jgyates/genmon #475: Cron job for backup

restauffer, 2020-11-24:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Be able to schedule backup of Genmon via Cron

### Actual Behavior

Not sure if this ability already exists

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

Is it possible to schedule a cron job on a Raspberry Pi to run the backup script for Genmon?

Thanks

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2020-11-24:
This command:

     /bin/bash /home/pi/genmon/genmonmaint.sh -b

will create a file named genmon_backup.tar.gz in the /home/pi/genmon directory.

If you are using a different directory than the default you can use:

     /bin/bash /home/pi/genmon/genmonmaint.sh -b -c /etc/someotherfolder/

restauffer, 2020-11-24:
Thank you. That was just what I needed.

closed 2020-11-24 by restauffer