jgyates/genmon #476: Genmon cellular connection?

nfunk70068, 2020-11-30:
This looks like a great project and upgrade over the existing limited Mobil link access.  

My only concern is if power does go down in the area, so will broadband access and connection.  I was thinking if the Raspberry Genmon can be accessed via a cellular connection.   I see many of the cellular dongles & boards are relatively inexpensive, but not sure if this is possible using the Raspberry.

Any thoughts on this idea?

Thanks,
Nick

jgyates, 2020-11-30:
There is an add on that can be enabled on the add on page that sends SMS with generator status via Raspberry pi based cellular modem. Here is more info:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gensms_modempy-optional

This implementation uses this hardware:

https://wiki.seeedstudio.com/LTE_Cat_1_Pi_HAT/

Cellular network data is not supported in this add on. If you want full TCP/IP communications over cellular and not just SMS then you would need to look into doing this at the operating system level.  If you want to use cellular data, then genmon would function as it normally does with the built in networking (Ethernet or Wifi). Setting up a cellular card to connect to the internet would be an operating system configuration / setup and is outside the scope of the project.

Let me know if you have any questions.

nfunk70068, 2020-11-30:
I will check it out!

Thank you for the prompt response!

closed 2020-12-01 by jgyates