jgyates/genmon #477: Genmon Box and Setup Pictures

cmanion1983, 2020-12-06:
Hey all, figured I'd post a few photos of my setup. Came out really well, thanks for all the help from you guys!

[image]
[image]
[image]
[image]
[image]

skipfire, 2020-12-06:
Where does the serial wire exit the generator enclosure?  My case is so tight I did not see a weather tight way of bringing the wire out.

cmanion1983, 2020-12-06:
I brought mine out right through the natural gas pipe exit point and the zip tied it up the electrical conduit. 

[image]
[image]
[image]

hokie21, 2020-12-10:
POE powered. Nice!

ravedoglv, 2021-02-13:
I noticed you have an LED indicator on the outside. What is it reflecting and what did you do in software to light it up? Would you share the process? Thanks!

jgyates, 2021-02-13:
There are two add on programs that will blink the LED

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gengpioledblinkpy-optional

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gengpiopy-optional

Both can be enabled on the Add  On page. I am guessing he is using the first one as this is the one that supports one led. the second one supports multiple LEDs.

cmanion1983, 2021-02-13:
> I noticed you have an LED indicator on the outside. What is it reflecting and what did you do in software to light it up? Would you share the process? Thanks!

Mine doesn't blink, I just wanted to be able to look out in the garage and see if it was on. I believe I used a 10k ohm resistor and a standard LED bulb. Then, I connected it to the 3.3v pin, or you can connect to the 5v pin if you want it brighter.

ravedoglv, 2021-02-13:
The am you both!

ravedoglv, 2021-02-13:
Question:on gengpioledblink.py it says to issue the command 

`python gengpioledblink.py`

Is that a one time command or if not, how does the system know to run the command at startup?

jgyates, 2021-02-13:
just enable it on the add on page. This is all you need to do. The other info is if you want to run it outside of the loader.

jgyates, 2021-02-13:
I updated the wiki to make this more clear for each add on program.

ravedoglv, 2021-02-13:
Thanks again!

sarahmva, 2021-03-05:
[image]

Fairly small box. Lab hat to connect a temperature sensor and using a prolific USB to RS232. Unfortunately, the 12V to USB Power device did not deliver enough power to keep the Raspi running.

Clarke219, 2021-05-18:
Nice looking job. Could you tell me where your enclosure label came from? It differs from the one here.
Thank you.

mikewesner, 2021-11-05:
I am adding some photos of the location I put mine because I was pretty unclear as to if this box that I found on the docs would even fit, or if the little cable would work.   I think a longer cable would give better reach with a hole near the bottom rather than up top like I had to mount it.  I bought the hi pi version of the HAT and a case,  I bought a Ras pi 3b+ because i didn't want to deal with fans, just little heat sync and lower power.  I ran an ethernet cable but did not use POE because I figured the shared ground would be more stable if I used the battery based connection. (especially with power outage, loads coming on/off messing with line voltages and things)

[image]

[image]

[image]

TopBanana123, 2021-11-06:
Here's my install with an RG027, I ran 1/2 Liquid tight conduit in the trench and used the spare knockout on the front

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sambeetm, 2022-03-03:
Thanks @jgyates for this great project and continuing support. Here is my setup. Everything is up and running since last sept flawlessly.

Some of the highlights of my setup.
1. Both mobile link and Genmon are working simultaneously.
2. Completely inside setup. Ran a shielded cable(with ground) from the generator to inside electrical panel. Around 12 feet.
3. External 5v, 3Amp power using wall adapter for PI 3B+. The power adapter is connected to a small a UPS along with bunch of other things. Common grounds.
4. Using GPIO Input and Otput add-ons. Works great.
5. A fan inside the box which kicks-off at 65F and stays on till 55F.
6. Integrated with Home Assistant using MQTT(just read only)
7. SSL does not work more than a day. I have to kill and restart the genserv every night at 2AM using cron job. Now, genserv stays up.
If anybody needs the template, parts list or more pictures etc, please let me know. I will add here.

[image]

davisgoodman, 2022-03-24:
> Thanks @jgyates for this great project and continuing support. Here is my setup. Everything is up and running since last sept flawlessly.
>
> Some of the highlights of my setup.
>
> 1. Both mobile link [...]

I would definitely be interested in getting your front panel template. I will also have an inside install having my generator right againt the back of my garage.

sambeetm, 2022-03-25:
@davisgoodman Here are some info.
Box......https://www.amazon.com/gp/product/B01N8SLCUM/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
[Box Label -  Final.pptx](https://github.com/jgyates/genmon/files/8346555/Box.Label.-.Final.pptx)

davisgoodman, 2022-03-25:
@sambeetm  Thanks so much. I already have the box but I'll definitely use the label. Looks great!

jgyates, 2022-06-10:
@gregmac, this is great! I added this to the wiki: 

     https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP#diy-serial-over-ip-using-and-esp32

mloebl, 2022-07-29:
Didn't want to drill a hole in the generator, and we're lucky where we are we don't need to lock it, so created a simple SMA antenna mount to replace the lock cylinder:
[image]
[image]
[image]

cmacnichol, 2022-10-05:
@sambeetm Can you share a Parts List?  I really like the look of what you did.  I am using a Pi-Zero W2 Knockoff and a phat, but I think it should work with this if I solder the wires to it directly.

sambeetm, 2022-10-06:
> @sambeetm Can you share a Parts List? I really like the look of what you did. I am using a Pi-Zero W2 Knockoff and a phat, but I think it should work with this if I solder the wires to it directly.

Here are some parts ordered from Amazon. Attached are the template/label and LED resister values.
 
https://www.amazon.com/gp/product/B01N8SLCUM/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B07J4ZCWNX/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B07QXR5MZB/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B08CVH8PVC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

[Box Label -  Final.pptx](https://github.com/jgyates/genmon/files/9719712/Box.Label.-.Final.pptx)
[LED Resistors.xlsx](https://github.com/jgyates/genmon/files/9719714/LED.Resistors.xlsx)

STiFTW, 2022-10-08:
A bit late to the party but happy to be here!

Just got everything setup on my 17KW Generac. Got a outdoor enclosure, but the cable glands would not fit the Molex connector.  I frustrated myself trying to de-pin the connector, then eventually decided to just cut it.  I was originally going to solder the connections, but decided to put Anderson Power Pole connectors on wires that I needed do that I could remove or service the wire in the future.  I ended up having to take it apart a few times during testing and installation, so very happy I left myself the option to disconnect easily.  I will watch for any errors. 

Mounted the weather resistant box just below where the lid hits when open.  Checked a few times to make sure there wasn't anything I was going to hit before putting in some self tappers to secure the box to the enclosure.  I ended up really tight on cable length, but I was able to make it work. 

[image]
[image]

Box I used: https://smile.amazon.com/dp/B09BJCLKPK?

TopBanana123, 2022-10-08:
> 

Might want a drip leg on that gas supply, unless its just out of shot

STiFTW, 2022-10-09:
>>Might want a drip leg on that gas supply, unless its just out of shot
That's a really good point.  I put a zip tie and left the tail near the grommet, will have to check next time it is raining.  Been installed 5 years like this, and I never considered that.

waynewarburton, 2023-03-04:
I unplugged the WIFI generac unit that was inside the enclosure and cut the molex connector 2 communication wires to tie in the Raspi.  I noticed the comment about having both Generac's wifi and their Pi both working at the same time.  I did the install a few years ago now and might revisit this to get both systems going.  How did you wire it in to the communication signal?

sambeetm, 2023-03-04:
My PI is externally powered using 5v Wall adapter and located inside the basement. So, I needed the GND. Ran a shielded cable with 5 wires(though you need only 3. GND, RX, TX) from connector to the PI. They are connected to pin 2, 7, 8 respectively. See the picture.
[image]
You need the "EPS 8 Pin Power Extension Cable - Power extension cable", if you don't want to cut the factory cables and revert the modification any time.
[image]

waynewarburton, 2023-03-04:
I did just change the oil and filter.  My hands were a bit cold by the time I finished.   It's still cold here so I'll check in the Spring when it's warmer and i can take the end cover off.

I recall buying that connector cable and cutting the 7 and 8 wires to feed the Pi.  I did unplug the WIFI unit connector.  If the Pi and WIFI can run in parallel, I'll plug it back in but at the time I was concerned about OTA firmware updates and encoding that was rumoured to be coming.  I have no idea if that ever transpired.

I've updated my desktop computer to Window 11 and now can't access the generator throught it.  The Pi is running fine and is connected to my home wifi and can be viewed locally with a monitor and keyboard plugged in.  I think it must be security settings...something stupid like 'trust unsecured apps' or something like that.  Diving into that next week.  Have you heard any issues with this?

On a side note, Generac has apparently increased the test run time to around 20 minutes to help with condensation accumulation in the oil.  I just increased my run time to 20 minutes.  First time I've changed it myself as it is now out of warrantee.  The oil was a bit milky looking so the longer run time is likely a good thing.

[molex view]<https://user-images.githubusercontent.com/18508058/222926212-1a3a151a-4557-49f5-93a4-4b57209f6851.PNG>
[81ILVoGoHyL AC_SL1500]<https://user-images.githubusercontent.com/18508058/222926472-6d531b77-7602-4476-a6eb-b588f513c3a5.jpg>

skipfire, 2023-03-04:
A tech noticed condensation in my oil as well, I've been doing a 1 hour transfer test every month in addition to the 5 minute weekly and that has taken care of the problem.  If your controller (I didn't see you listing what controller you have) supports that you may give it a try, it also ensures that the generator works under load and that the transfer works so you get extra validation from it.

It is possible that Windows is blocking the site due to not being secure, but it isn't supposed to do that for local addresses.  You say that it's fine with a keyboard and monitor plugged in, was that pulling up the website on localhost?

For the firmware, I'm guessing you are referring to the encryption update, it did happen and jgyates worked some magic and GenMon works again, but I don't trust Generac to not try something like that again so I keep my WiFi disconnected.

You can also DIY a 3-way cable with a couple of the yellow cables sambeetm listed and some 3-way crimp splicers.  Or buy a splitter from [PintSize.Me](https://pintsize.me) (transparency: I'm the guy behind PintSize.Me).

waynewarburton, 2023-03-04:
I have the Evolution Controller on a 2018 Generac 20KW unit.  I was thinking of reconnecting the wifi because I lost communication on my desktop.  Now you have me going back to plan A and leaving it disconnected.  Companies are always looking for more money and personally, thier App is terrible.  It offers little information and an expensive monthly fee for the service.  Genmon is far more superior.

With the keyboard and monitor plugged in to the Pi, localhost works fine.  I think my communication issue is to do with trusted sites or a router setting.  I'll pick away at it over the next couple weeks and let you know what I find or don't find.

skipfire, 2023-03-04:
You can always add telnet (no clue why Microsoft decided to not have it install by default anymore) and try connecting to the web port that way.  A successful connection will give you just a blank screen, unsuccessful will give you an error message.  That can help you determine if it is a problem accessing the port or a security configuration issue on your computer, though I would recommend taking it to a new issue if you need further help since it's off topic for this issue.

el0552t, 2023-03-08:
Can anyone tell me if this program is still working with the Evolution 2.0 controllers? I see activity in the respiratory that leads me to believe there could be changes made by Generac to block communication through the serial connection but can't locate a definitive answer anywhere. My unit was installed 12/2022

skipfire, 2023-03-08:
Are you referring to the encryption change a few years ago?  That did happen and GenMon works fine with my Evo2 controller.

el0552t, 2023-03-08:
Yes, that is what I was referring to. I believe I'll build one of up and try it out. I already have a method of monitoring my power consumption and frequency but was interested in seeing and logging as much of the data as possible directly from the generator. Thanks for the response.

waynewarburton, 2023-04-01:
My communication issues were due to a dual router configuration I was running.  Got rid of one router and added a mesh extender and all is good now.

I have updatd the firmware and looks awesome.  More information is always better.

I do have a question though.  Can you point me to where I can change the exercise running time?  Not the schedules start time, but the actual exercising time.  Trying to lengthen it to get rid of the moisture issue with the shorter times.  I have found it before but it now escapes me.

Thanks in advance,
Wayne Warburton

jgyates, 2023-04-01:
It is in the dealer edit menu

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#access-the-dealer-menu

Dealer - > Dealer Edit, then scroll thru the parameters until you get to Exercise Duration.

waynewarburton, 2023-04-01:
Went into the dealer section and believe it or not, there is no duration selection.  I contacted our local dealer that was servicing it and they said it is not a parameter you can change.  He also said not to worry about the moisture in the synthetic oil as there have been no reported issues.  I don't understand why they wouldn't just change it to a longer run time.  Eventually this will cause more wear than normal and premature damage.

I'm retired and mostly around the house.  I'll just run it manually for 20-30 once a week during the colder weather.  I don't think summer should be an issue just east of Toronto.

Thanks for your excellent service.  Keep up the good work.

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#access-the-dealer-menu

jgyates, 2023-04-01:
you can, depending on your controller model. Some models do not allow the duration of the exercise to change. There is an add on that will allow you to add a 2nd exercise cycle. See the Add On page in the genmon web interface.

waynewarburton, 2023-04-01:
Very nice.  I've never looked there before.  I'll have to checkout the other add-ons too.

capindc, 2023-05-12:
Thank You JGYates what a great peace of software, I start my project less then a week ago and now its up
Genmon running in a Ubuntu/VM on a Truenas Server 24 Core 128 Gig 34 TB server
also I did donate but did not hit the sponsore button.
Thank you again

[image]
[image]
[image]

jgyates, 2023-05-12:
Thanks for the feedback and donation. I am glad it is working well for you.