jgyates/genmon #479: Feature Request: Ability to disable Output Voltage guage on status page.

lwbeam, 2020-12-17:
I have an air-cooled Pre-Nexus model, which was installed in 2010. When I first installed my monitor, I noticed that it failed to show the Output Voltage when the machine was running. This was the case during exercise runs and actual utility outages. In both cases, the Frequency and RPMs are displayed correctly, but no Output Voltage.

At first I thought that the software might be looking at the wrong register for this vintage controller. However, after having a good look at the maintenance manual wire diagrams for this machine, as well as the machine's wiring, I've come to the conclusion that there is no way for the controller to monitor the output voltage. Simply put, the controller is not connected in any way to the generator's output.

I've since noticed that there is a setting that allows the Power and Current displays to be disabled for Evolution controllers. I was wondering if you could add a similar setting for Pre-Nexus controllers, that would allow the display of the Output Voltage to be disabled?

Having it displayed isn't a huge problem; however, I normally monitor the status of my generator on my iPad, which only displays four gauges across the screen. Eliminating the unused gauge would prevent the unnecessary wraparound, and significantly tidy up the display. Thanks!

jgyates, 2020-12-17:
What value is read from register 0012, is it always zero or does it change?

jgyates, 2020-12-17:
I am trying to determine if I should disable the output voltage gauge and display for all PreNexus units or just make this an option to not show the gauge.

FYI you can display the low bandwidth version of the web interface (with no gauges) with this url:

      http://YourIPAddressGoesHere:8000/low

lwbeam, 2020-12-17:
The value of register 0012 is zero, and never changes. Checked this several time under various operating states.

I've just had another good look at the maintenance manual with respect to the differences between the various size machines. Frankly, I'm at a loss as to how the controller detects the machine's Frequency and RPMs. However, the following observations may provide some guidance.

8kW machines - simple controller with no Remote Wireless Connection (page 98), and no Voltage Regulator (page 164). Not relevant, as no way to connect monitor to controller. 

10kW machines - Pre-Nexus controller with Remote Wireless Connection (page 99), and no Voltage Regulator (page 168). Although there is a way to connect monitor, controller likely unable to sense Frequency, RPMs or Output Voltage.

14kW, 17kW and 20kW machines - Pre-Nexus controller with Remote Wireless Connection (page 99), and Voltage Regulator (pages 172, 176 and 180). Controller likely able to sense Frequency and RPMs, but not Output Voltage.

I was aware of the low bandwidth version, but having the other gauges is very useful, have not used it.

Once again, many thanks for having a look at this.

jgyates, 2020-12-18:
I checked in version 1.15.13. In this version pre Nexus models will not display output voltage data. Let me know if you have any issues with this.

closed 2020-12-18 by jgyates

lwbeam, 2020-12-18:
Have loaded it, and it seems to work as fine. If I see any issues, will let you know. Again, many thanks.

jgyates, 2024-10-04:
@lwbeam 
do you see anything like the behavior described in this thread: https://github.com/jgyates/genmon/issues/1191