jgyates/genmon #480: Syntax errors reported when running /startgenmon.sh after upgrade from 1.7.? to the latest

tmorse305, 2020-12-19:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior
Expected genmon to start

### Actual Behavior
./startgenmon.sh: 7: ./startgenmon.sh: Bad substitution
./startgenmon.sh: 15: ./startgenmon.sh: Syntax error: "(" unexpected

### Steps to Reproduce (including precondition)

I ran these commands
git fetch origin
git reset --hard origin/master
sudo chmod 755 ./startgenmon.sh
sh ./startgenmon.sh start
Then I got the errors

Then I tried:
sudo pip install pytz
 sudo pip install configparser
Same errors

I then tried 
bash ./genmonmaint.sh -i
bash ./genmonmaint.sh -r
But same error was reported
### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac
- Genmon Logs: Submit via the About page
- Genmon Version: was 1.7.? (I don't remember, I think is was 4)

Thank you

jgyates, 2020-12-19:
I am not really sure what is happening. I can not duplicate this with the current version. The two bash scripts in question have not changed in a while. 

What linux distribution are your running or are you running this on another operating system?
Are you getting the *exact* same errors when running genmonmaint.sh as startgenmon.sh

Can you run the following without error?

      ./genmonmaint.sh -?

tmorse305, 2020-12-19:
Sorry, I confused you, here is the sequence again:

I ran these commands
git fetch origin
git reset --hard origin/master
sudo chmod 755 ./startgenmon.sh
sh ./startgenmon.sh start
Then I got the errors

Then I tried:
sudo pip install pytz
sudo pip install configparser
sh ./startgenmon.sh start
Same errors

I then tried
bash ./genmonmaint.sh -i
bash ./genmonmaint.sh -r
sh ./startgenmon.sh start
But same error was reported

Only startgenmon.sh fails with errors, everything else completed successfully.  I'm running it on:
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l

jgyates, 2020-12-19:
OK. I think the problem is that you are using 'sh' instead of 'bash'. Try this:

     bash ./startgenmon.sh

If I use 'sh' instead of 'bash' I get the same error. At first glance I thought you left of the 'ba' in 'bash' but it looks like you are actually using 'sh'.  The default in raspbian is bash so your could just type:

      ./startgenmon.sh start

I will look at this closer. I think I can make this work on both sh and bash. but this should correct the issue you are seeing.

tmorse305, 2020-12-19:
Thank you, that worked!  I am a novice with RPi, I googled how to run the script and came up with the sh command first so went with it.  I appreciate your help, thanks again.

jgyates, 2020-12-19:
No problem. I am glad it is working now.

closed 2020-12-19 by jgyates