jgyates/genmon #482: New install, model shows unknown

wesgipe, 2020-12-30:
### Expected Behavior

Model number should be detected.

### Actual Behavior

I have GenMon up and running great.  Everything is reading correctly, commands are passed perfectly, and serial com looks good.  The first time I ran GenMon, it was not hooked up to serial, and I'm wondering if that made it so that the model shows unknown in the Maintenance tab.  I saw the other posts about editing genmon.conf and removing the version line, but that file on my system does not have the version line in it (I have searched it just to make sure I wasn't going crazy).  Is it possible that there is a new way to force a new lookup to correct?

### Screenshot or Pictures relating to the problem (if possible)

Here's what shows in the maintenance tab:

Maintenance :
    Model : Unknown
    Generator Serial Number : 3004586880
    Controller Detected : Evolution 2.0, Air Cooled
    Nominal RPM : 3600
    Rated kW : 22 kW
    Nominal Frequency : 60 Hz
    Fuel Type : Propane
    Engine Displacement : 999 cc
    Ambient Temperature Sensor : 41 F
    kW Hours in last 30 days : 0.00 kWh
    Fuel Consumption in last 30 days : 0.02 gal
    Total Power Log Fuel Consumption : 0.02 gal
    Run Hours in last 30 days : 0.02 h

    Controller Settings :
        Calibrate Current 1 : 1489
        Calibrate Current 2 : 1504
        Calibrate Volts : 1040
        Nominal Line Voltage : 240 V
        Rated Max Power : 22 kW

    Exercise :
        Exercise Time : Weekly Wednesday 13:35 Quiet Mode On

    Service :
        Service A Due : 199 hrs or 12/29/2022
        Service B Due : 399 hrs or 12/28/2024
        Battery Check Due : 12/29/2021
        Total Run Hours : 21.00 h
        Hardware Version : V1.00
        Firmware Version : V1.11

### Your Environment

- Generator Model: 22Kw, Evo2
- Genmon Logs: Submit via the About page - Done
- Genmon Version: 1.15.13

jgyates, 2020-12-30:
the file you need to edit is /etc/genmon/genmon.conf. This file should have the version number in it. The file /home/pi/genmon/conf/genmon.conf is not the file used at run time. It is only a copy of the original version so this file would not have the version info in it. If you remove the line with version info from this file, the line to remove should look something like this:

     version = 1.15.13

Then restart genmon. From the /home/pi/genmon directory type:

     ./startgenmon.sh restart

On restart the software will attempt to lookup the model number from your serial number over the internet.

Also, if the model number is unknown the next software update would also force a lookup.

wesgipe, 2020-12-30:
Thanks so much!  That worked perfectly.  What you've created here is absolutely fantastic.

closed 2020-12-30 by wesgipe