jgyates/genmon #483: Lets Encrypt Cert Error

wesgipe, 2020-12-30:
### Expected Behavior

"sudo certbot certonly –manually –preferred-challenges dns" prompts for domain name

### Actual Behavior

Command generates the following error:

Usage: 
  certbot [SUBCOMMAND] [options] [-d DOMAIN] [-d DOMAIN] ...

Certbot can obtain and install HTTPS/TLS/SSL certificates.  By default,
it will attempt to use a webserver both for obtaining and installing the
certificate. 
certbot: error: unrecognized arguments: -preferred-challenges dns

### Your Environment

- Generator Model: 22kw Evo2
- Genmon Logs: Submit via the About page
- Genmon Version: 1.15.13

jgyates, 2020-12-31:
I duplicated your issue. I am am guessing that certbot changed their command line. I will look into this and see if I can figure it out. I don't use a cert on my genmon setup as this info was provided by another user, but I will try to look into this. In the meantime if anyone know the latest Let's Encrypt / certbot usage please feel free to post here. Once I get the updated info I will also update the wiki entry

skipfire, 2020-12-31:
Be sure you use double-hyphen before the arguments not single or dash.  Those frequently get lost in docs or copy/paste and I had that problem as well.

wesgipe, 2020-12-31:
Thanks guys.  I put a double hyphen: "sudo certbot certonly -manually --preferred-challenges dns"

Now this is what I get:

Saving debug log to /var/log/letsencrypt/letsencrypt.log

How would you like to authenticate with the ACME CA?
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
1: Spin up a temporary webserver (standalone)
2: Place files in webroot directory (webroot)
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Select the appropriate number [1-2] then [enter] (press 'c' to cancel): 1
Plugins selected: Authenticator standalone, Installer None

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Please read the Terms of Service at
https://letsencrypt.org/documents/LE-SA-v1.2-November-15-2017.pdf. You must
agree in order to register with the ACME server at
https://acme-v02.api.letsencrypt.org/directory
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
(A)gree/(C)ancel: a
There seem to be problems with that address. Enter email address (used for
urgent renewal and security notices)

If you really want to skip this, you can run the client with
--register-unsafely-without-email but make sure you then backup your account key
from /etc/letsencrypt/accounts

 (Enter 'c' to cancel): c
Unable to register an account with ACME server

I canceled, as I didn't want to mess anything up.  I'm sure I'm doing something stupid here - I just don't know what it is...

jgyates, 2020-12-31:
What is odd is that 'manually' takes one hyphen but gives and error if you give it two. 'preferred-challenges' gives errors unless you give it two hyphens.

jgyates, 2020-12-31:
You do have a DNS name registered, correct. Dynamic DNS will not work with this method as you need to update your DNS records as one of the later steps in the process.

I updated the wiki to use the double hyphen as shown above.

skipfire, 2020-12-31:
If you are getting an error with double hyphen before manual something else is wrong as spelled out words require double hyphen and single letters require a single hyphen.  I used the standalone web server with port 80 forwarded to the Pi as the replication for DNS entries was too slow for me to reliably go that path.

jgyates, 2020-12-31:
I think it should be 'manual' instead of 'manually'. This works:

    sudo certbot certonly --manual --preferred-challenges dns

wesgipe, 2020-12-31:
That's it!  Thanks so much.

closed 2020-12-31 by wesgipe