jgyates/genmon #484: Missing Dashboard on new install

kc-nj, 2020-12-31:
### Expected Behavior

Dashboard with status gauges at top of webpage. 

### Actual Behavior

No Dashboard with gauges, text only. 

I am in the process of setting up a new Pi to replace a pre-built unit that has been failing. I've got a new Pi 3B+, seem to have everything set up except for the physical connection to the genset. I'm waiting for the RS232 adapter to come in yet. 

I was able to export/restore the backup file from the old to the new, which was successful, all settings are there, and I got the SMS & email messages when it restarted. I should note the Dashboard was missing before I did the restore as well. 

Is it possible the dashboard will not show until it communicates with the genset?

### Your Environment

- Generator Model: Generac 17K, air cooled, Evolution controller. 
- Genmon Logs: not yet submitted. 
- Genmon Version: V1.15.13

jgyates, 2020-12-31:
If the serial comms are not working then genmon will signal to the web server that the initialization is not complete, which will not allow the web app to initialize the gauges since the software does not know what type of generator is attached. Once the serial comms are working you should see gauges. Let me know if you have any other questions.

closed 2020-12-31 by jgyates

kc-nj, 2020-12-31:
Thanks for the fast reply & that's what I was hoping. 

It's great you make this available. It was long overdue, but I just donated also.