jgyates/genmon #485: Issues with tankutility after update

SethL, 2020-12-31:
Having an issue with the external tank functionality after updating genmon.  It shows 0 now for some reason, used to work fine before.  I tried multiple browsers and also cleared my cache.  I submitted the logs as well.

Thanks!

jgyates, 2020-12-31:
I received your logs. the entry in the file /var/log/gentankutil.log that is interesting is this one:

     Error : The program gentankutil.py is already loaded

A few version ago I added checks to prevent the loading of multiple copies of add on programs. If you have another python program named gentankutil.py running or if you are running multiple copies of genmon this would cause this error. You can bypass this check (for example if you want to run multiple copies of genmon) by enabling the option "Allow Multiple Genmon Instances" on the Advanced page (double click the gear icon in the upper right of the web interface for the advanced options).

If you have another python program named gentankutil.py then you can just rename it something else and this should resolve the issue. 

Let me know if this resolves your issue.

SethL, 2020-12-31:
Hmm the weird thing is I never did anything like that, I just installed it and it worked 😊

Maybe I should reboot?

jgyates, 2020-12-31:
Yes, that may fix it. If could be that an existing copy of an older version if not exited. You log indicates that this has occured before. The last three entries are:

      2020-11-18 15:26:33,378 : Error : The program gentankutil.py is already loaded
      2020-12-28 12:28:47,221 : Error : The program gentankutil.py is already loaded
      2020-12-31 05:07:36,153 : Error : The program gentankutil.py is already loaded

SethL, 2020-12-31:
Yeah that’s weird…. I just rebooted and all is well 😊

 

Thanks!

closed 2020-12-31 by jgyates