jgyates/genmon #487: Invalid time entry in power log

zbourdeau, 2021-01-02:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Log parsing on startup doesn't result in error.

### Actual Behavior

Log reports:

2021-01-01 16:49:44,566 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:02' does n                                                                                ot match format '%x %X'
2021-01-01 16:49:44,570 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:01' does n                                                                                ot match format '%x %X'
2021-01-01 16:49:44,666 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:02' does n                                                                                ot match format '%x %X'
2021-01-01 16:49:44,669 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:01' does n                                                                                ot match format '%x %X'
2021-01-01 16:49:44,788 : Invalid time entry in power log: time data '\x00\x00\x00\x00\x00\x00\x0                                                                                0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:01' does                                                                                 not match format '%x %X'
2021-01-01 16:49:44,790 : Invalid time entry in power log: time data '\x00\x00\x00\x00\x00\x00\x0                                                                                0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:02' does                                                                                 not match format '%x %X'
2021-01-01 16:49:44,793 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:02' does n                                                                                ot match format '%x %X'
2021-01-01 16:49:44,794 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:01' does n                                                                                ot match format '%x %X'
2021-01-01 16:51:03,805 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:02' does n                                                                                ot match format '%x %X'
2021-01-01 16:51:03,809 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:01' does n                                                                                ot match format '%x %X'
2021-01-01 16:51:03,898 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:02' does n                                                                                ot match format '%x %X'
2021-01-01 16:51:03,900 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:01' does n                                                                                ot match format '%x %X'
2021-01-01 16:51:04,016 : Invalid time entry in power log: time data '\x00\x00\x00\x00\x00\x00\x0                                                                                0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:01' does                                                                                 not match format '%x %X'
2021-01-01 16:51:04,018 : Invalid time entry in power log: time data '\x00\x00\x00\x00\x00\x00\x0                                                                                0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:02' does                                                                                 not match format '%x %X'
2021-01-01 16:51:04,021 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:02' does n                                                                                ot match format '%x %X'
2021-01-01 16:51:04,023 : Error in GetPowerLogForMinutes: time data '\x00\x00\x00\x00\x00\x00\x00                                                                                \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0001/13/20 10:25:01' does n                                                                                ot match format '%x %X'

### Steps to Reproduce (including precondition)

Restart of RPi or upgrade to latest version.

### Screenshot or Pictures relating to the problem (if possible)

See above.

### Your Environment

- Generator Model: G0062602
- Genmon Logs: Done
- Genmon Version: 1.15.13

jgyates, 2021-01-02:
I have seen this once before, but I do not have enough info to attempt a software fix. If you can do two things it would help me try to discover the issue.

1) Send your complete logs. If your outbound email is working you can send your software logs for genmon by clicking the submit logs button on the About page on the web interface.
2) Get the file /etc/genmon/kwlog.txt and attached it to this thread. This is your power log. To attach the file to this thread, on the non mobile github site, drop the file into the edit window of this thread. 

Once you send the above data you can correct the bad log entires in one of two ways.

1) Edit the file /etc/genmon/kwlog.txt and remove the lines with odd looking entries. Normal entries should look like this:

        12/23/20 14:03:20,3.805

This is  <DATE and TIME, Power in kW>
This is the prefered as you do not have to reset the power log and your fuel estimates will be entact. 

2) Alternatively you can reset the power log. This will delete the entire power log, which will also reset your fuel estimate calculations. You can reset the power log at the bottom of the Maintenance page. If you want to keep your fuel estimates somewhat intact, you can enter the amount of fuel that has been used (i.e. the fuel to subtract from a full tank to get to your current fuel level) on the Advanced page in the "Subtract Fuel" entry.

Let me know if you have any questions.

jgyates, 2021-01-02:
you can also get the kwlog.txt file by extracting the backup file obtained from the About page

jgyates, 2021-01-02:
I checked in version 1.15.14. This has additional checks that *should* fix the problem and also avoid it in the future. If you could still send your logs and kwlog.txt that would still help.

Let me know how this works.

zbourdeau, 2021-01-03:
I have attached my full log. Thanks for the release. BTW, At a glance it looks like there are a few extra nulls chars that are causing the issue. I am not sure why they are geting inser
[kwlog.txt](https://github.com/jgyates/genmon/files/5761267/kwlog.txt)
ted into the file.

zbourdeau, 2021-01-03:
It looks like removing the two erronous lines in the log fixed the issue. The patch didn't seem to help before the bad entries were removed. 

I am not sure why the extra chars got into the log almost a year ago. I was probably running a current release at that time. I have updated since and it looks like that pattern has not existed since so perhaps it was a logging issue in an earlier release?

jgyates, 2021-01-03:
Thanks, the log helps a lot. I checked in an update that will not fix the issue you are seeing. It removes the unwanted characters. I am still not sure how they go there. The update I just checked in did not change the version number. I am going to close this issue now but let me know if you see it again.

closed 2021-01-03 by jgyates