jgyates/genmon #488: Service Journal entries

berenguerj, 2021-01-02:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Service Journal entries 

### Actual Behavior

Lost Service Journal entries

### Steps to Reproduce (including precondition)

For some reason, after an update, (using other than Firefox) lost access to the web console. Was able to SSH into RPI and reinstalled, without luck. Changed MicroSD card and did a full installation and all is now working without problems. 
Since I still have the original card, what is the Service Journal file named and located? I will try to copy the original file so that I can recover my previous journal entries

I'm not an expert in Linux, so any tips are greatly appreciated. Thanks and happy holidays everyone.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}
[image]

### Your Environment

- Generator Model: Guardian 9Kv Evolution Controller Air Cooled
- Genmon Logs: Submit via the About page
- Genmon Version: Version V1.15.14

jgyates, 2021-01-02:
If you have the old SD card then you can copy the files from /etc/genmon from the old SD card to the new one. This directory contains all conf file and power log (kwlog.txt), outage log (outage.txt), service journal (maintlog.json) and, if enabled the fuel log (fuellog.txt).

I am not sure what happened to your old system, possible some corruption in your SD card. Nothing has changed in the web interface in a while.

berenguerj, 2021-01-02:
Thanks for your fast reply. That's was my guess, something wrong with the SD card...I will copy the directory and post results. Thanks again

berenguerj, 2021-01-02:
Was able to copy back the journal. Thanks again!!

[image]

closed 2021-01-02 by berenguerj