jgyates/genmon #490: i have customers that would like to add Genmon monitoring to their units

grsthegreat, 2021-01-04:
but i can no longer find the pre-assembled units for sale on Ebay

back when i added mine, the pre-assembeled package was a god send to someone like me with limited knowledge on stuff computer related... especially Pi related.

Does anyone know of anybody still selling these units?

skipfire, 2021-01-04:
The cable is really easy to make, cutting and stripping wires, then screwing them into terminals.  You can also use an RS232 HAT instead of the little RS232 board that you have to use jumper wires for.  Add it with a HighPi enclosure and it's a clean build.

liltux, 2021-01-04:
I was the one building them on Ebay.  Just like skipfire says the the HighPi enclosure and the MODMYPI RS232 hat is what I have been using my self lately.  It is not sealed, but if a sealed enclosure is needed you can place that into a sealed enclosure.  If you need me to make you more off of this build let me know liltux at liltux.com.

liltux, 2021-01-04:
@skipfire I was also setting up a little webui for non-linux people to setup their wifi or static IP for them.  Basically they plug in, connect, setup via the webui and rock in roll.  Should not have any need for terminal access or little to null linux knowledge.

skipfire, 2021-01-04:
@liltux cool, didn't realize it was a completely ready-to-go unit.

grsthegreat, 2021-01-04:
yes, i have one of your units. works great.
i wanted to program it to auto reconnect to wifi network, but just too complicated. i just chose to leave it well enough alone.

i participate in several generator websites and people are always asking how to get a Genmon unit. would you mind if i pass along your email address?

liltux, 2021-01-04:
That is fine.  Just to calm anyone reading, I do give back portion to Mr. jgyates anytime I build one for someone.  @grsthegreat I can add that script to the crontab no problem.  I have been needing to update my image for awhile.

closed 2021-01-05 by jgyates

aquaman30k, 2021-03-01:
> I was the one building them on Ebay. Just like skipfire says the the HighPi enclosure and the MODMYPI RS232 hat is what I have been using my self lately. It is not sealed, but if a sealed enclosure [...]

Would this board work in place of the MODMYPI RS232 board? Having a hard time locating the MODMYPI HAT.
https://www.amazon.com/2-Channel-Isolated-RS232-HAT-Protection/dp/B083D27NJD

jgyates, 2021-03-01:
No, this isolated board appears to use SPI for the serial port.

These will likely work if you are wanting a HAT

https://thepihut.com/collections/raspberry-pi-serial-i2c-hats/products/serial-plus

https://thepihut.com/collections/raspberry-pi-serial-i2c-hats/products/serial-pizero

If you are not dead set on a HAT then this will work:

https://www.amazon.com/Ableconn-PI232DB9M-Compact-Expansion-Raspberry/dp/B00WPBXDJC/ref=pd_sbs_1?pd_rd_w=PhK1o&pf_rd_p=527ea27c-adf6-4b67-9c5f-265eb29e0622&pf_rd_r=P1FQ8CQR41DVFBDY8FP8&pd_rd_r=19279e6d-8b2c-4bc7-bfd0-5d410c25819f&pd_rd_wg=WHZQV&pd_rd_i=B00WPBXDJC&psc=1

grsthegreat, 2023-09-07:
Liltux. I now have 4 to 5 people interested in purchasing a complete genmon box like you sold me. Are you still making them?

skipfire, 2023-09-28:
@grsthegreat I have an image that uses a WiFi portal to configure WiFi. Feel free to contact me directly, there is a contact page at the bottom of [https://pintsize.me](https://pintsize.me).