jgyates/genmon #491: MQTT on Docker Help

mefbgil, 2021-01-04:
Hi everyone - very exciting day for me - got Genmon running inside Docker connected to my Generac thru a USR-W610. Everything worked on the first try - powered cable to the USR, USR setup on my WiFi, Dockerfile build, and TCP connectivity to the USR - all good - EXCEPT - MQTT. Any ideas where I should look for issues? The MQTT log is empty and no trace of error message in any other logs. Any help is good help. Python3 Dockerfile and docker-compose attached.

Thanks!

[python3.zip](https://github.com/jgyates/genmon/files/5767309/python3.zip)

jgyates, 2021-01-05:
You should be able to use this command or something like it to subscribe to the topic and see all published data:

     mosquitto_sub -h localhost -t generator/# -u username -P password -F %t:%p

Where localhost is the IP address of your MQTT broker server.

Also, you can set 

      debug = True 

in /etc/genmon/genmqtt.conf to get more info logged.

mefbgil, 2021-01-05:
OK - tried mosquitto_sub and nothing shows.  Added debug to the conf file and the "genmqtt.log" file is still completely blank.  It's as if it's not even getting initialized...

jgyates, 2021-01-05:
First try rebooting your pi, you may have a stranded thread in genmqtt that is preventing it from loading a new copy.

      sudo reboot now

Then let genmon start and test with mosquitto to see if messages are making it to the broker server. If they are not then get into the genmon directory and type this:

     sudo python genmqtt.py 

Do you get this message:

     Error : The program genmqtt.py is already loaded

or some other message?

mefbgil, 2021-01-05:
OK - now we're getting somewhere.

root@56bca075add1:/git/genmon# sudo python3 genmqtt.py
Error in MyMQTT init: [Errno 110] Connection timed out

jgyates, 2021-01-05:
I found one issue with the logging on genmqtt. I had made a change a couple of version ago that broke logging for this addon. I just checked in a fix. I did not bump the version number as this was a minor tweak so just update from the About page and this should get the logging going.

mefbgil, 2021-01-05:
Logging is back.  Does this tell you anything?  I've tried multiple MQTT clients connecting to this same broker and all work fine.  

2021-01-04 20:50:17,459 : Connecting to 10.0.0.146:1883
2021-01-04 20:50:48,843 : Error in MyMQTT init: [Errno 110] Connection timed out : genmqtt.py:413
2021-01-04 20:51:54,134 : Connecting to 10.0.0.146:1883
2021-01-04 20:52:25,611 : Error in MyMQTT init: [Errno 110] Connection timed out : genmqtt.py:413
2021-01-04 20:52:38,064 : Connecting to 10.0.0.146:1883
2021-01-04 20:53:09,643 : Error in MyMQTT init: [Errno 110] Connection timed out : genmqtt.py:413

jgyates, 2021-01-05:
It tells me that genmqtt.py can not connect to your broker. First validate that it is at IP address  10.0.0.146 and using port 1883. 

After that you sould use the same IP address, port, username and password with mosquitto_sub (to monitor) and mosquitto_pub (to publish) something to your broker. It is possible that your broker is setup to use a different TLS version. What are your other settings on genmqtt?

mefbgil, 2021-01-05:
OK - I'm going to try to bring the MQTT container into an internal vs common network connection.  That will avoid genmon needing to go out of the host and back to see it.  Stay tuned.  Will share the updated docker_compose if it works.

mefbgil, 2021-01-05:
Success!  Network section of the docker-compose file fixed it.  Turned off cert in the Genmon config.  Thank you @jgyates for guiding me thru it.

[docker-compose.yml.zip](https://github.com/jgyates/genmon/files/5770739/docker-compose.yml.zip)

closed 2021-01-05 by mefbgil