jgyates/genmon #494: Engine Displacement and Rated Max Power are incorrect

chia9876, 2021-01-15:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Engine Displacement: 999 cc
Rated Max Power : 24 kW (or 21kW for NG)

### Actual Behavior

Engine Displacement: 530 cc
Rated Max Power : 11 kW

### Steps to Reproduce (including precondition)

Checking the Maintenace page shows the incorrect values.
 

 Also, I also believe this needs to be added to generac_evolution.py
  ["24KW", "60", "120/240", "1", "999 cc", "240"]         # Evo G0072100 

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}
[image]

### Your Environment

- Generator Model: G0072100
- Genmon Logs: Submit via the About page
- Genmon Version: V1.15.15
-
-*Sent the Registers & Logs

jgyates, 2021-01-15:
I have not received a copy of your logs/registers. Do you have outgoing email configured and working and did you submit your registers / log on the about page?

Adding the line to generac_evolution.py is likely the solution however I can not test this without register . The value in list selected in generac_evolution.py is dependant on registers. 

To date I do not have any registers for a 24kw Evolution 2 generator so it would be helpful to see your registers.

chia9876, 2021-01-15:
Sorry about that. I changed my email password a while ago but I use Twillo for text messages so I didn't notice that that email had stopped.

I just submitted both Registers & Logs again after verifying that email works. Hopefully, you have them now?!?1

jgyates, 2021-01-15:
Got them, thanks. I will make this update soon. I will let you know once it is check in.

chia9876, 2021-01-15:
Thanks. I have had the generator and Genmon since Dec 1st and have been waiting for someone to put a ticket in to add the 24kW but I guess I'll be the one.

jgyates, 2021-01-16:
@chia9876 ,

I checked in V1.15.16 which should fix your problem. Let me know if you have any issues.

Thanks
Jason

closed 2021-01-16 by jgyates

xjrguy, 2021-02-15:
I have the same issue as the OP but mine is a 22kw and reports Rated Max Power as 15kw which is incorrect.

[image]