jgyates/genmon #495: Starting with 1.15.15, after upgrade, web GUI not accessible.  Have to reboot PI then accessable.

DC072012, 2021-01-16:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}. In the past when I ran an upgrade on the “about” page of the Genmon gui, the update occurred, the program restarted, and I was immediately able to re-access the GUI.

### Actual Behavior

{Please write here}
Starting with 1.15.15 (or maybe 1.15.14, I can’t remember), I hit the upgrade button,  I get the downloading message, then the restarting message.  Then my browser says “server not found” when I try to re-access the GUI.  Have to reboot the PI by power cycling or accessing over wifi and rebooting.  PI is connected to wifi with good signal, that is not the issue.

### Steps to Reproduce (including precondition)

{Please write here}
Problem started on update 1.15.15 and continues every time I update.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- G0058820
- Genmon Logs: Submit via the About page
- Submitted
- Genmon Version: {Please write here}
- 1.15.16

USAFPride, 2021-01-16:
I had a similar issue.  Try the steps[ here.](https://github.com/jgyates/genmon/issues/450)

DC072012, 2021-01-16:
USAFPride, thanks for the quick response.  I read the entire thread and it appears you got it working but the thread is a long one and I’m not clear on what the eventual solution is.  Also, I bought this unit pre-built on ebay and my Linux / Pi knowledge is somewhat limited.  I have figured out how to SSH into the unit but it came with some Node Red coding and other ‘add-ons’ and I’m not sure what changes I can make without breaking something else.

If everything I need to do is in the Genmon directory then I might be OK.

Thx again

jgyates, 2021-01-16:
The fix in the above mentioned thread was this:

       sudo rm /etc/genmon/genloader.conf

Then type:

       sudo reboot now

This will delete the genloaer.conf file and restart the pi, which should restart everything and re-copy genloader.conf to /etc/genmon

DC072012, 2021-01-16:
OK, thanks.  Looks like this may have solved the problem.  Any idea how it got messed up? 

I had noticed my email messages didn't come this morning when it exercised.  I just did another upgrade from the About page and I got email notices this times and the GUI came right back up.

Hopefully solved.  Thank you again.

jgyates, 2021-01-16:
By best guess is that this could a SD card failure, possibly a sign that more failures are to come, but that is just a guess.

closed 2021-01-16 by jgyates

DC072012, 2021-01-17:
USAFPride, did you find out what caused your similar problem?  Specifically, was it the SD card going bad?  My SD card is about 1 1/2 years old.

Thanks in advance