jgyates/genmon #497: Unable to get IP Address

RoyRomeo, 2021-01-22:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

Unable to access GenMon

### Steps to Reproduce (including precondition)

The other day my Fios ONT failed and the tech came today to replace it.  After everything was back up and running, I tried to access the GenMon but could not.  I went out to the generator and hit the reset on the GenMon.  I was able to access the GenMon network and went to "wifisetup.pi:1880/GenPi".  I entered the network name and password and received a text message that the GenMon started.  I always access the generator by typing "192.168.1.170:8000/", but now nothing happens.  When I go to "Fing" and refresh, I do not see an IP for the GenMon.  Any help will be greatly appreciated.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

liltux, 2021-01-22:
@RoyRomeo when you reset your WiFi settings it cleared the static ip settings as well.  You will need to configure a static ip via the router or through the Genmon.  If you need more help with wifi you can message me at liltux at liltux .com.  here is about Genmon only.

jgyates, 2021-01-22:
Sounds like your pi as assigned a different IP address by your new router. Depending on your router, you could try to ping genmon.local  (assuming that gnemon is the name of your pi). Linux uses a protocol to allow the local IP name resolution local network access. For that matter you should be able to just use this as your address in your browser:

     http://genmon.local:8000/

RoyRomeo, 2021-01-22:
Thanks very much for the info.

closed 2021-01-25 by jgyates