jgyates/genmon #498: "External Tank Fuel Monitor" conflicts with "Estimated Fuel In Tank"

bbrown-sc, 2021-01-25:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

I'm using a TankUtility external tank fuel monitor. It integrates well into the Status page with a gauge and the Maintenance page with values (e.g. Fuel Level Sensor : 81.94 % - Fuel In Tank (Sensor) : 409.70 gal). Also displayed on the Maintenance page is an "Estimated Fuel in Tank" (e.g  : 498.70 gal) which is misleading because we have an actual accurate value provided by the TankUtility Fuel Level Sensor. I don't think the estimated fuel in tank should be displayed at all or calculations derived from, if there's an external fuel sensor installed.

I think the values for "Estimated Load", "Fuel Rate Half Load" and "Fuel Rate Full Load" values from the advanced page should be used only with the values that come from from "Fuel In Tank (Sensor)" to calculate hours of fuel remaining and not the "Estimated Fuel In Tank".

Also, when the generator is running only the "Estimated Fuel In Tank" is displayed and "Fuel In Tank (Sensor)" is not. The calculation for "Hours of Fuel Remaining" is based on "Estimated Fuel In Tank" and not "Fuel In Tank (Sensor)".

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac H-100 Industrial Generator - QT02524
- Genmon Logs: Submit via the About page
- Genmon Version: 1.15.16

jgyates, 2021-01-25:
I think you have three questions / request in here. Let me know if I am misunderstanding what you are requesting.

1) Regarding not displaying Estimated Fuel in tank if you have a sensor. I chose to have this included for a specific reason: Not all sensor are accurate and this was just a simple display. In my experience it tracks nicly (within a few gallons) with the sensor on my tank. If memory serves the issue you are having is that your power log is reporting invalid numbers due to your controller only outputting 0,1,2,3, or 4 for the power reading. This will cause your calculated fuel to be off which is likely why your calculated vs actual vary so much.

2) Your second request concerning using the fuel sensor for calculating the hours fuel remaining, this is how the software acts presently.

3) Your final question: "Also, when the generator is running only the "Estimated Fuel In Tank" is displayed and "Fuel In Tank (Sensor)" is not. The calculation for "Hours of Fuel Remaining" is based on "Estimated Fuel In Tank" and not "Fuel In Tank (Sensor)".

Based on the fact that your controller is reporting zeros for power when it is running, this will cause the "Hours of Fuel Remaining (Current Load)" to not be displayed. This value is not displayed when power is zero. If this is not what you are asking, I will need more details as I do not know how "Fuel in Tank (Sensor)" is not displayed while the generator is running. This is displayed all the time if you have an internal or external sensor.

bbrown-sc, 2021-01-25:
Here are screen shots from early this morning. This is when the generator was running at 12:34am. As you can see "Fuel Level Sensor" and "Fuel in Tank (Sensor)"  and not displayed.

This is at 4:32am when the power came back on and the generator had shut down,

It looks to me the "Hours of Fuel Remaining" on the first screen was calculated based on the "Estimated Fuel in Tank" and the second screen looks like "Hours of Fuel Remaining" is calculated on "Fuel in Tank (Sensor)".

I agree there's something wrong with how my generator is reporting power which may help clarify things, but there doesn't seem to be a reason to report fuel estimates when there is a fuel tank sensor present.

The other thing I just noticed is the "Run Hours in last 30 days" reports zero. Not including exercise, the generator has run ~100 hours in the past 30 days. How is that calculated?

jgyates, 2021-01-25:
I did not get your pictures. Pictures need to be attached in the web interface. If you attach them via email they will be stripped away by github.

any value "in the last 30 days" is calculated form the power log. You power log has a bunch of zero entries when there should be a valid value. The zero entries are interpreted as the generator is not running. Any calculation based on the power log will be off since your controller is outputting invalid data for power and that value is being logged.

also, how did you enable the External Fuel Sensor support? By directly editing the conf file or did you enable it via the GUI?

jgyates, 2021-01-26:
Also, if you are using any of the add on programs for your fuel sensor, when you first start genmon and the add on, the sensor reading will not display until the first successful reading from the add on. The above condition could occur if you had just started genmon. Once you get a successful tank level reading from your sensor then the value and all subsequent readings will be displayed.

bbrown-sc, 2021-01-27:
Hear are the 2 screen shots. The first was when the generator was running. I just now  reloaded genmon using the startgenmon.sh and the tank utility sensor information was there when I logged in within 30 seconds. I don't remember if I used the UI or edited the gentankutil.conf file. You recommended that I remove the Fuel Rate Half Load and Fuel Rate Full Load values from the advanced page. Will this in effect disable the power log since it's not working for me?

[image]
[image]

jgyates, 2021-01-27:
For an H100, yes, removing the half and full fuel consumption will disable the power logging.

jgyates, 2021-01-27:
I just checked in an update that may give you an option to try to make this work for you.

1) Update the software on the about page. The version number did not change 
2) on the advanced page enable the "Use Calculated Power" option and set the values for Half and Full Fuel Rate
3) Save your settings which will restart genmon 

When your run your generator you can check the power output on the status page. This value should be some value other than 0, 1, 2, 3, or 4. The new option *should* calculate the power based on average current, average voltage and power factor. I have tested this in a limited way as I do not actually have an H100 controller.

If the power displayed is accurate then this should also make the power log use the correct value. 

Let me know how it goes.

bbrown-sc, 2021-01-27:
Thanks for adding that. Next time my generator runs I will check the power output for the correct vales. I (or you) can close this now or wait until I can verify this.

closed 2021-01-27 by jgyates