jgyates/genmon #499: MQTT Last Will and Testament and Keep Alive

RubyBailey, 2021-01-31:
I was wondering if there was an reason why the Last Will and Testament or Keep Alive were not implemented in the mqtt module?
I use it to ensure that my mqtt clients are functional. In the case of genmon it would allow the mqtt broker and any monitoring software to respond appropriately on loss of the genmon mqtt client or the generator monitoring system.
Many thanks.

jgyates, 2021-01-31:
the default keep alive for genmqtt is 60 seconds. This is not presently a settable option because no one has asked for this to date. If you want it to be configurable to values other than 60 seconds let me know.

RubyBailey, 2021-01-31:
How is it then reported when the keep alive fails? I believe the standard approach is for the broker to close the connection and send the last will and testament (if one has been specified by the client). Since there isn't a LWT, does the broker just close the connection?
Regarding the keep alive timing, I think it's probably fine leaving it as is. I thinks is likely more important that the other clients and automation systems know that a client has dropped off the system. I just not sure how that's current achieved with genmon.
Thanks for the quick response.

jgyates, 2021-01-31:
This would depend on the broker, but the assumption is that if they did not receive a keep alive then the broker would close the connection. The add on genmqtt.py is using [the paho-mqtt python library](https://pypi.org/project/paho-mqtt/) library as the client implementation. This library does appear to have a last will feature however I am not currently using it simply because no one has asked. I do not use the mqtt add on with my generator so the features of the genmqtt program are based on user feedback. I can add this, but should this be always enabled or an option?

jgyates, 2021-01-31:
FYI, I do have a MQTT server that I use for other things (openenergeymonitor.org). I use the broker (mosquitto) for testing.

RubyBailey, 2021-01-31:
Thanks, yes I did check the paho-mqtt python library before asking the question. I didn't want to ask about something that couldn't be implemented. My question would have been answered at that point.
I also have a mosquito broker running on my system. Regarding always enabled versus an option; implementation of LWT seems to vary between systems, some enable LWT the moment mqtt is enabled others allow the user to configure it as an option. I have both types in my system. Personally, it really doesn't matter which approach is chosen, for those cases when it's always on when mqtt has been enabled, users can just ignore the LWT messages if they decide not to use it. There's no detriment to the system overall. The LWT message appears to be consistent between all the systems I use, "Online" and "Offline", although I do have a few that give you both the option of enabling it as well as customizing the LWT message. Retained is set to true (I know it's in the documentation, just thought I'd mention it as well)
In my opinion it really boils down to whenever is easier to implement, always enabled versus an option. Let me know if you want anything tested.

jgyates, 2021-01-31:
I checked in V1.15.17 which adds support for MQTT last will and testament. This line was added to the wiki:

The MQTT last will and testament topic is "generator/client_status" and the valid states are "Online" and "Offline". The MQTT keep alive value is 60 seconds.

Let me know if you have any issues.

RubyBailey, 2021-01-31:
I downloaded the updated code and it seemed to work fine. Then I subscribed to the topics on the broker and noticed that the LWT status message was "Offline" when genmon was online. Restarted genmon and I got and "Offline", then "Online" status. Looking at the updated code I believe the LWT topic needs to be set to "Online" **and retained** on_connect and then after connecting set will_set topic (same one) to "Offline" and retained. 
I think what's happening now is that since "Offline" is the only retained message that any client newly subscribing to the topic will only receive the "Offline" message when requesting genmon status. 
See example: [https://mntolia.com/mqtt-last-will-testament-explained-with-examples/](url)

jgyates, 2021-02-01:
I checked in an update (did not increase the version) that looks like it may fix the problem. Let me know how your testing goes.

RubyBailey, 2021-02-01:
I downloaded the update and tested it today. Works as expected. Thanks

closed 2021-02-01 by jgyates