jgyates/genmon #500: Outage, Error, Warn and Info in Notifications - is there a way to determine generator state elsewhere

RubyBailey, 2021-02-01:
It is possible to receive an email for Outage, Error, Warn and Info events through the Notification section of genmon. Is there a way either through mqtt or otherwise to determine quickly if any of these conditions are present on the generator? 
I know through mqtt I can get the various states, but how would I determine quickly if the state is say and "error" or "warning"? Or is there a different method that would provide this information? 
I'm assuming the error would mean it's critical and the generator will not start until the condition has been resolved, versus warning which means please check and service but it will still start. Like for example the difference between the red and yellow LED's.
I checked the issues but couldn't find anything about these questions. Thanks.

jgyates, 2021-02-01:
Since the Outage, Error, Warn and Info are classifications of messages instead of a state there is not direct correlation. 

However you can determine if the system is in outage (on and Evolution or Nexus controller) via the Outage page  ( or mqtt topic generator/Outage/System In Outage = Yes or No)

The Error classification is for internal software errors like reading the product ID register and finding an unsupported value. This is not a one for one but your can get the health of the generator monitor software on the Monitor page (mqtt generator/Monitor/Generator Monitor Stats/Monitor Health = OK or error condition)

The Warn classification is for generator alarms and other events related to maintenance. This is on the status page (mqtt topic generator/Status/Engine/System in Alarm = type of alarm, e.g. Overcrank, etc)

Other types of warnings are fuel low (your can get the fuel level in the Maint page) if you are using the fuel calculation or have a fuel sensor.
power log size is near the max (monitor page)

The mqtt topics generally match the layout of the we interface so you can monitor most things.

RubyBailey, 2021-02-01:
Thanks, I now understand that better. 
I was looking for a efficient way to mimic the green, yellow and red LED's on the generator to provide a summary of its current status. In some respect similar to what you do with genpgio.py which summarizes the status to eight (8) gpio pins.
It's easier to take actions based on the status of a single bit, byte or numeric value than to evaluate an incoming text message. Of course the other challenge is that I'm not familiar with all the options for those text messages.  
I'll have to think about this one a bit more.
Thanks again for your response and feedback.

jgyates, 2021-02-01:
I would look at generator/Status/Engine/Switch State and generator/Status/Engine/Engine State

Switch State is going to be "Off", "Manual", "Auto" or "System In Alarm"
Engine State is going to be one of the values returned from this function:

     https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L2791

I never worked out the entire logic for the LED states since some alarms are warning and some prevent the generator from running however this function:

    https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L3489

will read both switch and engine state and give one of eight different states. This is what the web interface uses to change the colors of the menu.

jgyates, 2021-02-01:
I am closing this issue for now. Feel free to post here if you have additional questions on this topic.

closed 2021-02-01 by jgyates

RubyBailey, 2021-02-02:
I am using your note from above to configure my automation system to recognize and react to engine and switch states. 
I am also curious to see if there's a way to determine the states of the red, yellow and green led's through software. I assume that you've done quite a bit of work on the modbus protocol and reverse engineering the register content. I don't want to redo anything that you've already done, but I would be interested in seeing if information about the state of those led's is in one of the registers. If your interested I would be willing to do some work on that topic. If so, please let me know and we can start a discussion on what you've done (so I don't repeat stuff) and where I should start.

jgyates, 2021-02-02:
You are welcome to look for this in the registers. 

There are five main different firmware variants for residential grade controllers (Nexus Air Cooled, Nexus Liquid Cooled, Evolution Air Cooled, Evolution Liquid Cooled, Evolution 2.0 Air Cooled). Additional firmware variants exist for PowerPact models.

If you want to watch the main registers change you can view them on the register page (single click the gear icon in the top right).

Another thing you can do is use the program in the  /OtherApps folder in the repository called modbusdump.py. This will dump a range of modbus registers. To show the program command line usage type:

        python modbusdump.py -h

in the OtherApps directory. This will dump registers 5 thru 100 decimal for a direct serial connection (serial over TCP is also supported)

      python modbusdump.py -r 9600 -p /dev/serial0 -a 9d -s 5 -e 100 

modbusdump.py can not be run will genmon is loaded as both programs expect exclusive access to the serial port.

If you have the Evolution Air Cooled diagnostic manual it states the conditions where red, blue, green and yellow lights are active. I had contemplated adding this to the UI however the issue is that I do not have a comprehensive list of alarm conditions and which ones keep the generator from running or starting. The LED color is different for these conditions. 

I have spent a lot of time looking at the functionality of the modbus data, however I have not been that interested in the LED status. There are a few registers that I have included in the set that genmon reads that change on some models and I do not know their function. These registers are labeled as "undefined" or "Unknown" on the register page. If you want to see if you can find a correlation between the LED color and these registers, this may be the best place to start. In general I have left out registers that do not change under any conditions.