jgyates/genmon #501: Nominal Line Voltage setting parameters

RubyBailey, 2021-02-01:
We are at the end of a long transmission line and our utility voltage is typically at the low end of the range, around 225-ish. On the status page the utility voltage gauge is in the low yellow range. I set the usenominallinevolts and nominallinevolts parameters in the genmon.conf file assuming that would adjust the utility voltage gauge so that 230V the center of the green area. Yet after making those adjustments the gauge still has 240V as the center of the green area. Did I misunderstand the description in the genmon.conf file?

jgyates, 2021-02-01:
Did you refresh your browser after making this setting? The gauges are only initialized with the max/min/nominal values one the page is loaded.

jgyates, 2021-02-01:
also, I would recommend not editing the genmon.conf file directly if you can make the changes in the GUI. The advanced page allows you to edit these two values.  The software will not read these unless it is restarted. You can access the advanced page by double clicking the gear icon in the top right of the web interface.

RubyBailey, 2021-02-01:
Ah, I didn't realize there was an advanced page. Well that makes it easier. Done. Now it works. Thanks

On a somewhat related issue; when I reverted my text file changes (prior to making the same changes in the advanced page) and then restarted genmon it complained that genmqtt.py was not running:
./startgenmon.sh restart
Restarting genmon python scripts
Stopping....
Stopping genmqtt.py
kill: (32691): No such process
Stopping genserv.py
Stopping genmon.py
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py
Starting /home/pi/genmon/genmqtt.py

Tried again and it said the same thing:
./startgenmon.sh restart
Restarting genmon python scripts
Stopping....
Stopping genmqtt.py
kill: (12714): No such process
Stopping genserv.py
Stopping genmon.py
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py
Starting /home/pi/genmon/genmqtt.py

It did start OK after sudo reboot.

I couldn't see any issues recorded in genmon.log (nor mymodbus.log and serial.log). Any suggestions where else to 
troubleshoot?

jgyates, 2021-02-02:
My guess is that you have a version of genmqtt.py that is hung and not exiting. I have seen this happen on occasion. My guess is that this is due to the mqtt python library. If you have a message in /var/log/genmqtt.log that states that genmqtt.py is already loaded then this is your problem.

RubyBailey, 2021-02-02:
Interesting, wonder why that would happen. I am running with the latest version of Paho MQTT Python, version 1.5.1 and python 2.7. "Error : The program genmqtt.py is already loaded" is the message in /var/log/genmqtt.log. I'll keep track of it and see if it comes up again.

RubyBailey, 2021-02-02:
To test I did sudo reboot and then back to back ./startgenmon.sh restart, (with about 1 minutes between restarts to ensure stability) results below. First restart appears fine, whereas the second notes the same problem as before. 
./startgenmon.sh restart
Restarting genmon python scripts
Stopping....
Stopping genmqtt.py
Stopping genserv.py
Stopping genmon.py
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py
Starting /home/pi/genmon/genmqtt.py

./startgenmon.sh restart
Restarting genmon python scripts
Stopping....
Stopping genmqtt.py
kill: (1313): No such process
Stopping genserv.py
Stopping genmon.py
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py
Starting /home/pi/genmon/genmqtt.py

genmqtt.log notes same issue as before.

jgyates, 2021-02-02:
Are you using a raspberry pi, if so then is it a pi zero or some other variant?

I checked in a new version that changes the order of the shutdown / cleanup for genmqtt.py that *may* help. I could not get it to fail. Let me know if you see any issues with this version.

RubyBailey, 2021-02-02:
Yes, genmon is currently running on a pi zero w. Was running on a pi3b and then I decided to use the pi zero because of the smaller physical footprint, it fit into an enclosure that I had available and in turn inside the generator compartment. 
I installed your newer version today and after three (3) ./startgenmon.sh restarts, now there are no issues and no error logs appear in genmqtt.log.
Now I'm curious; what did you change and any thoughts on why there would have been an issue with it running on a pi zero?
Thanks again.

jgyates, 2021-02-02:
The program genmqtt.py has one thread for monitoring genmon and the mqtt library uses another thread. I changed the order of the thread cleanup so the genmon monitor thread was stopped first and mqtt lib thread is stopped second. the pi zero has much less processing power than the regular pis so the thread cleanup can take longer on the zero.

closed 2021-02-02 by jgyates