jgyates/genmon #503: Pushover info and customizations

ravedoglv, 2021-02-09:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac 20K NG
- Genmon Logs: Submit via the About page
- Genmon Version: 1.15.18

Is there more information on what is sent to pushover? Which events? And could it be configurable as well? 

A cool thing would be to add in the notifications section if pushover is added and active to add custom notification parameters there as well.

thank you for such amazing software.

jgyates, 2021-02-09:
The add on programs genpushover, gensms, gensns_modem and genslack all use the same type of notifications. These notifications are sent when the generator changes state. Valid states are:

running
running in manual mode
exercising
generator entered auto mode
generator entered manual mode
generator entered off mode
generator service is due
generator is in alarm
generator controller detected an outage (if supported by your controller, Evo and Nexus only)

These notification states or types of notifications are not currently configurable. The notification method listed above is implemented in /genmon/mynotify.py and is designed for short single line messages.

The "Notifications" section in the web interface classifies email messages, which can be lengthy. It is basically the same types of events, just the email gets more data. Email gives you finer grain information and in a few cases it notifies you of additional data. The cases that email will notify but the add on notifications will not include: if an internal software error occurs (error type notifications), if the software detects fuel level is low by reading a sensor or calculating the fuel level (warn)l, if the power log has reached the maximum size (warn) or if a software updates is available (info). All other emails (like a change in generato state or an alarm) are covered in some way by the add on notifications.

I could add a few more add on notifications like software update, internal software errors to the types of notifications. I will put that on the list of things to add.

ravedoglv, 2021-02-09:
Wow. Thank you. Great info.

ravedoglv, 2021-02-09:
Welp. I could not find the mynotify.py file in the genmon dir...

jgyates, 2021-02-09:
sorry, it is [/genmon/genmonlib/mynotify.py](https://github.com/jgyates/genmon/blob/master/genmonlib/mynotify.py)

ravedoglv, 2021-02-09:
Thanks. Appreciate everything you’re doing.

jgyates, 2021-02-10:
I checked in V1.15.19 which includes the following change:

Updated mynotify.py to add notifications for software update, low fuel notice, and internal errors (e.g. communication errros etc.). This allows genslack, genpushover, gensms, gensms_modem, gensyslog and genemail2sms to support these additional notices.

Let me know if you have any issues. I am going to close this thread, but feel free to post here if you have questions or issues on this topic.

closed 2021-02-10 by jgyates