jgyates/genmon #504: Cannot connect to Genmon after upgrade and restart.

jcherney, 2021-02-11:
### Expected Behavior

Being able to login to my Genmon after upgrade and restart

### Actual Behavior

Genmon restarts, sends notification emails, but I cannot connect to it unless I pull the power and do a cold restart. I can ping the IP address, but it does't seem to be keeping the port information. 

### Steps to Reproduce (including precondition)

Seems to be happening with both of the last couple of upgrades. 

### Screenshot or Pictures relating to the problem (if possible)  

{Please write here}

### Your Environment

- Generator Model: {Please write here}QT04842ANAX
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}Version V1.15.19

jgyates, 2021-02-11:
My guess is that this is due the the flask library not cleaning up correctly.  Send your logs if you have them. It may contain some useful info, but other than that I don't have a lot to go on.

Instead of pulling the plug you should be able to log in via ssh and restart genmon via the command line:

once you log in via ssh then type:

      cd genmon
      ./startgenmon.sh restart

jcherney, 2021-02-11:
I sent the logs just after I posted. Do you need them again?

jgyates, 2021-02-11:
I have not received them. Is your outbound email working?

jcherney, 2021-02-11:
Yes. I just sent it again

jgyates, 2021-02-11:
Sill nothing, can you send a test email to yourself from the settings page?

jcherney, 2021-02-11:
Weird, because I got the restart emails. But this popped up after the test email:
An error occurred:
Error sending email: {'': (450, '4.7.1 : Sender address rejected: Access denied')}

Please try again.

jgyates, 2021-02-11:
Since the email address used to receive the log emails is a gmail address, this means that the gmail server is rejecting your email address or domain or it can not validate that your message is coming from a host with your domain. Are you using a private domain address in your sending email account?

jcherney, 2021-02-11:
Yes. I use ZoneEdit for my SMTP server. The email is Generator@universalbearings.com if you can whitelist it.

jgyates, 2021-02-11:
I can not white list it. Google is the one that is stopping it not me.

You can send them manually if that is easier. 

Enter this URL in your browser:

      http://192.168.1.10:8000/cmd/get_logs

 (change the IP address and port if needed for your setup. the "/cmd/get_logs" at the end is what is important. This will download a compress file with your logs. Send that file as an attachment to generatormonitor.software@gmail.com

jcherney, 2021-02-11:
Just sent from my gmail account. Should work. Thanks.

jgyates, 2021-02-11:
got them.

jgyates, 2021-02-11:
I see this error in the genserv.log file:

     2021-02-10 18:59:36,550 : Error : The program genserv.py is already loaded

This looks like it is in line with the flask library not shutting down properly theory. If you are running on a slower pi (like a pi zero or pi 2) this could be fixed by changing the file /etc/genmon/genloader.conf. This file can be edited with this command:

     sudo nano /etc/genmon/genloader.conf

In the [genmon] section, change this value:

      postloaddelay = 4

to this:

      postloaddelay = 6

This is the delay in seconds between loading genmon and trying to load another module (genserv.py, the web server). Increasing this delay may help, but that is just a guess.

To save the file type Ctrl+X, then Y, then Enter

On the next start or restart this delay will be used.

jcherney, 2021-02-11:
Just changed that, thanks. I'll see what happens on the next upgrade.

jcherney, 2021-02-11:
BTW- Its a more recent PI 3 Model B Rev 1.2

jgyates, 2021-02-20:
Closing for now. Let me know how your testing goes.

closed 2021-02-20 by jgyates

jcherney, 2021-02-27:
Did the upgrade this morning- same problem as before. Lost browser connection to the PI. Emails are working.

jgyates, 2021-02-27:
About all I can do is add more delay or completely change the web server to not rely on the internal flask server. The latter is not something I am willing to do at this point. I did just check in a minor change that will all up to 4 seconds delay for each module, if need. I did not change the version number so you can upgrade on the about page, but the fix will not be in place until your 2nd upgrade attempt.

jcherney, 2021-02-27:
Thanks. I’ll go pull the power and reboot and try that.

jcherney, 2021-02-27:
I think that may have worked! Rebooted twice, and it came back up twice. Not sure why it started acting weirdly after these years but that delay seems to have fixed it, thanks!

jgyates, 2021-02-27:
we will know more after the next upgrade. You can click the upgrade to test now and that will be the same type of test. The problem (as I see it now) is that an existing program is not exiting fast enough before an attempt is made to try to reload it.

To test you can just click the upgrade button, wait for the reload, if it comes back OK then try it again. repeat this 3 or 4 times,This should be a good test.

jcherney, 2021-02-28:
It’s working now. Did at least 5 reboots and they all came back up. Thanks for your help.

griffths, 2021-10-31:
Having similar/exact same issue.....

Can ping but no joy on web interface.

Tried to login via SSH but not sure what username/password combo to use.

Tried "root" with the password I set during original installation but get access denied.

Am able to pull logs via HTTP and have attached here.

[genmon_logs.tar.gz](https://github.com/jgyates/genmon/files/7449815/genmon_logs.tar.gz)

Also, is there a command to restart Pi via HTTP?

Thanks,

Jay

skipfire, 2021-10-31:
@griffths for SSH the default Raspbian login is pi:raspberry.

griffths, 2021-10-31:
@skipfire ,

That was it!

Thank you.

While I have you on the line, three suggestions/requests

I just redid my install with your Pi HAT and optional 36" harness with battery leads., much cleaner hardware install - satisfies my OCD well.

- What is your thought on adding a fuse to positive battery lead on your harness?
- I used the following wire loom on the battery leads to keep them a bit more protected and neat (OCD again) - maybe you can add to your harness as it would be easier before terminals are installed.

[image]

- Lastly, if/when you revise your HAT layout, it would be awesome if you could add a few terminals for the DS18B20 temp probes.  I am by no means an expert but it seems like it is simply a connection to the GPIO pins, which are now occupied by your HAT.

[Temperature probe kit](https://smile.amazon.com/gp/product/B087JQ6MCP/ref=ox_sc_act_title_1?smid=A1ALB8TZW3RG3D&psc=1)

Thanks again for the login name.

Regards,

Jay

skipfire, 2021-10-31:
I don't see a problem with you adding a fuse to your setup, just keep in mind that a fuse big enough to allow enough power to the Pi will be a fuse that will not protect against connecting things wrong, like plugging the HAT end of the cable into the generator controller would still fry the generator controller.

As for the temp probe, the one you sent appears to have 2 chips, I'm guessing a resistor and capacitor.  The bigger problem is physical space on the edges and increasing the cost for users not interested in it.  Some others have requested adding the appropriate stuff to allow the current transformers to be added, which would be another space taker.  I am considering either making a stackable HAT that has those other features, but I have 3 other projects related to GenMon I want to get available first.

griffths, 2021-11-01:
An update on my efforts...

Changed postdelay to 6

Was posting another no joy message when the GUI finally loaded. 

I have a feeling that my link speed is not optimal.  I temporarily moved the Pi into the gen housing since it didn't fit in my enclosure anymore with the new HAT installed.

Wi-Fi signal is marginal per the monitor screen and I suspect I am no longer getting full speed LAN to the Pi

Will move it back to the exterior once I get a new enclosure and update this thread with my results.

skipfire, 2021-11-01:
@griffths how bad is your signal, and does the value stay really steady?  My signal is a steady at -70 or -69 and I don't have any problems.

griffths, 2021-11-02:
I would classify it as marginal at best.  steady at -80 to -84.

[image]

Really just a whisper in a hurricane so I am sure the data rate has really taken a hit BUT so far no timeouts or other errors.

Ordered a new project box that will accommodate the Pi and your HAT.  I am sure once I get this thing back outside my generator enclosure my signal will improve substantially along with the data rate.

skipfire, 2021-11-02:
Yeah that is pretty bad.  I did find I was able to get about 5db improvement just on orientation of my Pi within my generator, and I did a lot of work to figure out what fit best for that.  When you get your project box please be sure to drop a couple of pics and part numbers.

griffths, 2021-11-02:
Will do

Also,  another suggestion for your consideration.   

Rather than adding additional options to your HAT, how about using an extended header (not sure of the official name) for the GPIO pins so I could stack another HAT above yours or, in my case, have access to the GPIO pins for my 1-wire temp probes?

I may attempt to swap the one on my HAT but see no reason why it could not have been used initially, although I am a NOOB with the Pi. 

Regards,

Jay

skipfire, 2021-11-02:
@griffths I do take requests for customizations.  I do not default to the stackable header because it is more expensive, significantly more effort for me to put on the board, requires larger boxing, additional packing to protect the pins, and additional risk of damage due to those long pins being bent, and I don't believe most people are looking for the stackable header in this case.  I am looking at multiple options to make temp & CT easily and cleanly done.

griffths, 2021-11-02:
@skipfire,

Thanks for the insight.  Anxious to see what you come up with. 

My OCD easily overcomes any fiscal common sense so you will likely have a customer for your accessory HAT or revised base HAT

Regards,

Jay

lmamakos, 2021-11-03:
What I did on [my board](https://github.com/lmamakos/RPi-pHat-Thermocouple) was to take a few extra (unused) GPIO pins from the raspberry PI header off to some 0.1" pins.  In my case, I selected the pins that are connected to an I2C controller, and [brought them over to some pin headers](https://github.com/lmamakos/RPi-pHat-Thermocouple#j9-i2c-expansion-connector).  Even had provision for pull-up resistors on the board.  That's a reasonably compact solution; and you can further reduce cost by not populating that connector, either making that a different SKU or just leaving it for those users to solder in the connector themselves.

griffths, 2021-11-20:
@skipfire 

Finally got around to completing the upgrade to a new project box to accommodate the Pi/HAT combo and temp sensor interface.

Posting a few pics and [part number](https://smile.amazon.com/dp/B098Q9YX46?psc=1&ref=ppx_yo2_dt_b_product_details) as requested.

[image]

[image]

bmg01907, 2022-05-02:
I was running v17.04 on a RPi 4B without any problems.  I just upgraded to v18.08 via the web interface.  After the upgrade finished I tried to access the Genmon web page (at 192.168.1.nnn:8000) and it just hung with the "Generator Monitor Loading" and rotating circle.  I rebooted the RPi and now Chrome reports the that site "refused to connect."  Firefox won't reach the site as well.  I've flushed the DNS Resolver cache and rebooted the router.  Neither of these actions had any effect.

I'd like to downgrade the Genmon software back to v17.04 and see if that solves the problem.  How can I do that?

Any other suggestions as to how to proceed?

jgyates, 2022-05-02:
Since it was a while since you updated the upgrade process would likely take longer. The firs thing I would suggest is loging into your pi via ssh and type

    cd genmon
     ./startgenmon.sh restart

The most note worthy change since V17 is the project now uses python 3 instead of python 2 since python 2 is deprecated and not included the raspbian anymore.

Also what is the output of this command on the pi console:

    cat /etc/os-release 

This will let us know if you are able to install python3 or if your version of raspbian is too out of date. If it is then you can run genmon with python2 but upgrades are not recommended until you upgrade the pi operating system version