jgyates/genmon #505: Gauges: drag n drop and resize, permanent?

ravedoglv, 2021-02-13:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: Version V1.15.19
- 

I noticed that you can rearrange and resize the gauges. However, for some reason, they don’t stick. When I navigate away and back, they are back in their default places. Is there something I’m missing?

jgyates, 2021-02-13:
No, you are not missing anything. You can rearrange and resize but saving the position is not currently implemented.

closed 2021-02-13 by jgyates

ravedoglv, 2021-02-13:
Good to know. One of my “am I crazy?” Moments. Not this time!