jgyates/genmon #507: Just did update, and now external fuel monitor tank utility does not work

grsthegreat, 2021-02-17:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

Prior to installing updates my external fuel meter worked, now it does not.

I went into settings and disabled and saved. Then I went in and re-enabled it and saved. Still shows empty and emails me tank is below 10%

jgyates, 2021-02-17:
If you are the one who just submitted their logs, you have this message in your log /var/log/gentankutil.log:

     Error : The program gentankutil.py is already loaded

Rebooting your pi will correct this.

grsthegreat, 2021-02-17:
Ok, thank you

closed 2021-02-17 by grsthegreat