jgyates/genmon #508: Office365 SMTP needs STARTTLS (not SSL/TLS)

jmone1, 2021-02-18:
Hi,
I've migrated my e-mail across to Office 365 and I'm trying to reconfig my E-Mail Settings, but I'm getting an error when using "Test" of:

SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:727)

I see in the [MS advice](https://support.microsoft.com/en-ie/office/pop-imap-and-stmp-settings-8361e398-8af4-4e97-b147-6c6c4ac95353?ui=en-US&rs=en-IE&ad=IE), I need to setup my SMTP settings as following, & I'm wondering if the error I see in Genmon is that the Encryption needing to be STARTTLS?
- Server: smtp.office365.com
- Port: 587
- Encryption: STARTTLS

Thanks
Nathan

jgyates, 2021-02-18:
You did not include your genmon mail settings. 

STARTTLS is the default for genmon's mail support. you settings should be:

Use SSL Encryption = Off
Disable TLS Encryption = Off ( this enabled STARTTLS)
Disable SMTP Auth = Off
your port should be 587

jmone1, 2021-02-18:
Fixed Thanks!  All the fun of moving from a (low reputation) E-Mail hoster to MS Exchange.  The Good news is you should have finally gotten my Registers / Log for the Australian Delivered 22KW + 3 Phase Transfer Switch.

closed 2021-02-18 by jmone1

jgyates, 2021-02-18:
Got them. Thanks! I am glad it is working for you.

jmone1, 2021-02-18:
:) - I've very happy now I can send logs etc.  The reason I wanted you send the logs, was the reported voltage seems waaaaay to high.  Australian 3-phase should be 400 volts with a tolerance range of + 10% to – 6%.  So I don't know if it truly is so high or that GenMon is reporting it oddly for my Unit.  I've not had any issues however so it may just be cosmetic.

jgyates, 2021-02-19:
I am not sure why it is so high. I don't have a lot to go on. The value that is normally utility voltage is reading 484V. The way I see it there are a couple of options.

1) you could do a more extended dump of your modbus registers. If you are interested in this let me know and I can give you instructions. I would need to know if you are using serial or serial over TCP. This may allow us to  find additional modbus registers that could allow the display of three phases, but that is just a guess, not guarantee we will find it or if this functionality exist.
2) You can modify the UI a bit on the advanced page (double click the gear icon in the top right of the web interface to get to the advanced page). On this page there are two settings that will allow you to adjust the nominal line voltage in the user interface. Set "Override nominal line voltage in UI" to the voltage you want (e.g. 484) and then set "Use Nominal Volts Override" to enabled. Then save you settings at the bottom of the page. Once the settings are saved, refresh your browser and the Utility Voltage gauge should be reset.

jgyates, 2021-02-19:
FYI, I did load your register data in my simulator and it does appear to be reading this value. I wonder if this is the combined voltage of the three phases?

jmone1, 2021-02-19:
I'm happy to test anyway you want.  I'm using Genmon on a PI connected via serial cable to the Gen Set.  I really don't think 484V makes any sense and I'd prefer not to just mask it in case there is a real issue.

jmone1, 2021-02-19:
FYI - my single phase rack UPS/ATS shows line voltage of 229-233v so that is correct (we are 230v nominal)

liltux, 2021-02-19:
Question how is the generator configured?  Air cooled or liquid cooled what type of transfer switch?  Are you sure the gen and switch are calibrated correctly?  I only say before looking too close at software may need to make sure the generator, generator controller, and possibly the switch controller are calibrated correctly.  Just a different view to add in.

jgyates, 2021-02-19:
Let get a dump of your modbus registers. In the /genmon/OtherApps folder there is a python program named modbusdump.py. once you log in to the pi via ssh, change directories:

     cd genmon

Then stop genmon:

     ./startgenmon.sh stop

Then get into the OtherApps folder:

     cd OtherApps

For the command line options type:

      python modbusdump.py -? 

Since you are using the serial port you need to use the sudo command to give you proper privileges:

     sudo python modbusdump.py -9600 -p /dev/serial0 -a 9d -s 0 -e 1527

 You can pipe the output to a file like this:

     sudo python modbusdump.py -9600 -p /dev/serial0 -a 9d -s 0 -e 1527 > output.txt

Then attached the output.txt file to this thread.

jmone1, 2021-02-19:
I've attached the file (FYI, missing a "r" --> sudo python modbusdump.py **-r** 9600 -p /dev/serial0 -a 9d -s 0 -e 1527 > output.txt)

[output.txt](https://github.com/jgyates/genmon/files/6007953/output.txt)

jgyates, 2021-02-19:
@liltux 

The register dump indicates it is an Evolution 2.0 Air cooled 20kw.

liltux, 2021-02-19:
OK, I was concerned it may have been a different controller.  Being that it is Utility volts, that is not able to calibrated with that controller.  @jmone1 may I ask what your voltage is measured at N1 and N2?  Do you have access to the Dealer menu on your Generator Controller?  You can view from the Dealer menus, Display menu and find Utility Volts listed.  We can compare what the controller value is to the registers being read.  It would be helpful.

jmone1, 2021-02-19:
I do have access to the Dealer Menus.  Do you mean measure the Volts physically at N1 and N2 with a multi meter or is it displayed in the Menu?

liltux, 2021-02-19:
Both would be ideal.  From the main screen (starts with status date/time) push this sequence into you controller pad: ESC, [up arrow], [up arrow], [down arrow], [down arrow], ESC, ESC.  Then you can navigate to the submenu tab, then to the dealer tab, then to the display tab.  There you can arrow down to Utility Voltage.

jmone1, 2021-02-19:
Utility Input Volts = 480 to 482.  (I used "Up Up Esc Dn Up Esc Up Enter" to get to the dealer menu).  I'm not tooooo keen to use my cheapo Multi Meter to try to measure what I presume is 480v (or am I wrong?).  I do see that I have access to what is marked as N1 and N2 in the Transfer Switch (which is easier to get to) but the wiring is light gauge compared to that used by the phases.  I've attached some pics

[image]
[image]

liltux, 2021-02-19:
That is enough to match up with the register information.  I would ask the local dealer why it displays 480 volt and not the correct 400 volt?  @jgyates it would appear that is the correct register.

jmone1, 2021-02-19:
It's a good idea..... Unfortunately these units are not common in Australia, and the "dealer" that sold it to me did not want to do the install (just supply) and my sparky had never seen one before.  

Another thought, is if I simulate a power failure so the unit kicks it, will GenMon capture the registers etc so I can send them to you from the GUI? (just too see what other weirdness is happening??)

jgyates, 2021-02-19:
yes, as long as the pi is powered that should work.

jmone1, 2021-02-19:
Will do now (it will take a bit for the Pi to power back on after the failure).

jmone1, 2021-02-19:
Ok, is is running now and the House is up and powered.  I've sent the registers and log.  Here is a pic, and the Volt looks good but the  (oddly Power Output is 0)
[image]

jmone1, 2021-02-19:
....and here is my measured house usage
[image]

jmone1, 2021-02-19:
OK - I'm about to shut it back down but I've also sent a 2nd set of Logs/Registers (has been running for about 10min now without issue).

jmone1, 2021-02-19:
FYI - I love how quick the transfer switch is in failing back.  No power drops to any of my appliances

Stadica, 2021-02-20:
Hi jmone1 / jgyates / liltux , I have a 2018 13kw Honeywell (rebranded generac) with genmon pi now installed recently in Australia, if I can be of any assistance to provide logs from a single phase 50hz unit i would be more then happy to help out and make such a great tool better for all. Out of curiosity jmone1, I measured my network voltage this morning at 241 and the generator was displaying 242 on the controller and in genmon, so am pretty happy to say its reading the correct voltage. If you dont trust your meter incase you are actually getting such high volts from your network distributor, in the transfer box measure across N1 and N2 and that will show you the single phase (240V ish) sent to the generator to sense the loss of the network supply. From what I understand genmon is only displaying what the controller shows, so if your volts at n1/n2 are around the 240v mark there is something wrong in the controller as it calculates approx what the 3 phase voltage would be based off the single phase supply it is sent. For the controller to be seeing 480V i would guess single phase would have to be 277V (277 x 1.732 =480Vish). If your volts are up that high, i would be straight on the phone to your network provider (not retailer) for your area to have them investigate.

jmone1, 2021-02-20:
OK - Not I know it is "only" 230v to look for, I tested (at the main fuse box), N1 to N2 and all Three Phases.  All of them read around 234v

Stadica, 2021-02-20:
I’d dare say there is a issue with the generac controller firmware with the calculation used to perform the single to three phase voltage conversion from n1 and n2. When the gen is running it supplies voltage from its winding to the controller on different terminals and I’m assuming as my diag manual only shows a single phase unit ( 3 phase was not available in oz when I didn’t my training)  but it possibly sensing only across one phase and doing the same calc. Might be worth raising  with generac themselves as I’d dare say 3phase units this size would be fairly uncommon in the states as their electrical system is different to ours and they havnt been quizzed about this issue before. My mate has installed a few 3 phase units now for people and I will ask him if he has noticed anything strange with volts.

jmone1, 2021-02-20:
Sounds good.  As it appears to be just "cosmetic" I don't really mind.

jmone1, 2021-02-22:
@jgyates  - any hints from the logs of why Genmon is showing no (or very low) Power Output when running?

jgyates, 2021-02-22:
The data I have seen is showing that the values displayed by genmon are what the controller is reporting. Genmon does not modify these values. I did look for additional registers that could be relevant to the 3 separate line voltages but I have not found anything that looks promising.

jmone1, 2021-02-22:
Thanks mate!