jgyates/genmon #511: Feature Request: Notification support for selecting notification type (email, Twillo, Pushover etc)

sarahmva, 2021-02-26:
It seems that the Notification configuration only allows configuring email notifications.

It would be nice, if there would be a way to add a notification and select the notification type (e.g. Twillo, Slack, Pushover, Email). The recipient field would then be the configuration for the notification service (e.g. recipient phone number in Twillo).

This would allow setting up multiple recipients for services like twillo and send different types of notifications to the recipients. 

As an alternative, it would be nice to set multiple recipients in Twillo and allow to select what triggers an SMS to be sent.

jgyates, 2021-02-26:
This thread describes the issue with your first request. https://github.com/jgyates/genmon/issues/503

In short, short message services are not suited for long messages. The notification tab is for long messages (email) short messages are handled by add ons. The contain very different info.

I can add multiple Twilio recipients to the list of features to add.

https://github.com/jgyates/genmon/blob/master/feature_requests.md

I will close this issue for now. Once this is implemented I will update this thread.

closed 2021-02-26 by jgyates

jgyates, 2021-02-26:
I have added the ability to send to multiple recipients using gensim (twilio client). Separate recipient numbers with a comma. Update to V1.16.00