jgyates/genmon #512: Can't connect after recent software upgrade

Newtonite, 2021-02-27:
After upgrading to the latest version, I'm getting a "ERR_CONNECTION_REFUSED" error when I try to connect from my Chrome browser. I think the upgrade went successfully at least, because I received an email saying that the generator was restarting.

Where do I go from here?

liltux, 2021-02-27:
Reboot your pi.  Can you ssh or vnc to your pi?  If so you can reboot.

jgyates, 2021-02-27:
This thread has more details on the solution.

jgyates, 2021-02-27:
https://github.com/jgyates/genmon/issues/504

jgyates, 2021-02-27:
In the thread mention above I reference a fix I just checked in. Let me know if the next upgrade corrects this issue.

closed 2021-02-27 by jgyates

Newtonite, 2021-02-27:
Yes, that worked. I'll report back what happens with the next upgrade. Thanks!