jgyates/genmon #514: question: does it matter if serial says `None - Controller has been replaced`

scyto, 2021-03-01:
My Nexus controller died in Dec, had them replace it.
For serial number it now says "None - Controller has been replaced" does this matter?

Alex

liltux, 2021-03-01:
Not a problem at all.  There is a place on the advanced settings page to force the serial number.  This is in the top right corner, once is the registers, twice is the advance page.

closed 2021-03-01 by jgyates

jgyates, 2021-03-01:
A little more info: the serial number is used to look up the model for Evolution / Nexus controllers. If the controller has been replaced the serial number is blank. Adding the serial number on the advanced page allows the model number lookup to take place. The look up only occurs during the loading of the software after an upgrade.

scyto, 2021-03-06:
> Not a problem at all. There is a place on the advanced settings page to force the serial number. This is in the top right corner, once is the registers, twice is the advance page.

thanks, just need to find my serial number now :-)

jgyates, 2021-03-06:
There should be a plate on the inside of the generator enclosure that has the model, SN, etc