jgyates/genmon #516: TankUtility integration not functional (2 tanks)

klinquist, 2021-03-01:
### Expected Behavior

Level of one of my tanks shown!

### Actual Behavior

Tank level shows empty

### Steps to Reproduce (including precondition)

I have two tanks in my TankUtility account.  I have correctly specified the name - as verified by manual calls to the TankUtility API:

Result of `GET /devices `
```
{"devices":["0023001d3531323603473833","00270019333031310c473832"]}
```

Result of `GET /devices/0023001d3531323603473833` (the tank I want to monitor here)

```
{"device":{"device_id":"0023001d3531323603473833","short_device_id":"J9LGDN96","name":"MotoRetreat_500","address":"xxxx, USA","account_id":"","fuel_type":"propane","status":"deployed","capacity":500,"orientation":"horizontal","consumption_types":"","battery_warn":false,"battery_crit":false,"fixed_transmission_time":-1,"reading_interval":21600,"transmission_interval":86400,"threshold_1":-1,"threshold_2":-1,"lastReading":{"tank":54.746,"temperature":39.097,"time":1614587126000,"time_iso":"2021-03-01T08:25:26.000Z","sw_rev":"4.078"}}}
```

Latest Genmon 1.16.0

[image]

klinquist, 2021-03-01:
I've changed the title as I think I've identified the issue
I found gentankutil.log and saw the following entries:
```
2020-11-26 07:53:49,188 : Error : The program gentankutil.py is already loaded
2020-11-26 07:55:51,510 : Error : The program gentankutil.py is already loaded
2020-12-11 08:55:53,057 : Error : The program gentankutil.py is already loaded
2020-12-14 09:01:09,491 : Error : The program gentankutil.py is already loaded
2020-12-15 07:59:40,105 : Error : The program gentankutil.py is already loaded
2020-12-16 06:46:09,801 : Error : The program gentankutil.py is already loaded
2021-01-07 08:30:05,978 : Error : The program gentankutil.py is already loaded
2021-01-07 08:31:13,096 : Error : The program gentankutil.py is already loaded
2021-01-07 08:34:33,323 : Error : The program gentankutil.py is already loaded
2021-01-16 08:48:14,874 : Error : The program gentankutil.py is already loaded
2021-02-07 14:04:51,417 : Error : The program gentankutil.py is already loaded
2021-02-10 14:10:30,786 : Error : The program gentankutil.py is already loaded
2021-03-01 12:11:33,043 : Error : The program gentankutil.py is already loaded
```

I think those are the dates/times that I've upgraded Genmon.

I rebooted the pi... and when it came back up, it correctly showed the level.

jgyates, 2021-03-01:
If you had not rebooted your pi since the first entry, then this is due to a single instance of the program not exiting correctly. Once a copy of the program is abandoned in memory you either have to reboot or manually kill the process. Let me know if this occurs again.

jgyates, 2021-03-02:
I checked in an update (V1.16.01) that should help prevent this from happening.

closed 2021-03-02 by jgyates